zephyrproject-rtos/zephyr #38340: Bluetooth:DirectionFinding: Disabling the MPU causes some compilation errors

saleh-unikie, 2021-09-06:
**Describe the bug**
In order to use "Settings" based on NVS file storage, it is needed to disable MPU by adding this config:
`CONFIG_ARM_MPU=n`

When this config is added, some compiler errors and warnings pop up, when trying to build the "direction_finding_connectless_rx" example:
https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/direction_finding_connectionless_rx

**To Reproduce**
Steps to reproduce the behavior:
1. Open <Zephyr Location>/samples/bluetooth/direction_finding_connectionless_rx
2. Append `CONFIG_ARM_MPU=n` to `prj.conf` file
3. west build -b nrf52833dk_nrf52833
4. See error

**Expected behavior**
Build without errors and warnings

**Impact**
It is not possible to use Settings while receiver of radio direction is enabled.

**Logs and console output**
Some parts of the errors:
```
[1/9] Building C object zephyr/subsys/bluetooth/contr...roller.dir/ll_sw/nordic/hal/nrf5/radio/radio_df.c.obj
FAILED: zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/ll_sw/nordic/hal/nrf5/radio/radio_df.c.obj
.
.
.
In file included from /home/zephyr/zephyrproject/zephyr/include/sys/util_macro.h:34,
                 from /home/zephyr/zephyrproject/zephyr/include/sys/util.h:17,
                 from /home/zephyr/zephyrproject/zephyr/include/devicetree.h:22,
                 from /home/zephyr/zephyrproject/zephyr/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/radio/radio_df.c:9:
/home/zephyr/zephyrproject/zephyr/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/radio/radio_df.c:44:7: warning: implicit declaration of function 'NRF_DT_GPIOS_TO_PSEL' [-Wimplicit-function-declaration]
   44 |      (NRF_DT_GPIOS_TO_PSEL(RADIO, dfegpio##idx##_gpios)),   \
      |       ^~~~~~~~~~~~~~~~~~~~
.
.
.
zephyr/include/generated/devicetree_unfixed.h:3910:38: error: 'DT_N_S_soc_S_radio_40001000' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_radio_40001000_ORD'?
 3910 | #define DT_N_NODELABEL_radio         DT_N_S_soc_S_radio_40001000
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
.
.
.
/home/zephyr/zephyrproject/zephyr/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/radio/radio_df.c:44:35: error: 'dfegpio0_gpios' undeclared here (not in a function)
   44 |      (NRF_DT_GPIOS_TO_PSEL(RADIO, dfegpio##idx##_gpios)),   \
      |                                   ^~~~~~~
.
.
.

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: 86f94eb

**Additional context**
Adding the `#include <soc.h>` to [radio_fc.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/radio/radio_df.c) file, solves the above errors (I am not sure if it is the solution or not), but still these warnings are visible:
```
[139/238] Building C object zephyr/subsys/bluetooth/c.../subsys__bluetooth__controller.dir/ll_sw/ull_df.c.obj
In file included from /home/zephyr/zephyrproject/zephyr/subsys/bluetooth/controller/ll_sw/nordic/hal/cpu_vendor_hal.h:7,
                 from /home/zephyr/zephyrproject/zephyr/subsys/bluetooth/controller/./hal/cpu.h:7,
                 from /home/zephyr/zephyrproject/zephyr/subsys/bluetooth/controller/ll_sw/ull_df.c:13:
/home/zephyr/zephyrproject/zephyr/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/cpu.h: In function 'cpu_sleep':
/home/zephyr/zephyrproject/zephyr/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/cpu.h:13:2: warning: implicit declaration of function '__WFE' [-Wimplicit-function-declaration]
   13 |  __WFE();
      |  ^~~~~
/home/zephyr/zephyrproject/zephyr/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/cpu.h:14:2: warning: implicit declaration of function '__SEV' [-Wimplicit-function-declaration]
   14 |  __SEV();
      |  ^~~~~

```

ppryga, 2021-09-24:
@saleh-unikie I couldn't add you to reviewers list in the PR: https://github.com/zephyrproject-rtos/zephyr/pull/38819. The github is not showing you when I search in add reviewer box. Could you take a look on the PR and confirm whether the issue is fixed in your application?

saleh-unikie, 2021-09-24:
Yes, It has fixed the warning problem. Thanks!

ppryga, 2021-09-24:
Great! 
The issue may be closed when the PR is merged.

ppryga, 2021-09-24:
PR is merged. I'm closing the issue.

closed 2021-09-24 by ppryga