zephyrproject-rtos/zephyr #38346: twister command line parameter clean up and optimizate twister documents

hakehuang, 2021-09-06:
in discussion #38140, we'd like to improve the twister commands, and its documentation.

**Describe the solution you'd like**
we need:

- all the supported features shall be listed in help
- all paramenter shall have an example
- all parameter combination and dependency shall be well documented
- for given command lint all parameters shall be print out and easy to be searched(grep)
- need find a way that parameter is organized in one place. e.g the way with zephyr\boards\arc\nsim\support

**Describe alternatives you've considered**
currently twister use ArgumentParser.

hakehuang, 2021-09-06:
@PerMac @abrodkin

nashif, 2022-11-19:
twister went through a major overhaul over last year, command line parameters reorganised and documented, for anything that might remains, which I am not aware of, please file a bug.

closed 2022-11-19 by nashif