zephyrproject-rtos/zephyr #38348: Bluetooth: Switch to inclusive terminology from the 5.3 specification

carlescufi, 2021-09-06:
The inclusive naming mappings can be found here:
https://btprodspecificationrefs.blob.core.windows.net/language-mapping/Appropriate_Language_Mapping_Table.pdf

Areas to be covered:

- [ ] include/bluetooth: @carlescufi 
- [ ] subsys/bluetooth/host
- [ ] subsys/bluetooth/controller: @cvinayak 
- [ ] subsys/bluetooth/mesh
- [ ] subsys/bluetooth/<other>
- [ ] drivers/bluetooth/
- [ ] samples/bluetooth
- [ ] tests/bluetooth

carlescufi, 2021-09-06:
CC @jhedberg @joerchan @alwa-nordic @asmk-ot @wopu-ot @kruithofa @Thalley @Vudentz @asbjornsabo @sjanc @andrzej-kaczmarek @KKopyscinski @trond-snekvik @LingaoM @michal-narajowski @ppryga

Thalley, 2021-09-06:
@carlescufi Is there anything in `subsys/bluetooth/audio` that would require a change?

carlescufi, 2021-09-07:
API meeting:

- The version of the specification that replaces the terms with inclusive ones is 5.3
- All Bluetooth subsystems will claim compliance with the 5.3 version of the spec in Zephyr 2.7.0
- General consensus is to allow for the change, as long as it only involves replacing the terms and is well documented in the release notes for 2.7.0
- Requires TSC approval

carlescufi, 2021-09-16:
> @carlescufi Is there anything in `subsys/bluetooth/audio` that would require a change?

I am not sure about that @Thalley. Can you check?

Thalley, 2021-09-16:
> > @carlescufi Is there anything in `subsys/bluetooth/audio` that would require a change?
> 
> I am not sure about that @Thalley. Can you check?

A quick course search reveals nothing. Do we have a list of words that has/should be changed? Should be able to write a fairly quick script to validate.

carlescufi, 2021-09-21:
All done!

closed 2021-09-21 by carlescufi