zephyrproject-rtos/zephyr #38349: XCC compilation fails on Intel cAVS platforms

kv2019i, 2021-09-06:
**Describe the bug**
Recently merged https://github.com/zephyrproject-rtos/zephyr/pull/37647 broke SOF builds with Xtensa XCC compiler. Gcc builds still work on (with Zephyr SDK version).

**To Reproduce**
Build SOF with Zephyr as RTOS.

**Expected behavior**
SOF build works.

**Impact**
Severe performance impact as GCC doesn't support xtensa intrinsics.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

```
244/252] Linking C executable zephyr/zephyr_prebuilt.elf
[...]
zephyr/linker_zephyr_prebuilt.cmd:481: non constant or forward reference address expression for section .noinit
```

kv2019i, 2021-09-06:
We are temporarily disabling XCC builds in SOF CI: https://github.com/thesofproject/sof/pull/4719 
This does mean we have no CI testing XCC builds, so risk of further regressions in Zephyr main is now a real risk. Thus this bug should have priority (not just the performance impact, but to ensure we can reenable XCC compilation testing).

marc-hb, 2021-09-07:
Note Zephyr PR #37647 (or any other Zephyr PR) could have been tested in SOF CI before merge like this:
```diff
--- a/scripts/xtensa-build-zephyr.sh
+++ b/scripts/xtensa-build-zephyr.sh
@@ -81,6 +81,11 @@ clone()
 	mkdir -p "$WEST_TOP"
 	git clone --depth=5 https://github.com/zephyrproject-rtos/zephyr \
 	    "$WEST_TOP"/zephyr
+
+	# Temporarily testing a zephyr PR not merged yet
+	git -C "$WEST_TOP"/zephyr fetch    pull/37647/merge
+	git -C "$WEST_TOP"/zephyr checkout FETCH_HEAD
+
 	git -C "$WEST_TOP"/zephyr --no-pager log --oneline --graph \
 	    --decorate --max-count=20
 	west init -l "${WEST_TOP}"/zephyr
```
https://github.com/thesofproject/sof/blob/e963aa0/scripts/xtensa-build-zephyr.sh

closed 2021-09-08 by nashif