zephyrproject-rtos/zephyr #38351: Custom radio protocol

mrninhvn, 2021-09-06:
**Describe the solution you'd like**
Some custom protocol like logitech unifying are very interesting, are there any instructions to add to my radio protocol?

jukkar, 2021-09-07:
For example IEEE 802.15.4 has a radio API described in `include/net/ieee802154_radio.h`, search `ieee802154_radio_api` struct for details. That API separates the L2 protocol layer from the device drivers. You can do things in similar ways in your radio protocol.

carlescufi, 2021-09-07:
@mrninhvn if what you are looking for is raw 2.4GHz access to the radio to implement a radio protocol, then I recommend looking at Nordic's implementation of ESB here: https://github.com/nrfconnect/sdk-nrf/tree/master/subsys/esb
There's also a PR to add Gazell support: https://github.com/nrfconnect/sdk-nrf/pull/5281

closed 2021-09-07 by cfriedt