zephyrproject-rtos/zephyr #38352: IEEE 1588-2008 support

alsrgv, 2021-09-06:
**Is your enhancement proposal related to a problem? Please describe.**
We would like to implement support of IEEE 1588-2008 time synchronization mechanism using Zephyr.

Our immediate need is to implement masterOnly and slaveOnly clocks for STM32H7 that would synchronize to other IEEE 1588-2008 clocks using traditional off-the-shelf unmanaged switches via delay request/response mechanism. We could implement it using L2 transport or UDP, and would need to build HW timestamping support in STM32H7 network driver. Since we have a specific use case, we were planning to keep the PTP implementation on the application side, and upstream STM32H7 driver improvements.

We've started attacking the problem from two sides - build out timestamping support into STM32H7 driver, while also prototyping PTP code using `frdm_k64f` board. The timestamping support seems straightforward, but we got few design questions trying to build out PTP code. Here they are:

1. In all network drivers, timestamping is predicated on the packet being gPTP packet. What is a good way to add timestamping support for PTP?

2. If we implement Ethernet-based PTPv2 protocol, we need to relax some constraints about treating all packets with type `NET_ETH_PTYPE_PTP` as gPTP packets. We're okay to consume those via network stack in the application (i.e. not having benefit of processing time-sensitive packets out of order), but they should not be dropped since they don't match gPTP requirements. What is a good way to un-tie that dependency?

3. Some network drivers, including `eth_mcux`, only allow gPTP multicasts to reach the network stack - even if one executes `net_ipv4_igmp_join` from the application. What is a good way to add other multicast subcriptions, e.g. for PTP?

In the short term, we could solve all (3) using ugly hacks, but we'd like to implement our changes in a way that would pave the way for them to be upstreamed.

If there is a better platform to have this design dialog, would be happy to move it there!

cc @jukkar

alsrgv, 2021-09-07:
*Update:*

It looks like if we use L2 transport for PTP and relax timestamping & multicast filtering to not require gPTP (e.g. with change like this: https://github.com/maka-autonomous-robotic-systems/zephyr/commit/5d9ca7b95befd2060477b011ee9dcff134b2887b), we can accomplish the rest of logic in the library/application code:

1. Receive L2 PTP packets using `net_context_get(AF_PACKET, SOCK_RAW, ETH_P_ALL, &raw_context)` - we can't use socket API because we need to retrieve net_pkt's timestamp.

2. Send L2 PTP packets using `net_if_queue_tx()` and track their transmission timestamps using `net_if_register_timestamp_cb` - can't use `net_context_sendto` since it doesn't allow tracking packets, even in the callback.

Does that approach and commit https://github.com/maka-autonomous-robotic-systems/zephyr/commit/5d9ca7b95befd2060477b011ee9dcff134b2887b look reasonable?

jukkar, 2021-09-07:
> It looks like if we use L2 transport for PTP and relax timestamping & multicast filtering to not require gPTP (e.g. with change like this: [maka-autonomous-robotic-systems@5d9ca7b](https://github.com/maka-autonomous-robotic-systems/zephyr/commit/5d9ca7b95befd2060477b011ee9dcff134b2887b)), we can accomplish the rest of logic in the library/application code:

The changes in that commit look a bit suspicious. IMHO the `CONFIG_NET_PKT_TIMESTAMP` is not proper check here, as that config only tells that the timestamp to be stored in net_pkt struct and that the net_if supports timestamping in the first place. Better would be to create a new config option that sets or depends on the `CONFIG_NET_PKT_TIMESTAMP` option so that it is easier to manage the dependencies.

>     1. Receive L2 PTP packets using `net_context_get(AF_PACKET, SOCK_RAW, ETH_P_ALL, &raw_context)` - we can't use socket API because we need to retrieve net_pkt's timestamp.

makes sense

> 
>     2. Send L2 PTP packets using `net_if_queue_tx()` and track their transmission timestamps using `net_if_register_timestamp_cb` - can't use `net_context_sendto` since it doesn't allow tracking packets, even in the callback.

looks ok
> 
> 
> Does that approach and commit [maka-autonomous-robotic-systems@5d9ca7b](https://github.com/maka-autonomous-robotic-systems/zephyr/commit/5d9ca7b95befd2060477b011ee9dcff134b2887b) look reasonable?

Without seeing the actual patch it is difficult to say if it would be ok or not.

I am not fully understanding what you are trying to do here. So you want to direct all PTP messages to application to handle the time sync and skip / disable the existing gPTP stack. Have you considered extending the gPTP code to handle IEEE 1588 messages/logic that are currently not there.

alsrgv, 2021-09-07:
Yes, the plan is to handle the time sync logic in the application.

We would love to extend gPTP stack to handle IEEE 1588. What's the best approach for that? gPTP (IEEE 802.1AS) is a restricted subset (profile) of PTP (IEEE 1588), so it seems a bit odd to start extending it to cover PTP use cases.

Perhaps good first step would be to introduce `CONFIG_NET_PTP` option which would be set by `CONFIG_NET_GPTP`, update checks discussed above that depend on `CONFIG_NET_GPTP` to use `CONFIG_NET_PTP` instead, and rename `net_pkt_(is|set)_gptp` to `net_pkt_(is|set)_ptp`?

jukkar, 2021-09-07:
> Yes, the plan is to handle the time sync logic in the application.
> 
> We would love to extend gPTP stack to handle IEEE 1588. What's the best approach for that? gPTP (IEEE 802.1AS) is a restricted subset (profile) of PTP (IEEE 1588), so it seems a bit odd to start extending it to cover PTP use cases.

I am just wondering why introduce duplication as zephyr has already lot of generic PTP code that can be used in IEEE 1588. So personally I feel that it would be better to extend the current gPTP code for IEEE 1588 so that we would get better support for time sync in the upstream code base.

> 
> Perhaps good first step would be to introduce `CONFIG_NET_PTP` option which would be set by `CONFIG_NET_GPTP`, update checks discussed above that depend on `CONFIG_NET_GPTP` to use `CONFIG_NET_PTP` instead, and rename `net_pkt_(is|set)_gptp` to `net_pkt_(is|set)_ptp`?

That sounds like a good plan :+1:

alsrgv, 2021-09-08:
Raised a PR: #38379. Would appreciate guidance to make it better!

jukkar, 2024-02-08:
Old issue with no activity for over two years so closing this one.

closed 2024-02-08 by jukkar