zephyrproject-rtos/zephyr #38354: stm32: stm32f10x JTAG realated gpio repmap didn't works

ck-telecom, 2021-09-07:
For stm32f103 series when remap JTAG related GPIO for SPI1 function, it seems didn't works

The pinmux driver seems to need disable JTAG enable SWD and enable RCC->APB2ENR clocks which related gpio port

FRASTM, 2021-09-07:
@ck-telecom  Could you please detail the error you are  seeing and the sequence you are applying 
Based on the stm32f103 datasheet pins, for the SPI1 are mapped on the PA4, PA5, PA6, PA7

ck-telecom, 2021-09-07:
&spi1 {
	pinctrl-0 = <&spi1_nss_master_pa15 &spi1_sck_master_pb3
			&spi1_miso_master_pb4 &spi1_mosi_master_pb5>;
	status = "okay";
	cs-gpios = <&gpioa 15 GPIO_ACTIVE_LOW>;

	l3g4200d: l3g4200d@0 {
		compatible = "st,l3g4200d";
		label = "GYROSCOPE";
		spi-max-frequency = <8000000>;
		reg = <0>;

		irq-gpios = <&gpiob 12 GPIO_ACTIVE_LOW>;
	};
};

my board dts show above and when reading chip ID, the chip id is 0xFE with JLINK and 0x00 whitout JLINK，for legacy stm32 library, we using the follow APIs to remap SPI1:
RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOB | RCC_APB2Periph_AFIO,ENABLE);
GPIO_PinRemapConfig(GPIO_Remap_SWJ_JTAGDisable, ENABLE); 

but I find that pinmux driver using stm32 LL library, the register should set correct
RCC->APB2ENR = 1 |  (1<<3);  /*AFIO/ GPIOB*/
	
AFIO->MAPR = (AFIO->MAPR & ~((uint32_t)0x7 << 24)) | (2 << 24); // disable JTAG and enable SWD

erwango, 2021-09-13:
@ck-telecom thanks for the details. Let me see if I understood correctly.
The pin configuration used in your SW is in conflict with SWJ default config (mapped on PA13/PA14/PB3/PB4).
So you need a way to remap these pins using AFIO_MAPR(SWJ_CFG[2:0]). Is that correct ?

If yes, then this could be implemented as a specific additional static configuration.
I don't see any easy way to implement this using device tree, so I propose to introduce a dedicated Kconfig symbol:
CONFIG_STM32_REMAP_SWJ_FULL
CONFIG_STM32_REMAP_SWJ_FULL_NO_NJTRST
CONFIG_STM32_REMAP_SWJ_NO_JTAG
CONFIG_STM32_REMAP_SWJ_NO_JTAG_NO_SWJ

Would it work for you ?

ck-telecom, 2021-09-14:
@erwango Yes I think it should be work, and just NOTE the stm32f10x errata http://www.st.com/content/ccc/resource/technical/document/errata_sheet/7d/02/75/64/17/fc/4d/fd/CD00190234.pdf/files/CD00190234.pdf/jcr:content/translations/en.CD00190234.pdf SPI1 remap with I2C1 conflict each other

erwango, 2021-09-14:
> @erwango Yes I think it should be work, and just NOTE the stm32f10x errata 

> http://www.st.com/content/ccc/resource/technical/document/errata_sheet/7d/02/75/64/17/fc/4d/fd/CD00190234.pdf/files/CD00190234.pdf/jcr:content/translations/en.CD00190234.pdf SPI1 remap with I2C1 conflict each other

Ok, thanks for the heads up but there is nothing we can do for that.
This should be taken care by users when configuring/designing their device.

FRASTM, 2021-09-15:
However, the pins which are possibly conflicting are:
PA15 : SPI_NSS or JTDI
PB3 : SPI1_SCK   or JTDO
PB4 :  SPI1_MISO or JNTRST

PB5  :      SPI1 MISO or I2C1_SMBAI 

Could we introduce a single flag to select the JTag or SPI1 for PA15, PB3, PB4  globally   (something like _CONFIG_SWJ_NOT_SPI_) or should have each pin selectionnable (one flag per PIN).

I think the question of PB15 should be handled separately.
In your case, because you are using the SPI1_MISO on PB5 (alternate function), the remap applies and consequently the I2C1 must not be clocked, according to the Errata sheet. 
--> could this be done with 
`status = "disabled";`
directly in your DTS ?

On the other side, there is no reason to have
```
&i2c1 {
	pinctrl-0 = <&i2c1_scl_pb8 &i2c1_sda_pb9>;
	status = "disabled";
```
in the _./boards/arm/nucleo_f103rb/nucleo_f103rb.dts_, because the PB5 is not used there (the errata does not apply).

FRASTM, 2021-09-15:
It appears that for the STM32F1x soc series, it already exists in the _./drivers/gpio/gpio_stm32.c_

Could you try to set **CONFIG_GPIO_STM32_SWJ_NOJTAG=y** in your  DTS
This will _released PB4 PB3 PA15_

ck-telecom, 2021-09-21:
but when SPI1 remaped in LL_GPIO_AF_EnableRemap_SPI1(), the LL library LL_GPIO_AF_EnableRemap_SPI1() API set CONFIG_SWJ_NO_REMAP status

FRASTM, 2021-09-22:
yes, 
So you need LL_GPIO_AF_EnableRemap_SPI1  to ('SPI1_REMAP = 1) remap SPI1 on 
SPI1_NSS  PA15,  SPI1_SCK  PB3, SPI1_MISO  PB4, SPI1_MOSI  PB5
and also LL_GPIO_AF_Remap_SWJ_NOJTAG() to set the SWJ_CFG[2:0] = 010: JTAG-DP Disabled and SW-DP Enabled which  frees pin PA15, PB3, PB4

FRASTM, 2021-09-22:
with this patch, trying to change only the SPI1_REMAP bit (not SWJ _CFG)  when remapping the SPI1
Changing the SWJ _CFG is done by the `gpio_stm32_afio_init()` function

ck-telecom, 2021-09-23:
sure, that's the point!

FRASTM, 2021-09-23:
@ck-telecom could you please test with drivers: pinmux stm32F1 remap spi1 without changing the SWJ_CFG #38741
including a change in the hal_stm32 https://github.com/zephyrproject-rtos/hal_stm32/pull/118

closed 2021-10-04 by cfriedt