zephyrproject-rtos/zephyr #38362: soc: ti cc13x2-cc26x2: PM standby + radio interaction regression

niflostancu, 2021-09-07:
**Description**
Hello, my project uses the TI CC13x2/CC26x2 platform and I have discovered a critical regression regarding power management after upgrading my Zephyr stack: the radio module won't work anymore after a PM Standby state wakeup  (actually, the entire program enters an infinite busy wait inside RFCDoorbellSendTo).

I have identified the root cause to be 65e4add74ecae7281299ffb0a90292e6536ccc41, because it eliminated the code which checks for disallowed the PM state (see Additional Context for explaination and potential solution).

**To Reproduce**
To reproduce, I used two `cc1352r1_launchxl`s and the echo example.
1. checkout https://github.com/niflostancu/zephyr/tree/cc13x2_radio_standby_bug (modified example to allow standby).
2. For the server board: `west build --pristine -b cc1352r1_launchxl samples/net/sockets/echo_server -- -DOVERLAY_CONFIG="overlay-802154-subg.conf" && west flash`
3. For the client board: `west build --pristine -b cc1352r1_launchxl samples/net/sockets/echo_client -- -DOVERLAY_CONFIG="overlay-802154-subg.conf" && west flash`
4. Use debugger on the client board, see it getting stuck on `RFCDoorbellSendTo`.

**Expected behavior**
The radio should work after waking up from standby.

**Impact**
This is a showstopper for anyone using the CC13x2/26x2 radio and power management.

**Logs and console output**
```
pwndbg> bt
#0  0x0000b546 in NOROM_RFCDoorbellSendTo (pOp=pOp@entry=67174401) at .../rfc.c:104
#1  0x0000c9b2 in RF_abortCmd (...) at .../RFCC26X2_multiMode.c:3818
#2  0x00014332 in RF_flushCmd (...) at .../C26X2_multiMode.c:5134
#3  0x00013ad2 in ieee802154_cc13xx_cc26xx_subg_stop (...) at .../ieee802154_cc13xx_cc26xx_subg.c:518
#4  0x0000a77e in ieee802154_cc13xx_cc26xx_subg_tx (...) at .../ieee802154_cc13xx_cc26xx_subg.c:371
```

**Environment:**
 - OS: Arch Linux
 - Zephyr SDK 0.13.0
 - Commit e1481fee4a

**Additional context**
The removed code (https://github.com/zephyrproject-rtos/zephyr/commit/65e4add74ecae7281299ffb0a90292e6536ccc41#diff-d848b34b5a2e740d33ff20ca9bc29306704d50e2798e931d178ac187e559d3cb) contains several state disallow sequences:
```c
if ((constraints & (1 <<
		PowerCC26XX_DISALLOW_STANDBY)) != 0) {
	disallowed = true;
}
```
The `DISALLOW_STANDBY` is set whenever the radio core is up (see [RFCC26X2_multiMode.c](https://github.com/zephyrproject-rtos/hal_ti/blob/master/simplelink/source/ti/drivers/rf/RFCC26X2_multiMode.c#L3031)), but `Power_sleep` doesn't check it to stop the standby procedure.
I think the solution would be to set a pm_constraint for  whenever a radio interface is up (I can come with a pull request).

The same happens with `PowerCC26XX_DISALLOW_IDLE` which, I see, is set when the XOSC_HF is pending to be configured.
Does zephyr do this? Someone should check if this case happens, too :D

niflostancu, 2021-09-07:
Actually, I see that the hal_ti's `Power_setConstraint` / `Power_releaseConstraint` is used through its driver lib and should be properly respected... I don't know which would be the best way to integrate it with Zephyr's PM constraint API in order to avoid all possible race conditions... maybe add a platform-specific constraints hook?

ceolin, 2021-09-08:
> Actually, I see that the hal_ti's `Power_setConstraint` / `Power_releaseConstraint` is used through its driver lib and should be properly respected... I don't know which would be the best way to [...]

Urghh, you are right about respect the platform constraint. Restore the policy would work and is the simplest thing to do, but works only in one direction - Constraints set in the driver.  Constraints set using Zephyr API are meaningless to the driver.
The hooks seem to much but may be the most flexible solution to interchangeable constraints

ceolin, 2021-09-09:
@niflostancu Unfortunately I don't have two compatible boards to reproduce exactly this test. Can you check if this branch  https://github.com/ceolin/zephyr/tree/issue/38362 works ?

niflostancu, 2021-09-09:
Hi, I tested your solution.

You reversed the logic of `pm_constraint_get`:
```c
ret = (constraints & (1 << PowerCC26XX_DISALLOW_IDLE)) != 0;
```
It needs to return true if the state is allowed, so DISALLOW needs to be UNSET.
If I make these changes, it works as desired:
```c
ret = (constraints & (1 << PowerCC26XX_DISALLOW_IDLE)) == 0;
```

ceolin, 2021-09-09:
> Hi, I tested your solution.
> 
> You reversed the logic of `pm_constraint_get`:
> 
> ```c
> ret = (constraints & (1 << PowerCC26XX_DISALLOW_IDLE)) != 0;
> ```
> 
> It needs to return true if the state is allowed, so DISALLOW needs to be UNSET.
> If I make these changes, it works as desired:
> 
> ```c
> ret = (constraints & (1 << PowerCC26XX_DISALLOW_IDLE)) == 0;
> ```
> 
> (repeat for STANDBY)

last time changes : / Cool. Let me fix it and move forward.

PR #38434 (2021-09-13): pm: cc13x2_cc26x2: Implement their own constraint
TI Hal has its own constraint API that is used by its drivers. These
constraints need to be correlated with Zephyr constraints to be
constraints set in the HAL be visible on Zephyr and vice-versa.

Fixes #38362
commit 351fd170 (2021-09-13): pm: cc13x2_cc26x2: Implement their own constraint

TI Hal has its own constraint API that is used by its drivers. These
constraints need to be correlated with Zephyr constraints to be
constraints set in the HAL be visible on Zephyr and vice-versa.

Fixes #38362
closed 2021-09-13 by cfriedt