zephyrproject-rtos/zephyr #38371: Port the Google ChromeOS USB-C Stack to Zephyr

sambhurst, 2021-09-07:
## Introduction

Add a USB-C subsystem to Zephyr that provides a mechanism for developers to create Type-C and Power Delivery applications.

**Requirements:**
- The USB-C subsystem is designed against the following specifications:
  - USB Type-C Port Controller Interface Specification, Revision 2.0, Version 1.2
  - USB Type-C Specification, Release 2
  - USB Power Delivery Specification, Revision 3.0, Version 2
- Sink, Source, and Dual Role Port Type-C state machines are implemented. Note: Other Type-C state machines are added as needed by particular applications.
- Minimum set of Power Delivery features are implemented. Note: Other features are added as needed by particular applications.
- Define tests for each implemented feature.   

**Out of scope:**
- Security related to authentication between port partners or a port and a cable plug.
- Firmware updates between port partners or a cable plug.
- Application specific functionality such as power supply design, charging batteries, hardware control, etc.
- Specific power delivery policies.
- USB PD Compliance.

**Background:**
From wikipedia:
USB-C (formally known as USB Type-C) is a 24-pin [USB](https://en.wikipedia.org/wiki/USB) connector system with a rotationally symmetrical connector.
 
The USB Type-C Specification 1.0 was published by the [USB Implementers Forum](https://en.wikipedia.org/wiki/USB_Implementers_Forum) (USB-IF) and was finalized in August 2014. It was developed at roughly the same time as the [USB 3.1](https://en.wikipedia.org/wiki/USB_3.1) specification.
 
A device with a Type-C connector does not necessarily implement USB, [USB Power Delivery](https://en.wikipedia.org/wiki/USB_Power_Delivery), or any [Alternate Mode](https://en.wikipedia.org/wiki/USB-C#Alternate_Mode): the Type-C connector is common to several technologies while mandating only a few of them.
 
[USB 3.2](https://en.wikipedia.org/wiki/USB_3.2), released in September 2017, replaces the USB 3.1 standard. It preserves existing USB 3.1 SuperSpeed and SuperSpeed+ data modes and introduces two new SuperSpeed+ transfer modes over the USB-C connector using two-lane operation, with data rates of 10 and 20 Gbit/s (1 and ~2.4 GB/s).
 
[USB4](https://en.wikipedia.org/wiki/USB4), released in 2019, is the first USB transfer protocol standard that is only available via USB-C
 
From Google Chrome OS:
The USB-C Power Delivery solution for Chromebooks is implemented as a layered, state-based design which tracks closely with the USB Type-C and USB PD specifications. The layers are three in total and referred to as the Type-C Layer (TC), Protocol Layer (PRL), and Policy Engine Layer (PE) and each layer consists of one or more state machines that are described in the specifications. For a more detailed description see [TCPMv2](https://chromium.googlesource.com/chromiumos/platform/ec/+/main/docs/usb-tcpmv2.md).

## Detailed RFC

The USB-C subsystem operates as an application agnostic framework that can be used to rapidly create Type-C and Power Delivery products. Said products are created through the use of canonical APIs exposed by the subsystem and the implementation of user defined callbacks.

Definitions used in this document:

- VCONN:  Fixed power provided by the Source or Sink that's used to power cables with electronics.
- VBUS: Adjustable power provided by the Source to power Sinks.

The subsystem supports the following Type-C features:
- Source only port: Provider of power over VBUS
- Sink only port: Consumer of power from VBUS
- Dual Role port (DRP): The port can be a Sink or Source
- Accessory mode: Reconfigures the device for a different mode of operation, such as Debug Accessory Mode or Audio Adapter Accessory Mode.
- Try Source: A DRP that prefers to be a Source on a sink connect detection.
- Try Sink: A DRP that prefers to be a Sink on a source connection detection.

The subsystem supports the following Power Delivery (PD) features:

- Chunking: Enables the transmission and reception of large messages by breaking the message up into chunks.
- GiveBack: Enables the Sink to respond to a GotoMin message by reducing its load to a Minimum Operating Current.
- VDM: Vendor Defined Message allows vendors to exchange information outside that defined by the PD specification. 
- DP: Display Port
- PRS: Power Role Swap
- DRS: Data Role Swap
- VS: VCONN Swap

### Upstreaming
This section describes how the subsystem is structured and segmented for easy upstreaming and testing. 

**Directory Structure**
The Zephyr USBC subsystem is structured as follows:

**zephyr/lib/smf/**
This directory contains a State Machine Framework ([SMF PR](https://github.com/zephyrproject-rtos/zephyr/pull/38088)). The state machine states have the option of enabling Entry, Run, and Exit actions, where the Entry action is executed once on state entry, the Run action is executed repeatedly, and the Exit action is executed once on state exit. States can also have multiple ancestor states.
The SMF is application agnostic but it's used as the foundation for the USB-C subsystem's state machines.

**zephyr/drivers/usbc/**
This directory contains all device drivers related to USBC operation. 

**zephyr/include/usbc/**
This directory contains the include file that should be added to any project that is using the subsystem.

**zephyr/include/drivers/usbc/**
This directory contains the include files that define the public interface to the USBC device drivers.

**zephyr/subsys/usbc/**
This directory contains the core of the USBC stack.

**zephyr/tests/subsys/usbc/**
This directory contains the tests for the USBC stack.

**zephyr/tests/lib/smf/**
This directory contains the tests for the State Machine Framework.

**Upstreaming Process**
To make the upstreaming process seamless, the USB-C subsystem is broken into Minimum Viable Products (MVPs) that executes on at least native_posix and perhaps a real device. Along with each MVP, a set of [twister](https://docs.zephyrproject.org/latest/guides/test/twister.html) and or [emulator](https://chromium.googlesource.com/chromiumos/platform/ec/+/refs/heads/main/zephyr/test/drivers/) tests are also upstreamed with optional tests that can be run on some device.

Although there are several ways to test the MVPs, it is encouraged to use the native_posix emulator. With the emulator, the MVPs can be tested in a way similar to real hardware. Also, code coverage can be monitored and improved with each additional emulator test. One drawback to this approach is that emulated hardware has to be written in addition to the MVP tests but this drawback will become less of a concern as more emulated hardware is added to the native_posix emulator test subsystem. Also, to help alleviate this issue, each MVP could be coded by two developers, one to write the code and the other to write the emulator hardware and tests.

**[Pull Request 1](https://github.com/zephyrproject-rtos/zephyr/pull/38088)** #
This PR upstreams the State Machine Framework, where the framework is integrated into Zephyr as a library.

**Pull Request 2**
The purpose of this PR is to introduce the mechanism for TCPC Device Drivers and a Sink state machine. To this end, a TCPC Device Driver for the stm32g071_disco board along with a Type-C Sink state machine is upstreamed. A sample application should be implemented to demonstrate how it's used and executed on the stm32g071_disco board.

**Pull Request 3**
This PR upstreams a Type-C Source state machine.

**Pull Request 4**
This PR upstreams a Type-C DRP state machine with TrySrc and TrySnk.

**Pull Request 5**
This PR upstreams the Google Honeybuns board. This board is used for testing the remaining Pull Requests and it uses the same SOC as the stm32g071_disco board.

**Pull Request 6**
This PR upstreams PRL and PE functionality to implement PRS. The PR is upstreamed as stacked commits, where the first commit implements the PRL functionality and the second commit implements the PE functionality.

**Pull Request 7**
This PR upstreams PE functionality for DRS.

**Pull Request 8**
This PR upstreams PE functionality for VS.

**Pull Request 9**
This PR upstreams PE functionality for Chunking.

**Pull Request 10**
This PR upstreams PE functionality for Giveback.

**Pull Request 11**
This PR upstreams PE functionality for VDM.

**Pull Request 12**
This PR upstreams PE functionality for DP.

**Pull Request 13**
This PR upstreams PE functionality to implement the Type-C accessory mode.

### Dependencies

None

### Concerns and Unresolved Questions

None

erwango, 2022-05-13:
@sambhurst Would you mind updating description so we can follow what is currently under review and what is already merged ?

laurenmurphyx64, 2022-10-18:
@sambhurst Hi, could you update this checklist to reflect what you're targeting for 3.3?

sambhurst, 2022-10-24:
> @sambhurst Hi, could you update this checklist to reflect what you're targeting for 3.3?
Currently, a Power Delivery Sink device can be created.

For the 3.3:
1) Update the **[DT USB Connector node](https://github.com/zephyrproject-rtos/zephyr/pull/45601#discussion_r972406784)**
2) Implement a Power Delivery Source

jgl-meta, 2023-05-02:
@keith-zephyr Do you have any context on this? Potentially could you break this down and add to the release? I assume this being a major effort it will not make it into 3.4?

keith-zephyr, 2023-05-02:
> @keith-zephyr Do you have any context on this? Potentially could you break this down and add to the release? I assume this being a major effort it will not make it into 3.4?

Yes, this is a major effort, with completion of all features targeted towards the end of 2023.

We should probably drop the release milestone on this RFC/design, and instead add release milestones for the smaller efforts identified as Pull Requests 1-13 above.  We can look to assign a release milestone on each PR as it gets created.

For the v3.4 release specifically, [Type-C source support](https://github.com/zephyrproject-rtos/zephyr/pull/52993) just merged, so the release notes should note that enhancement.

No other features (Pull Requests 4-13) are targeting the v3.4 release.

donusry-privoro, 2024-01-17:
Can we get an update to completed pull requests and ones being worked on?

davedesro, 2024-03-05:
Hey hey @keith-zephyr.   What's the latest with this feature? Can you let us know where we are with respect to the original PR plan above?  Thanks for getting USB PD into Zephyr!

te-johan, 2024-10-25:
FYI, i just created a [pull request](https://github.com/zephyrproject-rtos/zephyr/pull/80418) for some of the fixes i had to do go get the sink and source ports to work.

oliverhuahua, 2026-09-22:
Hello @sambhurst @keith-zephyr,

I have been looking at test coverage for the usb_c subsystem and opened #119878
with unit tests for `usbc_timer.c`, which as far as I can tell are the first
tests in the subsystem.

That is about as far as this can go without a TCPC to test against. `i2c_emul`,
`gpio_emul` and `adc_emul` are all in tree, so the stack could run on
`native_sim` if there were an emulated port controller; without one there is
nothing for a test of the TC, PE or PRL layers to talk to.

So I would like to make the stack testable. The state machines only reach the
TCPC through the driver API, so the likely first step is a mock TCPC driver and
tests for the TC layer against it. Register-level emulation of the TCPCI
interface could follow if the driver layer turns out to warrant it, since
`tcpci.c` already centralises what the compliant drivers share and one emulator
would then serve ps8xxx, rt1715, fusb307 and the others.

@keith-zephyr, you noted here that the ChromeOS EC TCPM is a good reference but
was written for cros-ec and does not map onto Zephyr's device model without
significant work. I take that as a reason to write something small against
Zephyr's own driver model and `emul` framework rather than to port anything,
which is what I have in mind.

Before going further: does this fit what you have planned for the subsystem? If
it does I will write it up as an RFC with the design details: what the mock
needs to model, what the test-facing API looks like, and where the line between
the mock and a fuller emulator should sit. Either is a new driver and would be
waiting for the v4.6 window in any case.