zephyrproject-rtos/zephyr #38376: Raw Socket Read Failure - zsock_recvfrom_ctx does not support Raw Socket

bob2oneil, 2021-09-07:
**Describe the bug**
I have a raw socket implementation on 2 virtual Ethernet interfaces using the Native-POSIX board target.  The raw socket associated with the standard Ethernet interface no longer works properly with the latest Zephyr build as of Sept 7th, but I have observed this issue in previous versions.  

What appears to happen is as shown in the callstack of the attached screenshot terminating in the zsock_recv_ctx() API for the failed raw socket read.  This API does not support reads for raw sockets, so it would appear that perhaps the kernel configuration is causing execution paths that are not expected?

The callstacks for the working and non-working raw socket interfaces shown in the following callstack excerpts between working and non-working implementations on different interfaces.
It is clear that for the working raw socket, the 1st 2 lines differ from the non-working, so the callstack is simply different for 2 raw sockets.  The POSIX read API as zsock_recv() goes to different flows for these two interfaces.

Debugging involved walking into the POSIX recv() API for the working raw socket interface versus the non-working one to produce the callstacks.

**To Reproduce**
Steps to reproduce the behavior:
1. west build
2. target Native-POSIX board
3. make

**Expected behavior**
Properly call stack routed behavior for raw socket on both virtual interfaces.

**Impact**
Showstopper relative to raw socket operation.

**Logs and console output**

Working callstack for raw socket:

```
zpacket_recvfrom_ctx(struct net_context * ctx, void * buf, size_t max_len, int flags, struct sockaddr * src_addr, socklen_t * addrlen) (/home/robocom/zephyrproject/zephyr/subsys/net/lib/sockets/sockets_packet.c:202)
packet_sock_recvfrom_vmeth(void * obj, void * buf, size_t max_len, int flags, struct sockaddr * src_addr, socklen_t * addrlen) (/home/robocom/zephyrproject/zephyr/subsys/net/lib/sockets/sockets_packet.c:347)
z_impl_zsock_recvfrom(int sock, void * buf, size_t max_len, int flags, struct sockaddr * src_addr, socklen_t * addrlen) (/home/robocom/zephyrproject/zephyr/subsys/net/lib/sockets/sockets.c:1251)
zsock_recvfrom(socklen_t * addrlen, sockaddr * src_addr, int flags, size_t max_len, void * buf, int sock) (/home/robocom/us/NetworkRed/software/redparrot/build/zephyr/include/generated/syscalls/socket.h:217)
zsock_recv(int flags, size_t max_len, void * buf, int sock) (/home/robocom/zephyrproject/zephyr/include/net/socket.h:404)
App::receivePacket(const char * ifName, const int sock, unsigned char * pkt, int pktSize, int & size) 
```

Failed callstack for non-working raw socket:

```
zsock_recvfrom_ctx(struct net_context * ctx, void * buf, size_t max_len, int flags, struct sockaddr * src_addr, socklen_t * addrlen) (/home/robocom/zephyrproject/zephyr/subsys/net/lib/sockets/sockets.c:1231)
sock_recvfrom_vmeth(void * obj, void * buf, size_t max_len, int flags, struct sockaddr * src_addr, socklen_t * addrlen) (/home/robocom/zephyrproject/zephyr/subsys/net/lib/sockets/sockets.c:2166)
z_impl_zsock_recvfrom(int sock, void * buf, size_t max_len, int flags, struct sockaddr * src_addr, socklen_t * addrlen) (/home/robocom/zephyrproject/zephyr/subsys/net/lib/sockets/sockets.c:1251)
zsock_recvfrom(socklen_t * addrlen, sockaddr * src_addr, int flags, size_t max_len, void * buf, int sock) (/home/robocom/us/NetworkRed/software/redparrot/build/zephyr/include/generated/syscalls/socket.h:217)
zsock_recv(int flags, size_t max_len, void * buf, int sock) (/home/robocom/zephyrproject/zephyr/include/net/socket.h:404)
App::receivePacket(const char * ifName, const int sock, unsigned char * pkt, int pktSize, int & size) (/home/robocom/us/NetworkRed/software/redparrot/src/app/app.cpp:608)
App::onWiredRawSocketReceive() 
```
[image]
[image]

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK 13.0, Native-POSIX target.)
 - Commit SHA or Version used is zephyr-v2.6.0-3072-g35d5443387a1

**Additional context**
Kernel configuration is as follows:

```
# C++ support
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
# CONFIG_STD_CPP2A=y
# CONFIG_CBPRINTF_FP_SUPPORT=y

# This setting in necessary for QEMU-RISCV, but will
# default to 'n' or external for NATIVE-POSIX
CONFIG_NEWLIB_LIBC=y

# General kernel options
CONFIG_INIT_STACKS=y
[... 157 lines cut ...]

# Gather network statistics
CONFIG_NET_STATISTICS_ETHERNET=y

# Name threads
CONFIG_THREAD_NAME=y

# Max number of IPv4 network interfaces in the system
# Will be a value of 1 for real radio
CONFIG_NET_IF_MAX_IPV4_COUNT=2

# How many Tx traffic classes to have for each network device
CONFIG_NET_TC_TX_COUNT=1

# How many network connections is supported (default is 4)
CONFIG_NET_MAX_CONN=8

# Total number of sockets to poll in select()
CONFIG_NET_SOCKETS_POLL_MAX=5

# Enable mostly-standards-compliant implementations of various POSIX (IEEE 1003.1) APIs.
# Necessary to support clock_gettime() for QEMU target
CONFIG_POSIX_API=y

```

bob2oneil, 2021-09-12:
Thanks for the response Chris, a test case does not yet exist, but I can create a minimal one to explore this raw socket to implementation mapping issue.  Would it be sufficient to attach say a ZIP compressed file to this issue for tracking, or is a Pull Request required?  As I understand GIT PR, it is typically submitted with proposed code to incorporate into the repo, for which I do not presently have a solution, only an observed problem.

cfriedt, 2021-09-12:
@bob2oneil - a PR would be ideal

bob2oneil, 2021-09-13:
Doesn't a PR require change code that solves the problem?  Currently, I have no idea as to why the 2 raw sockets, both associated with virtual Ethernet interfaces running on Native-POSIX, using the same create socket function, deviate when it comes to the mapping of the socket handle using POSIX networking APIs to the underlying implementation.  That is, I do not fully understand the Zephyr networking stack architecture sufficient to provide a solution, I do not have years of experience using it nor domain knowledge in that area.

rlubos, 2021-09-13:
> Currently, I have no idea as to why the 2 raw sockets

> That is, I do not fully understand the Zephyr networking stack architecture sufficient to provide a solution

It's not expected from you to provide a solution. A simple test case however (likely an extension of existing `tests/net/socket/af_packet` test suite) which allows to easily reproduce the issue would be a great starting point for us to find one. Ideally we could incorporate the test into the tree to avoid regressions in the future.

bob2oneil, 2021-09-14:
Thanks Robert, I will create a test case and attempt to use the PR mechanism to submit it.  Presumably I would submit an entire Zephyr snapshot where the additions made by myself would just be this test case?

cfriedt, 2021-09-14:
Also, @bob2oneil - if you use `#` followed by any PR or Issue number, you create an automatic github reference, which is handy, so in your PR, remember to do that (in the commit message or in the web interface) to create a reference back here.

carlescufi, 2021-09-29:
@bob2oneil please provide the information required.

bob2oneil, 2021-09-29:
I intend to submit a test case for this, have not had time to do that at the moment.  It is still an important issue for me to resolve.

carlescufi, 2021-11-17:
> I intend to submit a test case for this, have not had time to do that at the moment. It is still an important issue for me to resolve.

@bob2oneil do you intend to pursue the resolution of this issue? Since it cannot be reproduced, you should either provide a test case or close the issue. Thanks!

hideogump, 2021-11-17:
I have tested this out with the Zephyr release that dates back maybe a week, and I believe it is fixed.  I will close this out and will revisit it again should testing indicate that a problem remains.

bob2oneil, 2021-11-17:
I have tested this out with the Zephyr release that dates back maybe a week, and I believe it is fixed. I will close this out and will revisit it again should testing indicate that a problem remains.

closed 2021-11-17 by bob2oneil