zephyrproject-rtos/zephyr #38382: stm32 uart finishes Tx before going to PM

FRASTM, 2021-09-08:
**Describe the bug**
On the stm32, when the PM is enabled, the testsuite `tests/kernel/tickless/tickless_concept/` cannot ends properly when running with the twister script
Even if the test is running successfully,  the twister is not really ended, the  `PROJECT EXECUTION SUCCESSFUL` is missing.
The end_report() function of the subsys/testsuite/ztest/src/ztest.c cannot display the _result_ line.

When flashing the board and running the test direclty, the test is complete as expected : `PROJECT EXECUTION SUCCESSFUL`

**To Reproduce**
Steps to reproduce the behavior:
1.  _./scripts/twister --device-testing --device-serial /dev/ttyACM0 -p nucleo_wb55rg -T tests/kernel/tickless/tickless_concept/_
2. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
test FAILED

**Logs and console output**
```
Running test suite tickless_concept
===================================================================
START - test_tickless_sysclock
time 20, 230
time 240, 440
PASS - test_tickless_sysclock in 0.465 seconds
===================================================================
START - test_tickless_slice
elapsed slice 110, expected: <100, 110>
elapsed slice 100, expected: <100, 110>
elapsed slice 100, expected: <100, 110>
elapsed slice 100, expected: <100, 110>
PASS - test_tickless_slice in 0.690 seconds
===================================================================
Test suite tickless_concept succeeded
===================================================================
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain : 
 - Commit SHA b78d6740ec3510cc99b41c998544bcac747b4e40
- Zephyr version: v2.7.0-rc1-23-g89f4adf2ef77

**Additional context**

FRASTM, 2021-09-08:
First fix was found on the tessuite with  the https://github.com/zephyrproject-rtos/zephyr/pull/38299.
But it is not valid

FRASTM, 2021-09-08:
When analysing the log, it appears that the other PM_DEVICE_ACTION than PM_DEVICE_ACTION_SUSPEND was not handled correctly. 
Especially, the PM_DEVICE_ACTION_LOW_POWER should be handled like the PM_DEVICE_ACTION_SUSPEND.waiting for the Tx completion. Then the uart device is left unchanged, like for other PM_DEVICE_ACTION_   states.

Thats 's the reason why the twister  testsuite execution cannot end with the "PROJECT EXECUTION SUCCESSFUL" line : the uart DEVICE enters low power before line is actually transmitted on the console.

erwango, 2021-09-08:
Here is the back trace:
```
#0  uart_stm32_pm_control (dev=0x8005a5c <__device_dts_ord_53>, action=PM_DEVICE_ACTION_LOW_POWER)
    at /local/mcu/zephyrproject/zephyr/drivers/serial/uart_stm32.c:1457
#1  0x0800539c in pm_device_state_set (dev=dev@entry=0x8005a5c <__device_dts_ord_53>, 
    state=state@entry=PM_DEVICE_STATE_LOW_POWER) at /local/mcu/zephyrproject/zephyr/subsys/pm/device.c:145
#2  0x080012c0 in _pm_devices (state=state@entry=PM_DEVICE_STATE_LOW_POWER)
    at /local/mcu/zephyrproject/zephyr/subsys/pm/device.c:40
#3  0x0800541a in pm_low_power_devices () at /local/mcu/zephyrproject/zephyr/subsys/pm/device.c:64
#4  0x080011b6 in pm_system_suspend (ticks=<optimized out>) at /local/mcu/zephyrproject/zephyr/subsys/pm/power.c:225
#5  0x0800376c in pm_save_idle () at /local/mcu/zephyrproject/zephyr/kernel/idle.c:45
```
What is not clear to me is why we're receiving PM_DEVICE_ACTION_LOW_POWER as suspend request.
This case is not handled in none of the drivers today, so we're returning -ENOTSUP.
(And for some unclear reason, as a consequence, test case is failed when run through twister.)

@ceolin can you have a check?
Seems there is a disconnection between System PM device and device drivers implementing PM.

gmarull, 2021-09-08:
One problem I see here is that the device driver is not using the PM callback to implement actual power management, but rather busy waiting until transmission has finished. There should be a way to tell the system PM to not run in such cases.

ceolin, 2021-09-08:
@erwango sure thing. Checking it now.

ceolin, 2021-09-08:
The driver was handling `PM_DEVICE_STATE_LOW_POWER` until 495672ab623f7a4301a6bb7b5765ee48a7a13884

ceolin, 2021-09-08:
I will restore it, while the discussion around having a low power state or not continue.

gmarull, 2021-09-08:
The proper way to fix this problem, I think, is to use the `pm_constraint_set` API. The UART driver is using the PM callback to block until it finishes TX, meaning that in such device state, the system should never have triggered the transition to a low power state. The PM callback doesn't really do any power management (device is left ON and consuming the same power), so it's an API misuse. Adding back `PM_DEVICE_ACTION_LOW_POWER` (and `PM_DEVICE_ACTION_RESUME`) will just paper over a bug.

erwango, 2021-09-08:
> The proper way to fix this problem, I think, is to use the `pm_constraint_set` API. The UART driver is using the PM callback to block until it finishes TX, meaning that in such device state, the [...]

There are more drivers doing so. For instance:
https://github.com/zephyrproject-rtos/zephyr/blob/293af9e8237b28d9ae21037c3ccf48e77861b463/drivers/serial/uart_npcx.c#L446-L453

So I think problem is not in one driver only, we need to review all drivers that were impacted by 495672a , and that potentially introduced a desync between existing drivers implementation and new system PM expectations.

ceolin, 2021-09-08:
> There are more drivers doing so. For instance:
>
> https://github.com/zephyrproject-rtos/zephyr/blob/293af9e8237b28d9ae21037c3ccf48e77861b463/drivers/serial/uart_npcx.c#L446-L453
>
> So I think problem is [...]

Agree, we can go a revisit all drivers implementation and fix them one by one. But that pr deliberately removed the capability for handle low power from drivers. Breaking them. @gmarull  agree with you that this driver is probably not doing pm properly. But we need to be careful with the constraint since it will affect the whole system, and it may be a valid case when the system goes to stand by and this driver keeps active or in a low power state.

gmarull, 2021-09-08:
> Agree, we can go a revisit all drivers implementation and fix them one by one. But that pr deliberately removed the capability for handle low power from drivers. Breaking them.

That statement is not correct. https://github.com/zephyrproject-rtos/zephyr/commit/495672ab623f7a4301a6bb7b5765ee48a7a13884 just fixes PM callback usage. The fact that  https://github.com/zephyrproject-rtos/zephyr/commit/495672ab623f7a4301a6bb7b5765ee48a7a13884 uncovered some API misuses/bugs it's a different story. 

@gmarull agree with you that this driver is probably not doing pm properly. But we need to be careful with the constraint since it will affect the whole system, and it may be a valid case when the system goes to stand by and this driver keeps active or in a low power state.

In this case, the system won't go to low power until driver finishes transmission, so it's equivalent to block low power states until it's done with TX, same as TI driver does.

erwango, 2021-09-08:
> In this case, the system won't go to low power until driver finishes transmission, so it's equivalent to block low power states until it's done with TX, same as TI driver does.

I wonder if this really has to be done inside driver.
So there's only need to do that on TX (as we can't control RX anyway).
For IRQ transitions case , we can control IRQ flags, as done in TI driver.
But for TX not interrupt driven, how can driver properly release the constraint ?

ceolin, 2021-09-08:
> > > > The proper way to fix this problem, I think, is to use the `pm_constraint_set` API. The UART driver is using the PM callback to block until it finishes TX, meaning that in such device state, the system should never have triggered the transition to a low power state. The PM callback doesn't really do any power management (device is left ON and consuming the same power), so it's an API misuse.
> > > 
> > > 
> > > There are more drivers doing so. For instance:
> > > https://github.com/zephyrproject-rtos/zephyr/blob/293af9e8237b28d9ae21037c3ccf48e77861b463/drivers/serial/uart_npcx.c#L446-L453
> > > 
> > > So I think problem is not in one driver only, we need to review all drivers that were impacted by [495672a](https://github.com/zephyrproject-rtos/zephyr/commit/495672ab623f7a4301a6bb7b5765ee48a7a13884) , and that potentially introduced a desync between existing drivers implementation and new system PM expectations.
> > 
> > 
> > Agree, we can go a revisit all drivers implementation and fix them one by one. But that pr deliberately removed the capability for handle low power from drivers. Breaking them.
> 
> That statement is not correct. [495672a](https://github.com/zephyrproject-rtos/zephyr/commit/495672ab623f7a4301a6bb7b5765ee48a7a13884) just fixes PM callback usage. The fact that [495672a](https://github.com/zephyrproject-rtos/zephyr/commit/495672ab623f7a4301a6bb7b5765ee48a7a13884) uncovered some API misuses/bugs it's a different story.

```
-	if (state != PM_DEVICE_STATE_ACTIVE) {
+	switch (state) {
+	case PM_DEVICE_STATE_SUSPENDED:
```
That pr clearly changed the states that this driver was handling. Not that this particular way to handle state is not particular wrong, the driver specific implementation can be, but the way it is was handling states not. So, from my perspective the pr broke it.
> In this case, the system won't go to low power until driver finishes transmission, so it's equivalent to block low power states until it's done with TX, same as TI driver does.

ceolin, 2021-09-08:
> > > > The proper way to fix this problem, I think, is to use the `pm_constraint_set` API. The UART driver is using the PM callback to block until it finishes TX, meaning that in such device state, the system should never have triggered the transition to a low power state. The PM callback doesn't really do any power management (device is left ON and consuming the same power), so it's an API misuse.
> > > 
> > > 
> > > There are more drivers doing so. For instance:
> > > https://github.com/zephyrproject-rtos/zephyr/blob/293af9e8237b28d9ae21037c3ccf48e77861b463/drivers/serial/uart_npcx.c#L446-L453
> > > 
> > > So I think problem is not in one driver only, we need to review all drivers that were impacted by [495672a](https://github.com/zephyrproject-rtos/zephyr/commit/495672ab623f7a4301a6bb7b5765ee48a7a13884) , and that potentially introduced a desync between existing drivers implementation and new system PM expectations.
> > 
> > 
> > Agree, we can go a revisit all drivers implementation and fix them one by one. But that pr deliberately removed the capability for handle low power from drivers. Breaking them.
> 
> That statement is not correct. [495672a](https://github.com/zephyrproject-rtos/zephyr/commit/495672ab623f7a4301a6bb7b5765ee48a7a13884) just fixes PM callback usage. The fact that [495672a](https://github.com/zephyrproject-rtos/zephyr/commit/495672ab623f7a4301a6bb7b5765ee48a7a13884) uncovered some API misuses/bugs it's a different story.
> 
> > @gmarull agree with you that this driver is probably not doing pm properly. But we need to be careful with the constraint since it will affect the whole system, and it may be a valid case when the system goes to stand by and this driver keeps active or in a low power state.
> 
> In this case, the system won't go to low power until driver finishes transmission, so it's equivalent to block low power states until it's done with TX, same as TI driver does.

gmarull, 2021-09-08:
@erwango @FRASTM I've setup a nucleo-wb55rg board to reproduce the problem in the same exact conditions.

I see that the problem doesn't happen in I do this:

```patch
diff --git a/drivers/serial/uart_stm32.c b/drivers/serial/uart_stm32.c
index 487de2c070..b94603f85d 100644
--- a/drivers/serial/uart_stm32.c
+++ b/drivers/serial/uart_stm32.c
@@ -458,6 +458,9 @@ static void uart_stm32_poll_out(const struct device *dev,
 	LL_USART_ClearFlag_TC(UartInstance);
 
 	LL_USART_TransmitData8(UartInstance, (uint8_t)c);
+
+	while (LL_USART_IsActiveFlag_TC(UartInstance) == 0) {
+	}
 }
 
 static int uart_stm32_err_check(const struct device *dev)
```

So it looks like the `uart_stm32_poll_out` is not fully synchronous, it can return without the transfer being complete. I guess that under some conditions, this function gets called, transmission is not yet finished, but the idle thread is reached. I'm not entirely familiar with the STM32 UART driver, but I guess the solution should be to make sync path truly synchronous and use constraints for the ISR/DMA paths. With that in place, all the PM code could go away.

erwango, 2021-09-09:
> So it looks like the `uart_stm32_poll_out` is not fully synchronous

That was indeed my point.
For sure, the change you propose will solve the issue in non IRQ driven case. Though, I wonder about the conrtact with API and impact on users.
```
 * @brief Output a character in polled mode.
 *
 * This routine checks if the transmitter is empty.
 * When the transmitter is empty, it writes a character to the data
 * register.
```
To me, API description seems clear that this function is not meant to be synchronous.
If it was implicit that we should only exit once transmission is complete, then why bothering checking if the transmitter is empty ? Looking to other drivers implementation, there are only 2 drivers implementing a fully synchronous implementations (one of which being TI).

The problem this raises to me is that, because we changed the meaning of the PM API, we have also to change the uart API (and we also need to check impact on other peripheral API). So that's definitely not one faulty driver issue (I which it was, as it would be much easier to fix).
@dpcleung, what's your take on this ?

gmarull, 2021-09-09:
It would also introduce too much overhead if `LL_USART_IsActiveFlag_TC` is used for every byte, so it's definitely not an option. It is sufficient to way for TXE. In such case, for the polling path, there's probably no other way than blocking before entering LP mode. Still, the IRQ/DMA path should make use of constraints to avoid starting unnecessary system PM transitions.

gmarull, 2021-09-09:
> There are more drivers doing so. For instance:
> https://github.com/zephyrproject-rtos/zephyr/blob/293af9e8237b28d9ae21037c3ccf48e77861b463/drivers/serial/uart_npcx.c#L446-L453
>
> So I think problem is [...]

I've just checked and `npcx` driver was explicitly checking for the suspended state, so behavior is not changed in that case. I wonder how it ever worked, though, since they seem to use a `PM_STATE_SUSPEND_TO_IDLE` which triggers the "low power" state. After checking `npcx` driver, I see that they can solve the case by using constraints for all cases, since polling doesn't uses FIFO buffer. The STM32 case was a bit different, since it was doing `if (state != PM_DEVICE_STATE_ACTIVE) {`. This pattern is not safe is new states are introduced.

PR #38432 (2021-09-13): drivers: serial: stm32: use PM constraints to prevent suspension
In the current implementation, the STM32 UART driver required to enable
`CONFIG_PM_DEVICE` when `CONFIG_PM=y` to function properly. The main
reason is that in some situations, like in polling mode, transmissions
are not fully synchronous. That is, a byte is pushed to the _queue_ if
it is empty, and then the function returns without waiting for it to be
transmitted to the wire. This makes sense to make things like per-byte
transmission efficient. However, this introduces a problem: the system
may reach idle state, and so enter low [... cut]
closed 2021-09-13 by nashif