zephyrproject-rtos/zephyr #38398: DT_N_INST error for TMP116 sample

1hannah11, 2021-09-09:
TMP116 sensor sample is unable to compile due to what seems to be an issue finding the tmp116 device driver. I am currently trying to compile for the BBC microbit_V2. However I additionally have tried to compile for the Nucleo STM32F401RE board which is the tested board for the sample. 
Running on Ubuntu VB from Windows, compiling using west.

The error message received is:
error: 'DT_N_INST_0_ti_tmp116_P_label' undeclared (first use in this function)
  305 | #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))

galak, 2021-09-09:
Have you added the ti,tm116 to the devicetree for those boards?

1hannah11, 2021-09-10:
I had not, compiling correctly now. Thank you

closed 2021-09-10 by 1hannah11

akhil-bhat, 2024-02-01:
i am also getting similar type of error 'DT_N_INST_0_arm_v7m_nvic_P_arm_num_irq_priority_bits' undeclared (first use in this function)
 what should i add into devicetree and how to add?
 this is my board devicetree
 
/dts-v1/;
#include <arm/nxp/nxp_s32g274aabvuc_a53.dtsi>
#include "s32g274aabvuc_a53.dtsi"

/ {
	model = "NXP S32G374AABVUC on Cortex-A53 cores";     //changed
	compatible = "nxp,s32g274";              //changed

	chosen {
		zephyr,sram = &sram0;
		zephyr,canbus = &can0;
		zephyr,console = &uart0;
		zephyr,shell-uart = &uart0;
	};

	aliases {
		watchdog0 = &swt0;
	};
};

&mru0 {
	rx-channels = <1>;
	status = "okay";
};

&enetc_psi0 {
	mboxes = <&mru0 0>;
	mbox-names = "rx";
};