zephyrproject-rtos/zephyr #38401: Builds fail due to a proxy error by launchpadlibrarian

benediktibk, 2021-09-09:
Builds fail reproducible during the step "Run coding guidelines checks on patch series (PR)" -> "Install Packages":
https://github.com/zephyrproject-rtos/zephyr/pull/38029/checks?check_run_id=3553183772

The URI https://launchpadlibrarian.net/484859880/coccinelle_1.0.8~20.04npalix1_amd64.deb seems to be down for me as well, so not really a zephyr issue. But as it blocks the CI I thought it might be nice to document the issue at least.

kurddt, 2021-09-09:
Same here: https://github.com/zephyrproject-rtos/zephyr/pull/37539/checks?check_run_id=3553068436

benediktibk, 2021-09-09:
Uri seems to be working for me again, gave it another try with the CI.

benediktibk, 2021-09-09:
Looking good again:
https://github.com/zephyrproject-rtos/zephyr/pull/38029/checks?check_run_id=3555136964

closed 2021-09-09 by benediktibk