zephyrproject-rtos/zephyr #38403: Cleanup `No SOURCES given to Zephyr library` warnings

tejlmand, 2021-09-09:
The https://github.com/zephyrproject-rtos/zephyr/pull/37512 cleanup PR created several new Zephyr libraries instead of adding sources to libzephyr.a.

Unfortunately, this also results in some libraries not having sources added to them in certain Kconfig combinations, resulting in the following warning being printed:
```
  No SOURCES given to Zephyr library: drivers__bluetooth

  Excluding target from build.
```

**Expected behavior**
No warning being printed

**Impact**
Annoyance, and bad user experience.

Thalley, 2021-09-09:
Related to https://github.com/zephyrproject-rtos/zephyr/issues/37770

galak, 2021-09-09:
It seems like there are two possible approaches from a Kconfig point of view.

1. introduce some form of `<FOO>_HAS_DRIVER` that gets selected by all the driver Kconfig symbols and is used to build the dir/lib.  (See `SERIAL_HAS_DRIVER` for example)
2. rework Kconfig's so things are menuconfig for each driver class, etc).  Has ripple in how various boards/soc's have been selecting drivers

tejlmand, 2021-09-10:
>     1. introduce some form of `<FOO>_HAS_DRIVER` that gets selected by all the driver Kconfig symbols and is used to build the dir/lib.  (See `SERIAL_HAS_DRIVER` for example)

I have been considering such approach, but I feel it's like asking Kconfig to know about Zephyr CMake internals, as it starts top have symbols intended to help CMake.
Why should kconfig select a promptless symbol just to make things cleaner in CMake.

Of course, if such symbol is available for internal Kconfig purposes, like the `SERIAL_HAS_DRIVER`, then CMake can of course use such knowledge.

That being said, it could end-up being the cleanest solution.

> 2. rework Kconfig's so things are menuconfig for each driver class, etc). Has ripple in how various boards/soc's have been selecting drivers

yes, this is a broader restructure, but could end-up with a more uniform way of selecting drivers in Kconfig in general.

tejlmand, 2021-09-10:
> > 1. rework Kconfig's so things are menuconfig for each driver class, etc). Has ripple in how various boards/soc's have been selecting drivers
> 
> yes, this is a broader restructure, but could end-up with a more uniform way of selecting drivers in Kconfig in general.

Actually, majority of drivers are already using a `menuconfig`, only a fex exceptions like Bluetooth, interrupt, and Timer drivers.
[image]

So maybe we should use this opportunity to align those few to the rest.

tejlmand, 2021-09-28:
still a pending PR to be fully fixed: #38908

PR #38471 (2021-09-28): Cleanup CMake code to avoid creation of empty Zephyr libraries
Fixes: #38403
Fixes #38622

This is a series of commits to followup on #37512 by ensuring empty Zephyr libraries are not created.

With this series of commits, then several occurrences of:
```
No SOURCES given to Zephyr library: <foobar>
    
Excluding target from build.
```

has now been fixed.

---
This change is [[image]](https://reviewable.io/reviews/zephyrproject-rtos/zephyr/38471)
commit 6ebe0b84 (2021-09-28): drivers: create BT_DRIVERS Kconfig entry

Fixes: #38403

Changing Bluetooth drivers from being a menu into a menuconfig.
This aligns the Bluetooth driver configuration with other driver
configurations as well as provides a setting which identifies if
Bluetooth drivers has been enable.

This further helps to avoid empty Zephyr libraries for bluetooth
samples.
commit 72a8520a (2021-09-28): drivers: console: remove unused CONSOLE selection

Fixes: #38403

Removing unneeded `CONFIG_CONSOLES=y` occurrences where no console
driver is selected.

Such selection results in an empty drivers__console zephyr library,
which again results in the following warning message.

> No SOURCES given to Zephyr library: drivers__console
>
> Excluding target from build.
commit 8e640385 (2021-09-28): drivers: net: adding NET_DRIVERS menuconfig

Fixes: #38403

Adding NET_DRIVERS menuconfig so that network drivers are grouped
together in its own menu entry under drivers, similar to most other
drivers.

This further has the advantages that `CONFIG_NET_DRIVERS` can be used
for testing to determine if network drivers has been selected.

This changed revealed a dependency loop where both `select` (for SLIP)
and `depends` (for PPP) which both depends on NET_DRIVERS` where in use
in the dependency tree for Qemu networking, especially NET_SLIP_TAP.

This is handled by defaulting `NET_DRIVERS` to `y [... cut]
closed 2021-09-28 by carlescufi