zephyrproject-rtos/zephyr #38414: Build of http client fails if CONFIG_POSIX_API=y

benediktibk, 2021-09-09:
**Describe the bug**
It is not possible to use CONFIG_POSIX_API together with the http client library.

**To Reproduce**
Steps to reproduce the behavior:
1. Activate for instance for samples/net/sockets/http_client CONFIG_POSIX_API
2. Build the example for a non-posix board, e.g.: west build -p auto -d build samples/net/sockets/http_client/ --board=stm32h735g_disco
3. zephyr/samples/net/sockets/http_client/src/main.c:110: undefined reference to `send'

**Expected behavior**
It should be possible to use the http client library together with CONFIG_POSIX_API

**Impact**
Hinders me from using the already existing http-library in zephyr as I need CONFIG_POSIX_API for a different reason.

benediktibk, 2021-09-09:
PR: #38415

cfriedt, 2021-09-10:
@rlubos - could you please assign a priority to this?

closed 2021-09-10 by cfriedt