zephyrproject-rtos/zephyr #38417: Add support for WeAct-F401CC board

pathnirvana, 2021-09-09:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
Weact f401cc board is the most common, most available, lowest cost and and hence most easily accessible stmf4 board. Yet zephyr does not support this board. I tried compiling and running zephyr samples on this board by modifying the f401ce (not available anywhere to purchase) board definition. I could kind of get them working but it is giving some errors when uploading via stlink

**Describe why you are asking for this support?**
I am new to embedded programming and do not understand internals of rtos nor device tree format

If this is a new board or SoC, please state whether you are willing to maintain the Zephyr support for it if it is included in the main tree

**Additional context**
The riot os support information is listed here https://doc.riot-os.org/group__boards__weact-f401cc.html

erwango, 2021-09-10:
@pathnirvana Zephyr project is open to contributions and current board support reflects zephyr contributors (users, developpers) work. Probability that this request gets implemented in short/medium term delay is quite small.
From what I can see stm32f401cc soc is already ported. So remaining work is to port the board description.
Maybe the best idea is to ask support in discord, reporting the issues you're facing.

pathnirvana, 2021-09-10:
Many thanks @erwango for the response and many thanks to the contributors who have done so much for the community.
For now I have got it working by making a copy of the f401ce board definition and making few modifications.

closed 2021-09-10 by pathnirvana