zephyrproject-rtos/zephyr #38427: mimxrt1050_evk and mimxrt1020_evk boards fail to boot some sample applications

ainguraXmarquiegui, 2021-09-09:
**Describe the bug**
The last codebase when I checkout the *main* branch from zephyr no longer boots some sample applications like  subsys/mgmt/mcumgr/smp_svr or net/gptp. If I compile the hello_world sample application it works on both platforms. 

All three sample applications boot if I compile the same code corresponding to the latest commit of the *master* branch.

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyrproject/zephyr
2. git fetch && git checkout main && git pull
3. west update
4. west -v build -p -b mimxrt1050_evk samples/net/gptp
5. sudo west flash
6. **system doesn't boot**

Steps to verify that hello_world boots:

7. west -v build -p -b mimxrt1050_evk samples/hello_world/
8. sudo west flash
9. **You can see hello world on your console**

Steps to verify that it used to work as expected:

10. git checkout master && git pull
11. west  update
12. west -v build -p -b mimxrt1050_evk samples/net/gptp
13. sudo west flash
14. **You can see the gptp application boot on your console**

**Expected behavior**
All sample applications should boot and work as expected.

**Impact**
This is a showstopper. Not being able to boot the OS on the latest stable version on some hardware platforms is highly undesirable.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: zephyr-sdk-0.13.0 + cmake version 3.20.2
 - Commit SHA or Version used: 40f4091d44cf7b435d4bcbf0eac70d1035df0d08 (bug happens here)
 - Commit SHA to verify that it used to work OK: bfd271932228f8ce33b68b82ffee5ee3b2386a17 (this works ok)

ainguraXmarquiegui, 2021-09-10:
@MaureenHelm , reverting commit f0b3146ff5538e5381318a309fec292d3fbb1f3f makes code point 40f4091d44cf7b435d4bcbf0eac70d1035df0d08 bootable again on the mimxrt1020_evk

Someone should review this.

MaureenHelm, 2021-09-10:
Thanks for the report. This is a duplicate of #38186

closed 2021-09-10 by MaureenHelm