zephyrproject-rtos/zephyr #38428: http_client response callback always reports final_data == HTTP_DATA_FINAL

benediktibk, 2021-09-09:
**Describe the bug**
The response callback triggered by http_client_req reported always HTTP_DATA_FINAL, although more data was left and also more calls to the callback function occurred.

**To Reproduce**
Steps to reproduce the behavior:
1. setup an instance of struct http_request with a buffer size of 256 bytees
2. execute http_client_req which will result in a response of 784 bytes
3. -> multiple response callbacks with HTTP_DATA_FINAL, actually every single one

**Expected behavior**
The flag final_data should be set to HTTP_DATA_FINAL only for the last call to the callback function.

**Impact**
It is more an annoyance, as the data_len field can be checked for 0. Which seems to be a good indicator for the last packet chunk.

rlubos, 2021-09-13:
@jukkar Do you recall what data does the `http_response_cb_t` is expected to receive? On one hand, it's called only within `on_body`/`on_message_complete` callbacks, but it appears to provide headers as well on the first call (which IMO makes sense to provide the full message, for particular fields application can register specific callbacks in `http_parser_settings` ). So it's not clear to me, whether it's expected to get body content only, or the entire response.

I'm asking, because if it's expected to provide the entire message, it'd be easier to call the response cb directly from the `http_wait_data` instead of parser callbacks. It'd also allow to prevent the dummy call at the end with `data_len` set to 0, currently it's called after the entire body is received (so we get the initial call with the actual data, and then the dummy call w/o any data).

jukkar, 2021-09-14:
> Do you recall what data does the `http_response_cb_t` is expected to receive? On one hand, it's called only within `on_body`/`on_message_complete` callbacks, but it appears to provide headers as [...]

I do not remember that but after looking the code, the `on_body()` calls `http_should_keep_alive()` from the http_parser library which does some checks and returns true in http response when http status is either 100, 204, 304 or if chuncked. See `http_should_keep_alive()` and `http_message_needs_eof()` functions in http_parser.c https://github.com/zephyrproject-rtos/zephyr/blob/50ff77fa910dcad29167b389a6c7c67ade8f6f17/subsys/net/lib/http/http_parser.c#L2284

> I'm asking, because if it's expected to provide the entire message, it'd be easier to call the response cb directly from the `http_wait_data` instead of parser callbacks. It'd also allow to prevent [...]

I would assume that the application needs to get the "sub" responses, especially if chunked encoding is used.  Anyway, what is described in the issue description sounds like a bug, as for some reason the `HTTP_DATA_FINAL` is returned even if there is more data to come.

rlubos, 2021-09-15:
I have submitted the fix proposal in https://github.com/zephyrproject-rtos/zephyr/pull/38553.

After investigatin the HTTP parser code a bit, I've came to conclusion that it's better to rely on the `on_message_complete()` callback do determine whether to report `HTTP_DATA_FINAL` or not. I've tested the solution with a local HTTP server and a regular repsonse (including Content Length) and some other server I've found online, that replied with chunked response, works fine in both cases. 

Additionally, moving the response callback reporting into `http_wait_data ()` allows to report the entire message correctly even for smaller buffers provided - previous solution reported rubbish if the buffer length was smaller than the total headers size. Current one works fine even for very small buffers (1 byte).

PR #38553 (2021-09-17): net: http: Fix HTTP_DATA_FINAL notification
HTTP_DATA_FINAL was incorrectly notified in case Content Length field
was present in the HTTP respone - in such case it was set for every
response fragment, not only the last one.

Fix this by relying on `message_complete` flag instead of
`http_should_keep_alive()` function to determine whether to notify
HTTP_DATA_FINAL or not. As the HTTP parser calls the
`on_message_complete()` callback in either case (response is chunked or
not), this seems to be a more reasonable apporach to determine whether
the fragment is final or not.

Additinally, instead of [... cut]
closed 2021-09-17 by nashif