zephyrproject-rtos/zephyr #38437: stm32: g0b1re: Serial UART timing issue after MCU entered deep sleep

ycsin, 2021-09-10:
**Describe the bug**
The log starts to print very slowly and wrong string after the MCU entered deep sleep, Im guessing that it is related to some uart timing issue.

**To Reproduce**
1. Build anything that logs something through serial UART, then enter LPM on Nucleo G0B1
4. See error

**Expected behavior**
The serial print shouldn't be affected by LPM.

**Impact**
LOG isn't working correctly after LPM.

**Logs and console output**
```
[00:00:46.981,000] <inf> modem_gsm: RSSI: -63
[00:00:47.594,000] <inf> telemetry_mqtt: MQTT Keepalive
[00:00:50.095,000] <inf> telemetry_mqtt: MQTT Keepalive
[00:00:50.606,000] <inf> telemetry_mqtt: MQTT client disconnected 0
<-------- entered LPM here -------->
øJ[00:00:51.35,000] <wrn> hwmanger: Alloe to sleep[m
[1;2mgmoc:~ $ [mÄø[00:051.5,000] <b> hw_mana
```

**Environment (please complete the following information):**
 - Zephyr v2.7 RC1

**Additional context**
Tipped by @erwango that it could be related to this commit introduced from a4f6a3a85afa7eac45b339266ccd04ce49e7e9f4 this PR #36759, and is seems to be the case.

After I edited `uart_stm32.c` like so:
```c
#ifdef CONFIG_PM_DEVICE
static int uart_stm32_pm_control(const struct device *dev,
				 enum pm_device_action action)
{
	USART_TypeDef *UartInstance = UART_STRUCT(dev);

	/* setting a low power mode */
	switch (action) {
	case PM_DEVICE_ACTION_RESUME:
		break;
	case PM_DEVICE_ACTION_SUSPEND:
		__fallthrough;
	case PM_DEVICE_ACTION_FORCE_SUSPEND:
		__fallthrough;
	case PM_DEVICE_ACTION_LOW_POWER:
#ifdef USART_ISR_BUSY
		/* Make sure that no USART transfer is on-going */
		while (LL_USART_IsActiveFlag_BUSY(UartInstance) == 1) {
		}
#endif
		while (LL_USART_IsActiveFlag_TC(UartInstance) == 0) {
		}
#ifdef USART_ISR_REACK
		/* Make sure that USART is ready for reception */
		while (LL_USART_IsActiveFlag_REACK(UartInstance) == 0) {
		}
#endif
		/* Clear OVERRUN flag */
		LL_USART_ClearFlag_ORE(UartInstance);
		/* Leave UartInstance unchanged */
		break;
	default:
		return -ENOTSUP;
	}

	/* UartInstance returning to active mode has nothing special to do */
	return 0;
}
#endif /* CONFIG_PM_DEVICE */
```

It is working correctly again.

```
[00:00:46.975,000] <inf> modem_gsm: RSSI: -61
[00:00:47.578,000] <inf> telemetry_mqtt: MQTT Keepalive
[00:00:50.079,000] <inf> telemetry_mqtt: MQTT Keepalive
[00:00:50.590,000] <inf> telemetry_mqtt: MQTT client disconnected 0
[00:00:51.341,000] <wrn> hw_manager: Allowed to sleep
[00:00:51.341,000] <dbg> hw_manager.hw_man_release: Released HW_MAN_DEV_SOC
[00:00:51.341,000] <wrn> hw_manager: Stopped modem
[00:00:51.341,000] <dbg> hw_manager.hw_man_release: Released HW_MAN_DEV_MODEM
[00:00:51.341,000] <inf> telemetry_mqtt: Stop OK
gmoc:~ $
```

ycsin, 2021-09-10:
I don't know how the PM works and what `pm_device_action` exactly represents, what I know from my observation is:

This is required
```c
	case PM_DEVICE_ACTION_RESUME:
		break;
```

Some or both of these are required
```c
	case PM_DEVICE_ACTION_SUSPEND:
		__fallthrough;
	case PM_DEVICE_ACTION_FORCE_SUSPEND:
		__fallthrough;
```

ycsin, 2021-09-10:
@erwango @gmarull

ceolin, 2021-09-10:
This pr https://github.com/zephyrproject-rtos/zephyr/pull/38432 much likely fix this problem.

erwango, 2021-09-15:
@ycsin Would you mind providing here all the info you shared in #38521 ?
Also provide details on your specific implementation.

ycsin, 2021-09-15:
> @ycsin Would you mind providing here all the info you shared in #38521 ?
> Also provide details on your specific implementation.

My board is a G0B1RE + Quectel EC21 modem combo.

I have this in my main to test the stability of `gsm_ppp` modem

```c
	hw_man_init();

	while (true)
	{
		printk("\r\n\r\n%d: Turn on modem \r\n\r\n", i);

		hw_man_request(HW_MAN_DEV_MODEM);

		k_sleep(K_SECONDS(40));

		printk("\r\n\r\n%d: Turn off modem \r\n\r\n", i);

		hw_man_release(HW_MAN_DEV_MODEM);

		k_sleep(K_SECONDS(20));

		i++;
	}
```

`hw_man` is the hardware manager I mentioned, basically a wrapper for Zephyr's `onoff_manager`. As I've said this mainly solves the hardware dependency problem where the modem sends unsolicited data to the MCU through uart and current stm32's uart driver lacks the ability to wake the MCU to process these data, so I have to keep the MCU awake while the modem is on.

Snippet of `hw_man.c`:

```c
int hw_man_init(void)
{
	int i;

	if (ctx.initialized)
	{
		return -EALREADY;
	}

	for (i = 0; i < HW_MAN_DEV_MAX; i++)
	{
		onoff_manager_init(&ctx.srv[i], &ctx.transitions[i]);
	}

	ctx.initialized = true;

	return 0;
}

int hw_man_request(enum hw_man_dev dev)
{
	int					ret;
	struct onoff_client cli;

	if ((dev < 0) || (dev >= HW_MAN_DEV_MAX))
	{
		LOG_ERR("Dev %d doesn't exist", dev);
		return -EINVAL;
	}

	if (!ctx.initialized)
	{
		LOG_ERR("HW manager not initialized yet");
		return -EACCES;
	}

	sys_notify_init_spinwait(&cli.notify);
	ret = onoff_request(&ctx.srv[dev], &cli);

	if (ret < 0)
	{
		LOG_ERR("onoff_request returned %d", ret);
	}

	LOG_DBG("Requested %s", hw_man_dev[dev].hw_dev_str);

	return 0;
}

int hw_man_release(enum hw_man_dev dev)
{
	int ret;

	if ((dev < 0) || (dev >= HW_MAN_DEV_MAX))
	{
		LOG_ERR("Dev %d doesn't exist", dev);
		return -EINVAL;
	}

	if (!ctx.initialized)
	{
		LOG_ERR("HW manager not initialized yet");
		return -EACCES;
	}

	ret = onoff_release(&ctx.srv[dev]);

	if (ret < 0)
	{
		LOG_ERR("onoff_release %s returned %d",
				hw_man_dev[dev].hw_dev_str,
				ret);
	}
	LOG_DBG("Released %s", hw_man_dev[dev].hw_dev_str);

	return 0;
}
```

Snippet of the modem pluggin:

```c
/**
 * @brief Transition-to-start function for modem
 */
void hw_man_modem_start(struct onoff_manager *mgr, onoff_notify_fn notify)
{
	const struct device *dev;

	LOG_INF("Starting modem");

#if CONFIG_MODEM_GSM_PPP
	dev = DEVICE_DT_GET(DT_INST(0, zephyr_gsm_ppp));
	gsm_ppp_start(dev);
#endif /* CONFIG_MODEM_GSM_PPP */

	hw_man_request(HW_MAN_DEV_SOC);

	LOG_WRN("Started modem");
	notify(mgr, 0);
	return;
}

/**
 * @brief Transition-to-stop function for modem
 */
void hw_man_modem_stop(struct onoff_manager *mgr, onoff_notify_fn notify)
{
	const struct device *dev;

#if CONFIG_MODEM_GSM_PPP
	dev = DEVICE_DT_GET(DT_INST(0, zephyr_gsm_ppp));
	gsm_ppp_stop(dev);
#endif /* CONFIG_MODEM_GSM_PPP */

	hw_man_release(HW_MAN_DEV_SOC);

	LOG_WRN("Stopped modem");
	notify(mgr, 0);
	return;
}
```

Snippet of the SOC plugin:

```c
/**
 * @brief Prevents the Zephyr from entering SUSPEND_TO_IDLE state, equivalent
 * to STM32's STOP mode.
 */
void hw_man_soc_start(struct onoff_manager *mgr, onoff_notify_fn notify)
{
	pm_constraint_set(PM_STATE_SUSPEND_TO_IDLE);
	if (!pm_constraint_get(PM_STATE_SUSPEND_TO_IDLE))
	{
		LOG_WRN("Disallowed from sleep");
		notify(mgr, 0);
	}
	else
	{
		LOG_ERR("Unable to wake SOC!");
		notify(mgr, -EFAULT);
	}
	return;
}

/**
 * @brief Allow the Zephyr to enter SUSPEND_TO_IDLE state, equivalent
 * to STM32's STOP mode.
 */
void hw_man_soc_stop(struct onoff_manager *mgr, onoff_notify_fn notify)
{
	pm_constraint_release(PM_STATE_SUSPEND_TO_IDLE);
	if (pm_constraint_get(PM_STATE_SUSPEND_TO_IDLE))
	{
		LOG_WRN("Allowed to sleep");
		notify(mgr, 0);
	}
	else
	{
		LOG_ERR("Unable to sleep");
		notify(mgr, -EFAULT);
	}
	return;
}
```

ycsin, 2021-09-15:
> @ycsin Would you mind providing here all the info you shared in #38521 ?
> Also provide details on your specific implementation.

#38552

erwango, 2021-09-15:
Closed in favor of #38552

PR #38432 (2021-09-13): drivers: serial: stm32: use PM constraints to prevent suspension
In the current implementation, the STM32 UART driver required to enable
`CONFIG_PM_DEVICE` when `CONFIG_PM=y` to function properly. The main
reason is that in some situations, like in polling mode, transmissions
are not fully synchronous. That is, a byte is pushed to the _queue_ if
it is empty, and then the function returns without waiting for it to be
transmitted to the wire. This makes sense to make things like per-byte
transmission efficient. However, this introduces a problem: the system
may reach idle state, and so enter low [... cut]
closed 2021-09-15 by erwango