zephyrproject-rtos/zephyr #38438: test-ci: test_flash_map:twr_ke18f: test failure

hakehuang, 2021-09-10:
**Describe the bug**

 Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/storage/flash_map/src/main.c:90: test_flash_area_get_sectors: (rc == 0 is false)

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=twr_ke18f
3. make
4. See error

**Expected behavior**
test assert

**Impact**
test failure

**Logs and console output**

```

Finished #31138834 storage.flash_map.mpu
Overview
Retries
History
platform: twrke18f
compiler: armgcc , release
runtimeout: 720 (s)
Image: zephyr.elf
project:
revision: None
comment:
RAW
RAW
RAW
RAW
======== Target.main.COM31 Output ========
*** Booting Zephyr OS build v2.7.0-rc1-20-g538968a77328  ***
Running test suite test_flash_map
===================================================================
START - test_flash_area_erased_val
 PASS - test_flash_area_erased_val in 0.1 seconds
===================================================================
START - test_flash_area_get_sectors

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/storage/flash_map/src/main.c:90: test_flash_area_get_sectors: (rc == 0 is false)
area not erased
 FAIL - test_flash_area_get_sectors in 0.672 seconds
===================================================================
START - test_flash_area_check_int_sha256
 PASS - test_flash_area_check_int_sha256 in 0.205 seconds
===================================================================
Test suite test_flash_map failed.
===================================================================
PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used:  v2.7.0-rc1-20-g538968a77328

hakehuang, 2021-09-10:
@MaureenHelm  @dleach02 @henrikbrixandersen 

looks like the flash setting for twr_ke18f need update for testing.

henrikbrixandersen, 2021-09-14:
I am unable to reproduce this issue here:

```
*** Booting Zephyr OS build v2.7.0-rc1-100-g50ff77fa910d  ***
Running test suite test_flash_map
===================================================================
START - test_flash_area_erased_val
 PASS - test_flash_area_erased_val in 0.1 seconds
===================================================================
START - test_flash_area_get_sectors
 PASS - test_flash_area_get_sectors in 0.638 seconds
===================================================================
START - test_flash_area_check_int_sha256
 PASS - test_flash_area_check_int_sha256 in 0.201 seconds
===================================================================
Test suite test_flash_map succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

```

dleach02, 2021-09-14:
@hakehuang can you retest? I'm also unable to reproduce this.

hakehuang, 2021-09-15:
it now passes on v2.7.0-rc1-94-gf83dc928f6&

closed 2021-09-15 by hakehuang