zephyrproject-rtos/zephyr #38446: intel_adsp_cavs15: fail to run mutex testcases on ADSP

LixinGuoX, 2021-09-10:
**Describe the bug**
run tests/kernel/mutex/mutex_api failed on ADSP, can not generate a log correctly.

**To Reproduce**
Steps to reproduce the behavior:

1. twister -p intel_adsp_cavs15 --device-testing -T tests/kernel/mutex/mutex_api/ --west- 
flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/flash.sh,/home/ztest/work/zephyrproject/modules/audio/sof/keys/otc_private_key.pem,/home/ztest/work/rimage/config,/home/ztest/work/rimage/build/rimage" --device-serial-pty="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py,--no-history"

**Expected behavior**
===
=== [ADSP Device Reset?]
===
Running test suite mutex_api
===================================================================
START - test_mutex_lock_unlock
PASS - test_mutex_lock_unlock in 0.1 seconds
===================================================================
START - test_mutex_reent_lock_forever
access resource from main thread
access resource from main thread
PASS - test_mutex_reent_lock_forever in 1.1 seconds
===================================================================
START - test_mutex_reent_lock_no_wait
access resource from main thread
bypass locked resource from spawn thread
access resource from main thread
bypass locked resource from spawn thread
PASS - test_mutex_reent_lock_no_wait in 1.1 seconds
===================================================================
START - test_mutex_reent_lock_timeout_fail
access resource from main thread
bypass locked resource from spawn thread
access resource from main thread
bypass locked resource from spawn thread
PASS - test_mutex_reent_lock_timeout_fail in 2.1 seconds
===================================================================
START - test_mutex_reent_lock_timeout_pass
access resource from main thread
access resource from spawn thread
access resource from main thread
bypass locked resource from spawn thread
PASS - test_mutex_reent_lock_timeout_pass in 2.1 seconds
===================================================================
START - test_mutex_recursive
PASS - test_mutex_recursive in 0.2 seconds
===================================================================
START - test_mutex_priority_inheritance
PASS - test_mutex_priority_inheritance in 6.1 seconds
===================================================================
Test suite mutex_api succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

**Impact**
block daily testing

**Logs and console output**
===
=== [ADSP Device Reset?]
===

**Environment (please complete the following information):**
 - OS:  Fedora28
 - Toolchain: Zephyr-sdk-0.13.0
 - Commit ID: 1dc8e8c5a03e7cc5e47781ba1523c5e62a451cd0

nashif, 2021-09-22:
Are you able to build and run the tests manually and without twister?

KangJianX, 2021-09-23:
@nashif We could build successfully, but can not get output after flashing. It should rely on a script to get output, and the root cause is loop time too short to get output.

KangJianX, 2021-09-24:
This issue is caused after changed the loop time in this commit ID 1dc8e8c

yerabolu, 2021-10-14:
@LixinGuoX : Could you further triage this issue on the setup and figure out if there is something else other than the timing issue?