zephyrproject-rtos/zephyr #38479: "west flash" command exiting with error

waqarhumayun, 2021-09-11:
**Describe the bug**
west flash command fails with error 

TypeError: expected str, bytes or os.PathLike object, not NoneType

**To Reproduce**
Steps to reproduce the behavior:
Follow getting started guide and then section for flashing the built image on the board.

**Logs and console output**

I am facing following error while running west flash command.

Traceback (most recent call last):
  File "c:\users\waqarh\appdata\local\programs\python\python36-32\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\waqarh\appdata\local\programs\python\python36-32\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Program Files (x86)\Python\Python36-32\Scripts\west.exe\__main__.py", line 9, in <module>
  File "c:\users\waqarh\appdata\local\programs\python\python36-32\lib\site-packages\west\app\main.py", line 779, in main
    app.run(argv or sys.argv[1:])
  File "c:\users\waqarh\appdata\local\programs\python\python36-32\lib\site-packages\west\app\main.py", line 106, in run
    self.run_command(argv)
  File "c:\users\waqarh\appdata\local\programs\python\python36-32\lib\site-packages\west\app\main.py", line 338, in run_command
    self.run_extension(args.command, argv)
  File "c:\users\waqarh\appdata\local\programs\python\python36-32\lib\site-packages\west\app\main.py", line 408, in run_extension
    command.run(args, unknown, self.topdir, manifest=self.manifest)
  File "c:\users\waqarh\appdata\local\programs\python\python36-32\lib\site-packages\west\commands.py", line 129, in run
    self.do_run(args, unknown)
  File "C:\Personal\zephyr_rtos\zephyrproject\zephyr\scripts\west_commands\flash.py", line 29, in do_run
    do_run_common(self, my_args, runner_args)
  File "C:\Personal\zephyr_rtos\zephyrproject\zephyr\scripts/west_commands\run_common.py", line 222, in do_run_common
    runner.run(command_name)
  File "C:\Personal\zephyr_rtos\zephyrproject\zephyr\scripts/west_commands\runners\core.py", line 482, in run
    self.do_run(command, **kwargs)
  File "C:\Personal\zephyr_rtos\zephyrproject\zephyr\scripts/west_commands\runners\openocd.py", line 114, in do_run
    self.require(self.openocd_cmd[0])
  File "C:\Personal\zephyr_rtos\zephyrproject\zephyr\scripts/west_commands\runners\core.py", line 502, in require
    if shutil.which(program) is None:
  File "c:\users\waqarh\appdata\local\programs\python\python36-32\lib\shutil.py", line 1113, in which
    if os.path.dirname(cmd):
  File "c:\users\waqarh\appdata\local\programs\python\python36-32\lib\ntpath.py", line 243, in dirname
    return split(p)[0]
  File "c:\users\waqarh\appdata\local\programs\python\python36-32\lib\ntpath.py", line 205, in split
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType

**Environment (please complete the following information):**
 - MS Windows 10

**Additional context**
A similar issue  #29722 was reviewed. OpenOCD is is already in path.

cfriedt, 2021-09-14:
@waqarhumayun - can you please provide additional reproducibility information? E.g. the specific west command you used?

mbolivar-nordic, 2021-09-15:
Hi, this looks like a misconfiguration.

Please provide the following information and reopen.

- zephyr version used
- board used
- west build command