zephyrproject-rtos/zephyr #38484: boards: beaglev_starlight_jh7100: integrate .bin processing script into tree

cfriedt, 2021-09-12:
**Is your enhancement proposal related to a problem? Please describe.**
In [the docs](https://docs.zephyrproject.org/latest/boards/riscv/beaglev_starlight_jh7100/doc/index.html), it mentions
> The bootloader expects size information at the start of the binary file, so the bin file needs to be processed first to include that information. Download the helper script from starfive-tech github repo here

with a link pointing to 

https://github.com/starfive-tech/freelight-u-sdk/blob/starfive/fsz.sh

**Describe the solution you'd like**
It would probably be helpful to simply integrate that process into some board-specific hook when west is invoked

**Describe alternatives you've considered**

**Additional context**
Reminded about this extra step when checking #38397
Not sure if this is necessarily a bug, but I feel it breaks the usual flow of development having to run an extra step.

henrikbrixandersen, 2021-09-12:
You may be able to find some inspiration in how post processing is done in https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/arty/CMakeLists.txt

cfriedt, 2021-09-14:
Damn... the other 2 links to Seeed Studio's site for that board return 404. I heard production was delayed until next year. Odd that they would take down the BeagleV Getting Started guide and the bootloader docs.

closed 2023-03-01 by cfriedt