zephyrproject-rtos/zephyr #38485: x_nucleo_iks02a1 won't compile error 'DT_N_INST_0_st_mpxxdtyy_FULL_NAME' undeclared

richbai90, 2021-09-13:
**Describe the bug**
I don't know if this is a bug with zephyr, with platformio, or just something I'm missing but I don't know where else to ask for help. When I try to compile the x_nucleo_iks02a1 microphone sample, it fails with this error

```
/Users/richbaird/.platformio/packages/framework-zephyr/include/devicetree.h:301:40: error: 'DT_N_INST_0_st_mpxxdtyy_FULL_NAME' undeclared here (not in a function); did you mean 'DT_N_S_otgfs_phy_FULL_NAME'?
 #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
                                        ^~~~~~~~~
/Users/richbaird/.platformio/packages/framework-zephyr/include/device.h:771:11: note: in definition of macro 'Z_DEVICE_DEFINE'
   .name = drv_name,     \
           ^~~~~~~~
/Users/richbaird/.platformio/packages/framework-zephyr/include/sys/util_internal.h:64:53: note: in expansion of macro '__DEBRACKET'
 #define __GET_ARG2_DEBRACKET(ignore_this, val, ...) __DEBRACKET val
                                                     ^~~~~~~~~~~
/Users/richbaird/.platformio/packages/framework-zephyr/include/sys/util_internal.h:59:2: note: in expansion of macro '__GET_ARG2_DEBRACKET'
  __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
  ^~~~~~~~~~~~~~~~~~~~
/Users/richbaird/.platformio/packages/framework-zephyr/include/sys/util_internal.h:54:2: note: in expansion of macro '__COND_CODE'
  __COND_CODE(_XXXX##_flag, _if_1_code, _else_code)
  ^~~~~~~~~~~
/Users/richbaird/.platformio/packages/framework-zephyr/include/sys/util_macro.h:117:2: note: in expansion of macro 'Z_COND_CODE_1'
  Z_COND_CODE_1(_flag, _if_1_code, _else_code)
  ^~~~~~~~~~~~~
/Users/richbaird/.platformio/packages/framework-zephyr/include/devicetree.h:637:2: note: in expansion of macro 'COND_CODE_1'
  COND_CODE_1(DT_NODE_HAS_PROP(node_id, prop), \
  ^~~~~~~~~~~
/Users/richbaird/.platformio/packages/framework-zephyr/include/device.h:180:2: note: in expansion of macro 'DT_PROP_OR'
  DT_PROP_OR(node_id, label, DT_NODE_FULL_NAME(node_id))
  ^~~~~~~~~~
/Users/richbaird/.platformio/packages/framework-zephyr/include/devicetree.h:464:36: note: in expansion of macro 'DT_CAT'
 #define DT_NODE_FULL_NAME(node_id) DT_CAT(node_id, _FULL_NAME)
                                    ^~~~~~
/Users/richbaird/.platformio/packages/framework-zephyr/include/device.h:180:29: note: in expansion of macro 'DT_NODE_FULL_NAME'
  DT_PROP_OR(node_id, label, DT_NODE_FULL_NAME(node_id))
                             ^~~~~~~~~~~~~~~~~
/Users/richbaird/.platformio/packages/framework-zephyr/include/device.h:223:4: note: in expansion of macro 'DEVICE_DT_NAME'
    DEVICE_DT_NAME(node_id), init_fn,  \
    ^~~~~~~~~~~~~~
/Users/richbaird/.platformio/packages/framework-zephyr/include/device.h:239:2: note: in expansion of macro 'DEVICE_DT_DEFINE'
  DEVICE_DT_DEFINE(DT_DRV_INST(inst), __VA_ARGS__)
  ^~~~~~~~~~~~~~~~
/Users/richbaird/.platformio/packages/framework-zephyr/include/sys/util_internal.h:84:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
 #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
                          ^~~~~~~~~~~~~~~~~~
/Users/richbaird/.platformio/packages/framework-zephyr/include/devicetree.h:301:31: note: in expansion of macro 'UTIL_CAT'
 #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
                               ^~~~~~~~
/Users/richbaird/.platformio/packages/framework-zephyr/include/devicetree.h:1807:27: note: in expansion of macro 'DT_INST'
 #define DT_DRV_INST(inst) DT_INST(inst, DT_DRV_COMPAT)
                           ^~~~~~~
/Users/richbaird/.platformio/packages/framework-zephyr/include/device.h:239:19: note: in expansion of macro 'DT_DRV_INST'
  DEVICE_DT_DEFINE(DT_DRV_INST(inst), __VA_ARGS__)
                   ^~~~~~~~~~~
/Users/richbaird/.platformio/packages/framework-zephyr/drivers/audio/mpxxdtyy.c:168:1: note: in expansion of macro 'DEVICE_DT_INST_DEFINE'
 DEVICE_DT_INST_DEFINE(0, mpxxdtyy_initialize, NULL, &mpxxdtyy_data, NULL,
 ^~~~~~~~~~~~~~~~~~~~~
*** [.pio/build/disco_l072cz_lrwan1/drivers__audio/zephyr/drivers/audio/mpxxdtyy.c.o] Error 1
================== [FAILED] Took 5.76 seconds ==================
```

It seems that it's not finding the mpxxdtyy device in any of the included headers. The full backtrace begins at line 52 of the sample

```c
const struct device *mic_dev = device_get_binding(DT_LABEL(DT_INST(0, st_mpxxdtyy)));
```

I'm just barely learning zephyr, and it's my first RTOS at that, so I really don't know where to go from here. I am using platform IO to simplify things a bit, but I have set it up to use cmake version 3.21.2. I also tried compiling the sample using west with this command `west build -b b_l072z_lrwan1 -s samples/shields/x_nucleo_iks02a1/microphone --cmake-only but I get an error about arduino I2s.

My CMakeLists.txt file is exactly as in the sample, as is my prj.conf file. Any help is appreciated, and sorry again for posting this here, is there a community where I can post these questions? I didn't see a forum on the zephyr project website, and I did join the discord but it is mostly about the project itself, not applications developed on the platform, and stackoverflow has only 72 followers of the zephyr tag.

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain cmake and platformio
 - Commit SHA or Version used 2.6.0.0

erwango, 2021-09-13:
@richbai90 Please provide details on the builds commands you used as well as modifications that you might have made on target files.

avisconti, 2021-09-13:
@richbai90  Can you try if this command work also n your case? west build -b nucleo_f411re -s samples/shields/x_nucleo_iks02a1/microphone
Thanks!

richbai90, 2021-09-13:
> @richbai90 Can you try if this command works also in your case?
> 
> ```
> west build -b nucleo_f411re -s samples/shields/x_nucleo_iks02a1/microphone                                                                                
> ```
> 
> Thanks!

That worked without an issue. I noticed that that board has the following in the built zephyr.dts file where the board I am using does not.

```
i2s1: arduino_i2s: i2s@40013000 {
			compatible = "st,stm32-i2s";
			#address-cells = < 0x1 >;
			#size-cells = < 0x0 >;
			reg = < 0x40013000 0x400 >;
			clocks = < &rcc 0x3 0x1000 >;
			interrupts = < 0x23 0x5 >;
			dmas = < &dma2 0x3 0x3 0x400 0x3 &dma2 0x2 0x3 0x400 0x3 >;
			dma-names = "tx", "rx";
			status = "okay";
			label = "I2S_1";
			pinctrl-0 = < &i2s1_ck_pa5 &i2s1_sd_pa7 >;
			mp34dt05@0 {
				compatible = "st,mpxxdtyy";
				reg = < 0x0 >;
				label = "MP34DT05";
			};
		};
```

If I'm not mistaken, this is defining the arduino compatible headers on the nucleo board, is that accurate? I tried adding it to the zephyr.dts file for my board in the build directory, but I am not sure that's the right place to add it.

avisconti, 2021-09-13:
> If I'm not mistaken, this is defining the arduino compatible headers on the nucleo board, is that accurate? I tried adding it to the zephyr.dts file for my board in the build directory, but I am not sure that's the right place to add it. Also, is there a detailed guide on understanding what all of this means, and how to use it / change it?

No, no, it is not the right place. The build directory contains things that are built.
You should modify the dts file in the board directory to enable arduino_i2s:
e.g. boards/arm/nucleo_f411re/nucleo_f411re.dts

Then here define the PLL for I2S of your soc
boards/shields/x_nucleo_iks02a1/boards/

And then most of the things are under dts/ already done for your soc. For example for F411RE board:
dts/arm/st/f4/stm32f411.dtsi

What is your soc?

richbai90, 2021-09-13:
> No, no, it is not the right place. The build directory contains things that are built.

Yeah that's what I figured, but I had to try.

> Then here define the PLL for I2S of your soc
> boards/shields/x_nucleo_iks02a1/boards/

Is there a guide for this?

> And then most of the things are under dts/ already done for your soc

Is there anything I will need to add to my soc dtsi file?

> What is your soc?

It is the B-L072Z-LRWAN1 by ST. In zephyr you label it b_l072z_lrwan1

manoj153, 2021-09-14:
> Yeah that's what I figured, but I had to try.
>
>
> Is there a guide for this?
>
>
> Is there anything I will need to add to my soc dtsi file?
>
>
> It is the B-L072Z-LRWAN1 by ST. In zephyr you label it [...]

a quick check by me and noticed none of the `l0` series has i2s defined.

avisconti, 2021-09-14:
> 
> > Then here define the PLL for I2S of your soc
> > boards/shields/x_nucleo_iks02a1/boards/
> 
> Is there a guide for this?
> 

The soc reference manual has a I2S section that describes the PLL_I2S. And then also the chapter describing the clocktree and how to feed I2S is important.

> > And then most of the things are under dts/ already done for your soc
> 
> Is there anything I will need to add to my soc dtsi file?

Mmmmh, maybe not. It is something that has to be checked.

> 
> > What is your soc?
> 
> It is the B-L072Z-LRWAN1 by ST. In zephyr you label it b_l072z_lrwan1

Do you have the reference manual of this soc? I saw @manoj153  is saying that i2s is not supported.

manoj153, 2021-09-14:
> The soc reference manual has a I2S section that describes the PLL_I2S. And then also the chapter describing the clocktree and how to feed I2S is important.
>
>
> Mmmmh, maybe not. It is something that [...]

https://www.st.com/resource/en/data_brief/b-l072z-lrwan1.pdf
According to the doc: SoC : `STM32L072CZ M`

avisconti, 2021-09-14:
@richbai90 @manoj153 

The soc in the B-L072Z-LRWAN1 kit is the STM32L0, and the reference manual is here:
https://www.st.com/resource/en/reference_manual/rm0377-ultralowpower-stm32l0x1-advanced-armbased-32bit-mcus-stmicroelectronics.pdf

I saw that i2s is supported. All the info (PLL I2S, clocktree, ...) can be found here. I don't know if @erwango would like to add anything to this.

erwango, 2021-09-14:
> I saw that i2s is supported. All the info (PLL I2S, clocktree, ...) can be found here. I don't know if @erwango would like to add anything to this.

If there's no I2S node defined in L0 dtsi, it also means that I2S driver is not validated on this series.
So it's possible it requires some care.

avisconti, 2021-09-14:
> > I saw that i2s is supported. All the info (PLL I2S, clocktree, ...) can be found here. I don't know if @erwango would like to add anything to this.
> 
> If there's no I2S node defined in L0 dtsi, it also means that I2S driver is not validated on this series.
> So it's possible it requires some care.

It seems that under dts/arm/st/l0/ i2s is not defined...

erwango, 2021-09-14:
> > > I saw that i2s is supported. All the info (PLL I2S, clocktree, ...) can be found here. I don't know if @erwango would like to add anything to this.
> > 
> > 
> > If there's no I2S node defined in L0 dtsi, it also means that I2S driver is not validated on this series.
> > So it's possible it requires some care.
> 
> It seems that under dts/arm/st/l0/ i2s is not defined in any dtsi ...

I think that's what @manoj153 reported [earlier](https://github.com/zephyrproject-rtos/zephyr/issues/38485#issuecomment-918832081)

manoj153, 2021-09-14:
> I think that's what @manoj153 reported [earlier](https://github.com/zephyrproject-rtos/zephyr/issues/38485#issuecomment-918832081)

yes, that's what I meant and during the quick glance, I noticed the `i2s` on this SoC is piggybacking SPI2, so some register config has to be done on SPI2 peripheral before using it as i2s. and I wonder how the dt definition will be like SPI2 or I2S?

richbai90, 2021-09-14:
I’m working in writing the definitions now. This is my first time doing anything like this so I may have a few questions as I go along

erwango, 2021-09-14:
> s what I meant and during the quick glance, I noticed the `i2s` on this SoC is piggybacking SPI2, so some register config has to be done on SPI2 peripheral before using it as i2s. and I wonder how the dt definition will be like SPI2 or I2S?

That's actually the same on F4, on which i2s is functional.
So you can check on F4 state of affairs.

richbai90, 2021-09-14:
Here is what I have so far, I've modified the `stm32l072.dts` file adding the following mostly copied from f4. I changed the address, clock, and interrupts to be those from the SPI1 definition in the `stm32l0.dtsi` per the data sheet this should actually be SPI2, but there exists no definition for SPI2 in the `stm32l072.dts` file. Maybe we need to add that first? I am not sure where to find the addresses and clock details for the SPI2 interface. I thought the datasheet would hold that information, but that doesn't appear to be the case.

> One standard I2S interfaces (multiplexed with SPI2) is available. It can operate in master or
slave mode, and can be configured to operate with a 16-/32-bit resolution as input or output
channels. Audio sampling frequencies from 8 kHz up to 192 kHz are supported. When the
I2S interfaces is configured in master mode, the master clock can be output to the external
DAC/CODEC at 256 times the sampling frequency.

https://www.st.com/resource/en/datasheet/stm32l072cz.pdf

```
	i2s1: i2s@40013000 {
		compatible = "st,stm32-i2s";
		#address-cells = <1>;
		#size-cells = <0>;
		reg = <0x40013000 0x400>; // Same as SPI1
		clocks = <&rcc STM32_CLOCK_BUS_APB2 0x00001000>; // same as SPI1 
		interrupts = <25 3>; // same as SPI
		dmas = <&dma2 3 3 0x400 0x3
		&dma2 2 3 0x400 0x3>;
		dma-names = "tx", "rx";
		status = "disabled";
		label = "I2S_1";
	};
```

richbai90, 2021-09-14:
I'm also unsure if anything needs to be changed to handle the multiplexing, or if that is done in hardware.

avisconti, 2021-09-15:
> yes, that's what I meant and during the quick glance, I noticed the `i2s` on this SoC is piggybacking SPI2, so some register config has to be done on SPI2 peripheral before using it as i2s. and I [...]

I'm getting this now. Sorry.

closed 2021-11-30 by github-actions[bot]