zephyrproject-rtos/zephyr #38494: Flooded logs when using CDC_ACM as back-end

emob-nordic, 2021-09-13:
**Is your enhancement proposal related to a problem? Please describe.**
If USB CDC_ACM is selected as logging back-end and the USB cable is not connected to the board at the start up, logs could be flooded with `LOG_WRN("Device not configured or suspended, drop %d bytes", len);`.

**Describe the solution you'd like**
The best option would be for logger back-end not to try to send anything on CDC_ACM back-end, until its fully operational and ready to send the data (backend_ready_callback).

**Describe alternatives you've considered**
* Changing logging priority of the CDC_ACM class so that no warning logs would be processed.
* Delaying logger processing thread with `CONFIG_LOG_PROCESS_THREAD_STARTUP_DELAY_MS`

Both above alternatives are workaround and are not satisfactory solutions.

**Additional context**
AFAIK CDC_ACM is using uart API so from logger perspective anything could be connected there and there is no good mechanism to know which back-end is chosen (except for Kconfig option).

To get above enhancement in i would propose to:
1. Separate setting usb_status_callback from usb_enable(). Add usb_register_status_callback() or whatsoever similar. 
2. Allow to set multiple status callbacks for single USB device.
3. In case USB CDC_ACM is used as logging back-end, use newly introduced function to register back-end callback during logger initialization.
4. Add USB state tracking in logger subsystem and allow to send the data only if the back-end is ready.

This could be more generic for other back-ends where the device may not always be ready to send the data.
I think the assumption that the back-end is always ready is incorrect.

jfischer-no, 2021-09-15:
> The best option would be for logger back-end not to try to send anything on CDC_ACM back-end, until its fully operational and ready to send the data (backend_ready_callback).

There will not be this kind of callback. UART driver API is sufficient to solve it, see https://github.com/zephyrproject-rtos/zephyr/pull/38203

> **Describe alternatives you've considered**
> 
>     * Changing logging priority of the CDC_ACM class so that no warning logs would be processed.

This is not a workaround it is mandatory configuration, CDC ACM logging should be off or at least configured to ERR, regardless whether backend is capable to wait for readiness of CDC ACM UART or not.

And, since it keeps being described incorrectly, even if USB device is connected to host and enumerated, the CDC ACM UART does not need to be "ready" unless a host application is accessing it. The correct way to detect it is like https://github.com/zephyrproject-rtos/zephyr/pull/38203.

emob-nordic, 2021-09-15:
@nordic-krch Could we have something similar to #38203 for uart back-end in logger?

nordic-krch, 2021-09-15:
The difference between logging and shell is that shell has dedicated thread for each instance and here there is one logging thread for all backends so approach must be non-blocking.

I'm currently working on multidomain logger and there an entity type called `link` is introduced. it is a pipe from other domain through which remote messages are added. Links are similar to backends in a way that it also has specified API and all are in the same memory section. All links are initialized by the logger at the same stage as backends. Initialization may be time consuming so i used an approach where link instead of `init` has `initiate` and `activate`. Both are supposed to be non blocking calls. At the beginning `initiate` is called for each link. Then in thread loop `activate` is called which returns success when link is ready. Logging thread keeps mask of all activated links and once all links in the system are activated it stops polling them.

I will look into something similar for backends but it must be backward compatible.

PR #43405 (2022-06-09): logging: Add option for prolonged backend initialization
Extended logging backend API with log_backend_is_ready call which
returns 0 is backend is ready. Logging core will make sure that
all autostarted backends are ready before they are enabled.
This option allows to handle backends which are not yet ready
after init function is called (e.g. usb backend that is not plugged
in). If this is the only backend in the system, logging processing
will not start util first backend is ready.

Function for checking readiness is optional and when backend has
no such function it is assumed that backend is [... cut]
closed 2022-06-09 by carlescufi