zephyrproject-rtos/zephyr #38502: Update mcumgr library to fix wrong callback state

lairdjm, 2021-09-13:
**Describe the bug**
mcumgr's library has callbacks which report the status of a firmware update with 3 states - start, ongoing and complete. However, complete is not used.

This is fixed in this PR to the upstream mynewt-mcumgr project code: https://github.com/apache/mynewt-mcumgr/pull/137

Therefore this is a request to pull the change into zephyr's version of mcumgr so that it will be included for Zephyr 2.7

de-nordic, 2021-09-21:
Included here: https://github.com/zephyrproject-rtos/mcumgr/pull/43

de-nordic, 2021-10-04:
@cfriedt This is low priority bug, so it will not go into the 2.7. Should we re-target it? Also the bug is in the mcumgr rather than in Zephyr, in common code, so is it needed to keep this as a bug here? It has already been fixed in mcumgr but we are not able to change west.yml to include the fix at this point.

cfriedt, 2021-10-04:
@de-nordic - we should be able to pull into main. What is preventing the west.yml update?

de-nordic, 2021-10-04:
@cfriedt The mcumgr updates come in snapshot updates. To bring this fix in we would need to bring in features/enhancements that ere not "blockers".
Alternative is to create mcumgr-zephyr-2.7-lts branch and start bringing the fixes there, and use this branch in west.yml for lts.
Do we want to start the mcumgr-zephyr-2.7-lts branch already? It may happen anyway.

lairdjm, 2021-10-04:
Is it possible to get this included in a 2.7 backport, just the bug fix not the addition of features? We currently manually have this patched in a modified 2.6 branch and the idea was we would scrap it when moving to 2.7 because the fix would be included, but if the idea is this won't be fixed in 2.7 then that really doesn't sit well for our development

de-nordic, 2021-10-04:
@cfriedt So this could go in: https://github.com/zephyrproject-rtos/zephyr/pull/39101 ?

cfriedt, 2021-10-04:
> @cfriedt So this could go in: #39101 ?

I think the issue there is that it would be tied to a PR rather than the main branch of mcumgr.

Would it make sense to merge zephyrproject-rtos/mcumgr#43 into the master branch of mcumgr and then update #39101 to use the resulting master commit?

Or if these are changes specific to v2.7 that have been cherry-picked, they will need to wait until after v2.7.0 unless they have "Blocker" status.

I'll bring this up in the release meeting in any case though.

de-nordic, 2021-10-04:
> I think the issue there is that it would be tied to a PR rather than the main branch of mcumgr.
>
> Would it make sense to merge [...]

@cfriedt  Yes I will merge it, need reviews from people, also can you join in. What I am asking is whether such set of changes, as done with the snapshot in that PR, is acceptable, and whether sha in west can be updated to point to it, when merged.
Review here https://github.com/zephyrproject-rtos/mcumgr/pull/43.

No cherry picking. this is normal snapshot update.

carlescufi, 2021-11-09:
Fixed by #39658

closed 2021-11-09 by carlescufi