zephyrproject-rtos/zephyr #38512: stm32f7: CAN: STM32F645VE CAN signal seems upside down.

easternblack, 2021-09-14:
**Describe the bug**
I have some custom PCB with STM32L433, and each board communicate by CAN iso-tp. and it pretty work well.   
after that, I need to add ethernet functions to my board, so I changed MCU to STM32F745VE.   
and then, CAN iso-tp does not work, so I checked there signal via measuring machines, and it's signal seems upside down compared with STM32L433. Here is a picture of measurement result of them.   
[image]   
STM32L433 using CAN1, and STM32F745VE one using CAN2, but CAN1 of STM32F745VE looks same with CAN2.   
they also have external CAN chip, but the signal on the picture is measured at between MCU and CAN chip.

**To Reproduce**
I used iso-tp sample program in zephyr origin.   
/zephyr/samples/subsys/canbus/isotp/   
Just compiled them and run.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

erwango, 2021-09-14:
@alexanderwachter Would you have time to check this ?

alexanderwachter, 2021-09-14:
@erwango never saw this behavior before. Is there a invert flag for the pinmux? CAN is defined to be a high signal in idle (recessive bit = 0).

easternblack, 2021-09-15:
I'm using default pinmux of zephyr origin.   
here's details for using CAN.
**zephyr/drivers/pinmux/stm32/pinmux_stm32f7.h**   
```
#define STM32F7_PINMUX_FUNC_PB5_CAN2_RX \
        (STM32_PINMUX_ALT_FUNC_9 | STM32_PUPDR_PULL_UP)
#define STM32F7_PINMUX_FUNC_PB6_CAN2_TX \
        (STM32_PINMUX_ALT_FUNC_9 | STM32_PUSHPULL_NOPULL)
```
and here is CAN1 and CAN2 dts node.   
```
can1: can@40006400 {
        compatible = "st,stm32-can";
        #address-cells = <1>;
        #size-cells = <0>;
        reg = <0x40006400 0x400>;
        interrupts = <19 0>, <20 0>, <21 0>, <22 0>;
        interrupt-names = "TX", "RX0", "RX1", "SCE";
        clocks = <&rcc STM32_CLOCK_BUS_APB1 0x02000000>;
        status = "disabled";
        label = "CAN_1";
        bus-speed = <125000>;
        sjw = <1>;
        prop-seg = <0>;
        phase-seg1 = <4>;
        phase-seg2 = <5>;
};

can2: can@40006800 {
        compatible = "st,stm32-can";
        #address-cells = <1>;
        #size-cells = <0>;
        reg = <0x40006800 0x400>;
        interrupts = <63 0>, <64 0>, <65 0>, <66 0>;
        interrupt-names = "TX", "RX0", "RX1", "SCE";
        clocks = <&rcc STM32_CLOCK_BUS_APB1 0x04000000>;
        master-can-reg = <0x40006400>;
        status = "disabled";
        label = "CAN_2";
        sjw = <1>;
        prop-seg = <0>;
        phase-seg1 = <4>;
        phase-seg2 = <5>;
};
```
of course status is set to okay in the overlay file.

alexanderwachter, 2021-09-15:
I had a look at the pictures again. It seems that you are measuring in CAN_H and CAN_L. In the second picture, it looks like most of the data, if not all, is zero. The dominant spices you see in the picture are probably stuffing bits.

easternblack, 2021-09-15:
Okay, I'll check with more simple and clear CAN sample apps.   
But next week is week-long national holiday in our country, so the result probably be late.   
   
I'll re-open this issue if there are something to notice.

closed 2021-09-15 by easternblack