zephyrproject-rtos/zephyr #38514: mqtt azure sample failing with net_tcp "is waiting on connect semaphore"

swethad11782809, 2021-09-14:
Issue observed with MQTT on the TCP2 stack. Specifically,  TCP connects fail while waiting on semaphore.
Tried with increasing the TCP_ACK and SOCKET_CONNECT timeouts to more than 10 seconds, still not able to connect.

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
<inf> net_config: Running dhcpv4 client...
<dbg> mqtt_azure.main: Waiting for network to setup...
<wrn> net_gptp: Reset Pdelay requests
<wrn> net_gptp: Reset Pdelay requests
<wrn> net_gptp: Reset Pdelay requests
<inf> net_dhcpv4: Received: 192.168.1.102
<inf> net_config: IPv4 address: 192.168.1.102
<inf> net_config: Lease time: 86400 seconds
<inf> net_config: Subnet: 255.255.255.0
<inf> net_config: Router: 192.168.1.1
<wrn> net_gptp: Reset Pdelay requests
<wrn> net_gptp: Reset Pdelay requests
<wrn> net_gptp: Reset Pdelay requests
<dbg> mqtt_azure.try_to_connect: attempting to connect...
<dbg> net_tcp.tcp_conn_ref: (main): conn: 0x600ca340, ref_count: 1
<dbg> net_tcp.tcp_conn_alloc: (main): conn: 0x600ca340
<dbg> net_tcp.net_tcp_connect: (main): context: 0x600a7f7c, local: 0.0.0.0, remote: 0.0.0.0
<dbg> net_tcp.net_tcp_connect: (main): conn: 0x600ca340 src: 192.168.1.102, dst: 0.0.0.0
<dbg> net_tcp.tcp_in: (main):  [LISTEN Seq=34736396 Ack=0]
<dbg> net_tcp.tcp_out_ext: (main): SYN Seq=34736396 Len=0
<dbg> net_tcp.tcp_send_process_no_lock: (main): SYN Seq=34736396 Len=0
<dbg> net_tcp.tcp_send: (main): SYN Seq=34736396 Len=0
<err> net_tcp: net_send_data()
<dbg> net_tcp.tcp_in: (main): LISTEN->SYN_SENT
<dbg> net_tcp.tcp_send_process_no_lock: (tcp_work): SYN Seq=34736396 Len=0 in_retransmission
<dbg> net_tcp.tcp_send: (tcp_work): SYN Seq=34736396 Len=0
<err> net_tcp: net_send_data()
<wrn> net_gptp: Reset Pdelay requests
<dbg> net_tcp.tcp_send_process_no_lock: (tcp_work): SYN Seq=34736396 Len=0 in_retransmission
<dbg> net_tcp.tcp_send: (tcp_work): SYN Seq=34736396 Len=0
<err> net_tcp: net_send_data()
<dbg> net_tcp.tcp_send_process_no_lock: (tcp_work): SYN Seq=34736396 Len=0 in_retransmission
<dbg> net_tcp.tcp_send: (tcp_work): SYN Seq=34736396 Len=0
<err> net_tcp: net_send_data()
<dbg> net_tcp.tcp_send_process_no_lock: (tcp_work): SYN Seq=34736396 Len=0 in_retransmission
<dbg> net_tcp.tcp_send: (tcp_work): SYN Seq=34736396 Len=0
<err> net_tcp: net_send_data()
<dbg> net_tcp.tcp_send_process_no_lock: (tcp_work): SYN Seq=34736396 Len=0 in_retransmission
<dbg> net_tcp.tcp_send: (tcp_work): SYN Seq=34736396 Len=0
<err> net_tcp: net_send_data()
<dbg> net_tcp.tcp_send_process_no_lock: (tcp_work): SYN Seq=34736396 Len=0 in_retransmission
<dbg> net_tcp.tcp_send: (tcp_work): SYN Seq=34736396 Len=0
<err> net_tcp: net_send_data()
<wrn> net_gptp: Reset Pdelay requests
<dbg> net_tcp.tcp_send_process_no_lock: (tcp_work): SYN Seq=34736396 Len=0 in_retransmission
<dbg> net_tcp.tcp_send: (tcp_work): SYN Seq=34736396 Len=0
<err> net_tcp: net_send_data()
<dbg> net_tcp.tcp_send_process_no_lock: (tcp_work): SYN Seq=34736396 Len=0 in_retransmission
<dbg> net_tcp.tcp_send: (tcp_work): SYN Seq=34736396 Len=0
<err> net_tcp: net_send_data()
<dbg> net_tcp.tcp_send_process_no_lock: (tcp_work): SYN Seq=34736396 Len=0 in_retransmission
<dbg> net_tcp.tcp_send: (tcp_work): SYN Seq=34736396 Len=0
<err> net_tcp: net_send_data()
<dbg> net_tcp.tcp_send_process_no_lock: (tcp_work): SYN Seq=34736396 Len=0 in_retransmission
<dbg> net_tcp.tcp_conn_unref_debug: (tcp_work): conn: 0x600ca340, ref_count=1 (tcp_send_process():520)
<dbg> net_tcp.tcp_conn_unref_debug: (tcp_work): conn: 0x600ca340 is waiting on connect semaphore
<wrn> net_gptp: Reset Pdelay requests
<dbg> net_tcp.tcp_conn_unref_debug: (main): conn: 0x600ca340, ref_count=1 (net_tcp_connect():2283)
<dbg> net_tcp.net_tcp_connect: (main): conn: 0x600ca340, ret=-116
<dbg> net_tcp.net_tcp_recv: (main): context: 0x600a7f7c, cb: (nil), user_data: (nil)
<dbg> net_tcp.net_tcp_unref: (main): context: 0x600a7f7c, conn: (nil)
<err> mqtt_azure: mqtt_connect failed -116
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

tbursztyka, 2021-09-14:
On which platform?
On which zephyr commit id?

tbursztyka, 2021-09-14:
> net_tcp.net_tcp_connect: (main): conn: 0x600ca340 src: 192.168.1.102, dst: 0.0.0.0

dst 0.0.0.0 ... does not sound like a proper destination

tbursztyka, 2021-09-14:
Have you properly configured the sample? 

CONFIG_DNS_RESOLVER is set there. Maybe your network does not allow you to do DNS stuff, or go to 8.8.8.8 at least. 

Try enabling DNS debug log

swethad11782809, 2021-09-14:
Platform is "Elkhart lake board"
Commit ID for our base line is "72bd8da095f857600536ffad9deb461a2cdb1d0a"
========================================
commit 72bd8da095f857600536ffad9deb461a2cdb1d0a
Author: Joakim Andersson <joakim.andersson@nordicsemi.no>
Date:   Mon Aug 2 14:32:12 2021 +0200

    Bluetooth: host: Fix key distribution not calling pairing complete

    Fix key distribution not calling pairing complete callback when
    distributing encryption information when not bonding.
    Encryption information is only distributed in legacy pairing so issue
    was only seen during legacy pairing.

    Signed-off-by: Joakim Andersson <joakim.andersson@nordicsemi.no>
===================================================

swethad11782809, 2021-09-14:
Please find the logs with DNS enabled, still can see the same failure on tcp connect

[image]

tbursztyka, 2021-09-14:
Ok, and can you access this destination out of zephyr? (40.83.177.42 on port 8883).
Because I cannot myself. nslookup does the same result as the dns resolver in zephyr, but tcp connection cannot be done.

swethad11782809, 2021-09-15:
This was our working Azure Server, I collected the below logs with Zephyr 1.14 ( TCP1 Stack ).   While migrating to Zephyr 2.X onwards I am facing this connectivity issue.

[image]

swethad11782809, 2021-09-15:
Also when I enable CONFIG_SOCKS, I see the build failure. 
Below is the diff of the prj.conf 

[image]

[image]

tbursztyka, 2021-09-15:
> This was our working Azure Server

this WAS or this IS? ;) (I guess you tested it on 1.14 just now so it probably is). 

> Also when I enable CONFIG_SOCKS, I see the build failure.

You need to provide CONFIG_SAMPLE_SOCKS_ADDR and CONFIG_SAMPLE_SOCKS_PORT

tbursztyka, 2021-09-15:
Now what will be necessary is logs from wireshark.
Log everything that comes from your EHL board running Zephyr.
- get the logs for 1.14 (working path)
- get the logs for current version (broken path).

And hopefully we'll see what is going on with the new TCP stack.

ana-rajput, 2021-09-24:
we are facing difficulty in getting wireshark logs, can you help in any alternative way to get logs from debug configuration.

rlubos, 2021-09-24:
One possible option would be to use `net_capture` library to monitor the traffic on an interface (https://docs.zephyrproject.org/latest/reference/networking/capture.html), note however this is quite recent addition and is not available for 1.14.

carlescufi, 2021-09-29:
@swethad11782809 @ana-rajput could you please try to provide additional information and traces in order to try and reproduce this?

kumarnac, 2021-10-04:
[image]
Comparison of 1p14 and 26.99 connect sequence

kumarnac, 2021-10-04:
[image]

kumarnac, 2021-10-04:
Comparison of 1p14 and 2px connect sequence

kumarnac, 2021-10-05:
The observation is same as what is seen in zephyr logs  earlier. Response to a connect is not received in case of 2.6.99 .
 In 1.14 the MSS field is set to 1460 when a connect request is sent. However, in case of 2.x , the  MMS field is absent when a request is sent.

kumarnac, 2021-10-05:
@rlubos -  In 2.6.99  how is the MSS set ? Can having this as zero lead to connect failures ?

kumarnac, 2021-10-05:
@jukkar - Any inputs regarding  setting MSS  for a SYN request ?

rlubos, 2021-10-05:
@kumarnac MSS option is not implemented in TCP2 at the moment.

carlescufi, 2021-10-06:
Closing this issue since MSS support is already being tracked in this issue: #30367

closed 2021-10-06 by carlescufi