zephyrproject-rtos/zephyr #38527: lwm2m: re-register instead of removing observer on COAP reset answer to notification

huelsenfruchtzwerg, 2021-09-14:
Right now, when the lwm2m_engine gets a Coap reset message in response to a notification, it removes the observer but continues normally otherwise.

This usually happens (afaict) when the LWM2M server the client was communicating with has been restarted and lost the state of registrations/observations. (Coap has the [following to say to Coap Reset](https://tools.ietf.org/id/draft-ietf-core-coap-09.html#reset): "A Reset message indicates that a specific message (confirmable or non-confirmable) was received, but some context is missing to properly process it. This condition is usually caused when the receiving node has rebooted and has forgotten some state that would be required to interpret the message.")

It seems to me that a better reaction to this would be to re-register instead, since it seems reasonable to assume that the server then has also lost the state of all other observations and the registration itself.

What happens right now is that the device will only re-register the next time it would send an registration-update, resulting in all notifications created up until then to be dropped/lost.

rlubos, 2021-09-15:
I don't think that LwM2M engine should make any assumption based on some common behavior. Yes, the Reset might be sent if the server has lost all its observations due to a reboot, but on the other hand it's also a perfectly vialbe mechanism to Cancel observation according to spec (the observer can send a Reset message if it's no longer interested in particular resource):
https://datatracker.ietf.org/doc/html/rfc7641#section-3.6

What we could consider is some better notification result reporting mechanism (as per #38531) to report such events to the application, and let the application decide on how to deal with the situation.

boaks, 2021-09-16:
Do you use [eclipse/leshan](https://github.com/eclipse/leshan)?

If so, at least two-three years ago, leshan used the californium function, to persist the coap-observe-client (that's the lwm2m-server). AFAIK, redis was used, though the use case was more a "leshan-cluster" then a graceful-restart. Maybe, you ask there in the eclipse/leshan, if this coap-observe-client with redis still works and could be used for restarts.

huelsenfruchtzwerg, 2021-09-19:
@boaks That's a very acute observation. :)
We had some more discussions internally and arrived at the conclusion that this is better solved in the backend by introducing a registration/observation store, as you suggest.
I still believe better visibility of the application into what observations do/don't exist/are canceled would be a very desirable feature, but as @rlubos this would probably fit well within the suggested more generic approach.

closed 2021-09-19 by huelsenfruchtzwerg

boaks, 2021-09-20:
Just for curiosity: which implementation do you use for the lwm2m-server/coap-client? (If allowed to publish that.)