zephyrproject-rtos/zephyr #38534: lwm2m: add api to inspect observation state of resource/object

huelsenfruchtzwerg, 2021-09-14:
RIght know there's no (*) way for the application to know whether a resource/object is observed from the lwm2m server. From an high/abstract level this is not an issue, the app should not need to care about what the server is or is not interested in. However there are a number of use cases where this knowledge would be very useful:

1. some indicator on the device to signal "ready to use" when expected functionality depends on some observations to be established
2. (in the absence of the "Notification storing when offline" feature) to buffer some data locally to transmit it when the observation has been established

(*): There actually is one very hacky way: lwm2m_notify_* returns the number of existing observation instances, but not without creating a most likely superfluous notification,

rlubos, 2021-09-15:
Would a generic observe callback proposed in #38531 do the job? Perhaps we could cover the enhancements under a single umbrella issue, instead of 3 separate ones, since they're all related?

huelsenfruchtzwerg, 2021-09-19:
Yes, I agree, a generic solution seems like the most desireable outcome.

MaikVermeulen, 2021-09-29:
For everyone interested:
```
Currently on Github I see that there are 3 or so issues all related to buffering/handling intermittent connectivity.

For focussed discussion, I think it would be nicer to have a central thread on it.
I’ve made a thread ‘Client offline mode, data buffering, etc.’ in the LwM2M channel on the Discord.
```

huelsenfruchtzwerg, 2021-09-29:
closing in favor of the more generic #38531

closed 2021-09-29 by huelsenfruchtzwerg