zephyrproject-rtos/zephyr #38544: drivers: wifi: esWIFI: Regression due to 35815

nandojve, 2021-09-14:
**Describe the bug**
The #35815 introduced a regression that create a BUS FAULT, probably because `socket->context` is invalid. Rollback patch problem disappeared.

**To Reproduce**
Just build [TagoIO IoT Cloud HTTP Sample](https://docs.zephyrproject.org/latest/samples/net/cloud/tagoio_http_post/README.html) using esWIFI with UART transport and flash board. You need follow sample instructions to add SSID and PASSWD to connect to your own wifi router.

**Expected behavior**
Fix problem.

**Impact**
showstopper

**Logs and console output**
```
uart:~$ *** Booting Zephyr OS build v2.7.0-rc1-94-g8956556cc7ff  ***
[00:00:00.600,000] <inf> tagoio_http_post: TagoIO IoT - HTTP Client - Temperature demo
[00:00:00.600,000] <inf> tagoio_http_post: WIFI try connecting to TRENDnet...
uart:~$ 
[00:00:03.462,000] <inf> tagoio_http_post: WIFI Connected
[00:00:03.504,000] <inf> tagoio_http_post: Temp: 87
[00:00:03.553,000] <err> os: ***** BUS FAULT *****
[00:00:03.553,000] <err> os:   Precise data bus error
[00:00:03.553,000] <err> os:   BFAR Address: 0x5154
[00:00:03.553,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x2002711e  r2/a3:  0x000050ce
[00:00:03.553,000] <err> os: r3/a4:  0x0000003e r12/ip:  0x00000000 r14/lr:  0x0800ddf9
[00:00:03.553,000] <err> os:  xpsr:  0x61000000
[00:00:03.553,000] <err> os: Faulting instruction address (r15/pc): 0x0800ed52
[00:00:03.553,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:03.553,000] <err> os: Current thread: 0x20022548 (main)
[00:00:03.633,000] <err> os: Halting system
```

At zephyr.lst file:
```
	net_context_set_state(socket->context, NET_CONTEXT_CONNECTED);
 800ed4e:	6862      	ldr	r2, [r4, #4]
	return 0;
 800ed50:	2000      	movs	r0, #0
	context->flags &= ~(NET_CONTEXT_STATE_MASK << NET_CONTEXT_STATE_SHIFT);
 800ed52:	f8b2 3086 	ldrh.w	r3, [r2, #134]	; 0x86
 800ed56:	f023 0306 	bic.w	r3, r3, #6
```
Failed at `net_context_set_state` introduced by #35815
https://github.com/zephyrproject-rtos/zephyr/blob/79952ad774caf380fca4d24cd12f397b9c337d29/drivers/wifi/eswifi/eswifi_socket.c#L201-L210

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK 0.13.0
 - Commit SHA 88b60f31c7

**Additional context**
Board: nucleo-f767zi
Shield: inventek_eswifi / inventek_eswifi_arduino_uart.overlay
CC: @chrisc11

cfriedt, 2021-09-21:
@chrisc11 - can you please take a look at this?

nandojve, 2022-01-30:
Hi @chrisc11,

May you please take a look at this?

nandojve, 2022-02-27:
Hi @bbilas ,

I was wondering if you did you saw this when looking at other PR?

bbilas, 2022-02-28:
Hi @nandojve, I have not seen it but I'm afraid that I won't be able to help with that due to lack of the necessary hardware.

erwango, 2022-05-02:
@chrisc11 Would you mind having a look to this point which relates to #35815 that you pushed?

erwango, 2022-05-03:
@nandojve Is this issue still valid?

nandojve, 2022-05-04:
> @nandojve Is this issue still valid?

Hi @erwango,

Last time I tested was in January and currently I don't have the shield to re-test. I 'll have access to this shield again only in August. It may be good keep this open until a next check once it is a regression.

fabiobaltieri, 2022-09-08:
@nandojve any update? Hard to justify keeping this as medium after almost 1y.

nandojve, 2022-09-10:
I still have a problem with the esWIFI driver. My feeling is that some kind of leak exists because after few hours changing the configs and memory I put it to ran the app. The other thing that I noted is that problem happen after socket be closed.

I'll close this issue because I'm not saw BUS FAULT anymore and app run with more memory for a little more time.

```
uart:~$ *** Booting Zephyr OS build zephyr-v3.1.0-5084-g11b1ef102541  ***
[00:00:00.608,000] <inf> tagoio_http_post: TagoIO IoT - HTTP Client - Temperature demo
[00:00:00.608,000] <inf> tagoio_http_post: WIFI try connecting to <AP>...
[00:00:06.314,000] <inf> tagoio_http_post: WIFI Connected
[00:00:06.316,000] <inf> tagoio_http_post: Temp: 89
[00:00:17.498,000] <err> os: ***** USAGE FAULT *****
[00:00:17.498,000] <err> os:   Illegal use of the EPSR
[00:00:17.498,000] <err> os: r0/a1:  0x204092b8  r1/a2:  0xe000ed00  r2/a3:  0x204092b8
[00:00:17.498,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x20408340 r14/lr:  0x00412ab9
[00:00:17.498,000] <err> os:  xpsr:  0x60000000
[00:00:17.498,000] <err> os: Faulting instruction address (r15/pc): 0x00000000
[00:00:17.498,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:17.498,000] <err> os: Current thread: 0x204089c8 (unknown)
[00:00:17.585,000] <err> os: Halting system
uart:~$ 
```

closed 2022-09-10 by nandojve