zephyrproject-rtos/zephyr #38552: stm32: g0b1: garbage output in log and hard fault when configuring modem

ycsin, 2021-09-15:
**Describe the bug**
This issue documents what happens after applying #38521, and make it easier to differentiate from #38519.

I'm observing a system hang (which is usually a hard fault but Zephyr didn't manage to print the message to the log) when the MCU attempts to configure the modem.

My setup is a STM32G0B1RET custom board with Quectel EC21 modem using the `gsm_ppp` modem driver. As the STM32's UART driver doesn't have the ability to wake the modem from LPM, it is necessary to constraint the MCU from entering deep sleep.

**To Reproduce**

I'm building my own application with the `gsm_ppp` set to autostart and mux-enabled. In my main I have `pm_constraint_set(PM_STATE_SUSPEND_TO_IDLE);` so that the MCU doesn't enter deep sleep.

**Expected behavior**
The application and the modem should work normally

**Impact**
showstopper

**Logs and console output**
```
I: Starting bootloader
I: Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
I: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: primary slot
I: Swap type: none
I: Bootloader chainload address offset: 0x10000
I: Jumping to the first image slot

ø1;3òmgmoc:~F¶´­û***** delaying boot 2000ms (per build configuration) *****

Build time: Sep 15 2021 17:03:18 Malaysia!
```

Output of the modem's TX line:
```
RDY
AT
OK
```

**Environment (please complete the following information):**
 - win10
 - gnuarmemb
 - f83dc928f6

**Additional context**
My `main.c`:

```c
void main(void)
{
	int i = 1;
	printk("\r\n\r\nBuild time: " __DATE__ " " __TIME__
		   " Malaysia!\r\n\r\n\r\n\r\n");

	while (true)
	{

		k_sleep(K_SECONDS(40));
	}
...
```

ycsin, 2021-09-15:
@erwango I've simplified my setup quite a bit so my hardware manager implementation isn't related anymore.

erwango, 2021-09-15:
@ycsin can you provide status with https://github.com/zephyrproject-rtos/zephyr/pull/38432 ?

ycsin, 2021-09-15:
> @ycsin can you provide status with #38432 ?

Status with #38432 is in my previous issue #38519

erwango, 2021-09-22:
@ycsin, I'm closing this point in light of the change merged yesterday. Don't hesitate to raise a new point with the latest status.

closed 2021-09-22 by erwango