zephyrproject-rtos/zephyr #38560: log v2 with 64-bit integers and threads causes invalid 64-bit value output

lairdjm, 2021-09-15:
**Describe the bug**
We are designing a product based around an nRF52840 module using NCS 1.6.1 which uses Zephyr 2.6, for this we are using the log output function and spawn a few threads for the application to run. To get 64-bit value outputting, we are using the v2 logging system in deferred mode but it seems that in this application, when we print a variable containing the timestamp, we get a large number output that is garbage, it's being printed from the shell with this code:
```
static int ats_test_cmd(const struct shell *shell, size_t argc, char **argv)
{
        int64_t testval = k_uptime_get();
        LOG_ERR("Uptime: %" PRIi64, testval);
        return 0;
}
```
And the outputs it gives:
```
uart:~$ attr test 
[00:00:05.954,925] <err> attr: Uptime: 2891370540558188544
uart:~$ attr test
[00:00:07.442,932] <err> attr: Uptime: 25769803776
uart:~$ attr test
[00:00:08.914,886] <err> attr: Uptime: 226374841270272
uart:~$ attr test
[00:00:10.307,067] <err> attr: Uptime: 106377749987328
uart:~$ attr test
[00:00:11.667,053] <err> attr: Uptime: 0
uart:~$ 
```

By disabling parts of our code that run threads, or for example a GPIO callback in one instance, the output from this can be reasonable, e.g.:
```uart:~$ attr test
[00:00:03.806,671] <err> attr: Uptime: 3806
uart:~$ attr test
[00:00:05.406,646] <err> attr: Uptime: 5406
uart:~$ attr test
[00:00:07.086,669] <err> attr: Uptime: 7086
uart:~$ attr test
[00:00:10.126,861] <err> attr: Uptime: 10126
```

I can't really work out what would be the cause of this, this is ONLY an issue with 64-bit values, 32-bit values output fine, as this shows with test2 using `k_uptime_get_32()`:
```
uart:~$ attr test2
[00:00:14.051,910] <err> attr: Uptime: 14051
uart:~$ attr test2
[00:00:15.491,973] <err> attr: Uptime: 15491
uart:~$ attr test2
[00:00:16.932,067] <err> attr: Uptime: 16932
uart:~$ attr test
[00:00:23.540,222] <err> attr: Uptime: 17179869184
```

This only seems to happen with v2 logging on deferred mode, switching to synchronous mode fixes it so I've got a suspicion that there is possibly a context switch happening after the data has been passed to the log, and that is then being trashed before it gets output at which point the reading seen is garbage. I don't believe this to be an issue in the nRF connect SDK hence why I'm posting the issue here, I've tried switching out the nordic parts e.g. bluetooth stack to use the native zephyr one and I see the same issue

**To Reproduce**
I've been unable to reproduce it with the shell sample application, it seems to happen in our application which makes extensive use of threads

**Expected behaviour**
For 64-bit integer output to work

**Impact**
Showstopper, if we can't trust the shell output, we can't trust the readings and if we can't trust the readings, we can't use the firmware or product

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.13.0
 - Commit SHA or Version used: It's using NCS 1.6.1 but that is built upon Zephyr 2.6

carlescufi, 2021-09-15:
Could this be related to https://github.com/zephyrproject-rtos/zephyr/issues/38268?

lairdjm, 2021-09-15:
@carlescufi It could be, we have an issue with packets just randomly dropping also but thought it was unrelated, even though the buffer and stack sizes are set to be pretty large, outputting 4 messages at roughly the same time generally results in seeing 1-3 messages although most times 2 are shown and 2 are what appears to be silently dropped.

nordic-krch, 2021-09-15:
That is interesting as timestamp, source id, level seems correct so it is not that the message is corrupted but only the parameter. Can you try to add some 32 bit value to that log msg? Something like:
```
LOG_ERR("%d Uptime: %" PRIi64 " %d", -1, testval, -1);

nordic-krch, 2021-09-15:
Please also make sure that stack guard is enabled.

lairdjm, 2021-09-15:
I think the issue was resolved in 2.7 with this commit: https://github.com/zephyrproject-rtos/zephyr/commit/d2f48ac35dada251b88a103960b49350e9a69209

I've manually pulled in that and 3 other changes and the 64-bit values now seem to be coming out fine. Stack guard is enabled as is canaries and null referencing checks

carlescufi, 2021-09-15:
> I think the issue was resolved in 2.7 with this commit: [d2f48ac](https://github.com/zephyrproject-rtos/zephyr/commit/d2f48ac35dada251b88a103960b49350e9a69209)
> 
> I've manually pulled in that and 3 other changes and the 64-bit values now seem to be coming out fine. Stack guard is enabled as is canaries and null referencing checks

OK, thanks for the info. Closing this then.

closed 2021-09-15 by carlescufi