zephyrproject-rtos/zephyr #38563: ISO broadcast cannot send if CONFIG_BT_CONN=n

Thalley, 2021-09-15:
**Describe the bug**
https://github.com/zephyrproject-rtos/zephyr/pull/38194 introduced a bug for ISO broadcast if CONFIG_BT_CONN=n. 
If CONFIG_BT_CONN=n then the `free_tx` FIFO is never intiialized, causes the FIFO to be empty and thus no able to use the `bt_conn_send_cb` function that `bt_iso_chan_send` now relies on. 

**To Reproduce**
Steps to reproduce the behavior:
Build and run `samples/bluetooth/iso_broadcast/` and see the `<err> bt_conn.bt_conn_send_cb: Unable to allocate TX context` error

**Expected behavior**
Expect broadcast ISO to be fully independent on CONFIG_BT_CONN. 

**Impact**
Showstopper for specific builds. Increased memory usage as a work around. 

**Logs and console output**
N/A

**Environment (please complete the following information):**
 - Commit SHA or Version used: cd5357ca56c2629b957a3a79a3dd71fe528e2626

**Additional context**
Add any other context about the problem here.

carlescufi, 2021-09-16:
@Thalley will you send a fix in time for 2.7.0?

Thalley, 2021-09-16:
> @Thalley will you send a fix in time for 2.7.0?

That's the plan :)

Thalley, 2021-09-16:
Updated the priority to low as the callback is just missing, but it is still sending data.

PR #38614 (2021-09-18): Bluetooth: ISO: Add ISO tx callback struct array
Add an array similar to the bt_conn (ACL/L2CAP)
tx sent callback, and initialize it.

This increases the number of bt_conn_tx available
such that ISO does not take any of "L2CAP's" buffers,
but also ensures that the sent callback is called
for a broadcast iso only build.

fixes #38563
closed 2021-09-18 by nashif