zephyrproject-rtos/zephyr #38565: Process: backlog handling

mbolivar-nordic, 2021-09-15:
Items under this issue:

- review backlog handling
- additional processes for unreviewed PRs, enhancements, RFCs, feature requests

Previous discussion:

https://docs.google.com/document/d/1LOhgronKx3TxUypYLy9nex0_3UC58FMMjbXV4wbiuiE/edit#

henrikbrixandersen, 2022-11-16:
I might make sense to include the "Hardware Support" label in the proposed new stale bot automation.

mbolivar-nordic, 2022-11-16:
Process WG:

- discussed proposal from @henrikbrixandersen 
- @nashif since I introduced the stale bot, I can give my perspective. I am fine with this as long as we have a process for triaging and dealing with these enhancements. If we don't want this, we can just remove 'enhancements' altogether.
- @henrikbrixandersen I generally agree but I don't think we triage PRs, do we? We triage bugs, but not PRs, and those are handled by the stale bot as well
- @nashif but pull requests get reviews
- @henrikbrixandersen hopefully they do, but if they don't, they become stale and they get closed
- @nashif yes, but in principle we have maintainers assigned for pull requests. Some things get missed, but this is where most of the work is happening, and we cover a high percentage of them. Enhancements, not really.
- @henrikbrixandersen that's fair; I tried to come up with a filter in github for open issues that only have the enhancement label and nothing else, but I'm not sure it's possible. I'm going through the incoming enhancements once a week and trying to put labels on them. I'd like to have a list of open issues with only 'enhancement' as the label, so when a maintainer looks for open issues in their area, it won't show up
- @nashif I noticed that and really appreciate it. One thing we can do is add this to the dashboard website I recently announced. I can add that really fast and we can track this.
- @henrikbrixandersen perhaps a more conservative approach would be to use another label instead of having these closed by automation. We could perhaps label enhancements that haven't been updated for a given amount of time, making it easier to triage.
- @stephanosio what's the point of relabeling 'old enhancement'? the issue count will remain high; I thought that was the main concern at the release engineering meeting.
- @henrikbrixandersen If we had an extra label, it would make it easier for a board to go through them and see if they are still relevant
- @nashif can we use projects for this? The main point is that we need a process for going through them. For example, I see issues opened in 2016, e.g. "better cache APIs needed", and Carlo is working on it right now, so something is being done after years. We need to make sure that if an enhancement is open, something is being done about it.
- @mbolivar-nordic so are you saying we are closing enhancements with no assignee automatically?
- @nashif no, after they're reviewed. And it doesn't have to be no assignee, it could be "waiting for response" for too long. And of course also allow for reopening; we aren't deleting anything
- @stephanosio I think we do need automation for closing issues, even if we have a review committee
- @nashif there is an initial effort needed to clean this up, but then we can maintain them more easily afterwards
- @stephanosio so let's say we have 500 issues in the stale queue; going through them would take a long time
- @nashif stale is the least of my concerns. I want to go through enhancements and just close the irrelevant ones, not having 'stale' just to be polite. I want a process where if I'm asking for something and we put it into a 'waiting for response' state that never gets answered, we can close them.
- @mbolivar-nordic but 'stale' is not just to be polite, it's to help automate the initial effort, and people can just reopen
- @nashif but then some issues will be closed and nobody will look at them
- @mbolivar-nordic that's fine!
- @nashif I don't think that's fine; I can come up with some examples. I think it's essential to do this; there is a lot of useful information in the enhancements that will get lost if we just close them, since people don't search through closed issues typically.
- @keith-zephyr I think I agree with @nashif that there needs to be a process for triaging before dismissing as stale. It's not a good signal to send to new users or contributors that their issues will get closed without being looked at.
- @henrikbrixandersen I don't think it's feasible to go through some 900 feature requests, even if we distribute the effort among TSC members or any other group. Not sure how to proceed
- @stephanosio my concern is how we decide if enhancements are relevant
- @nashif you can assign the enhancements to maintainers 
- @stephanosio we are missing maintainers for a lot of drivers and subsystems, and they don't get addressed
- @mbolivar-nordic should we move this back to "todo", then?
- @nashif no, please leave it "in progress". This has been on my mind and I'll work with @henrikbrixandersen on this. I am not agains the 'stale' issue, but we need it to be part of something bigger.
- @mbolivar-nordic let's leave it here then

mbolivar-nordic, 2022-11-23:
Process WG:

- @nashif looking to develop a proposal on where we're stuck with backlog issues. We discussed this in the context of enhancements, but we have other categories of issue that are suffering from the same problem (RFCs, feature requests, ...). Not a new issue; will post a link to past discussions about this on the TSC mailing list. Need to figure out how to clean this up. I managed to close 50 enhancements. Definitely need a triaging process which is triggered when there is a new enhancement (or RFC, ...) filed. Need to be more aggressive about closing stale issues or issues nobody will look at. Need to discuss with individual maintainers where these items are being filed. There are also a few historical posters of many enhancements that are still open and need to be triaged. It's also been observed that the triage meeting is refiling some bugs as enhancements. Bluetooth engineers have also been using upstream zephyr enhancements to manage their team deliverables, which is fine, but not how we are expecting enhancements to be managed, as the enhancement issues themselves do not contain any content in the description besides the template text. There are over 100 such items tracked in a private project; this is something we need to avoid. Probably we need to provide large teams or areas a dedicated GitHub project where managing ongoing work can be done in their own space without interfering with the overall project enhancement request issue list. Need to come up with a more detailed proposal based on these observations.
- @mbolivar-nordic put this in "to do", or ready for next week?
- @nashif maybe by next week, but let's leave this in "in progress" and check in next week
- @keith-zephyr idea of addressing the backlog by triaging into individual subsystems might be a way to break it down to avoid having it all fall on one small team. We can have an initial effort to split it up into individual areas, and ask the individual maintainers to refine further.
- @nashif agreed; I started doing that. Am still looking at that. Maybe don't want to create a project for each label -- that would be too much -- but instead graph the areas that have the most areas and put those into a project to be managed by the maintainers of that area. There's potential for automation if we limit the number of labels applied to each issue.

marc-hb, 2022-11-30:
> there is a lot of useful information in the enhancements that will get lost if we just close them, since people don't search through closed issues typically.

"Closed" can have and often has two conflicting definitions:
1. closed = not active right now but could be re-opened. The "project management" definition.
2. closed = done or invalid, should never be re-opened. The "engineering" definition.

Definition 1. can have a "stale bot"; definition 2. can not.

Either definition can work - _but not at the same time in the same project_. The entire project must agree on just one definition (not on some complex, grey area in the middle) otherwise discussions like the one above never end. Been there, done that.

Once one definition is agreed and clearly advertised then people can adjust their search queries. For instance when closed = "not active" then engineers must search bugs closed as stale too. When closed = "done or invalid" then project managers must use some other label or state to track work in progress and deadlines.

Other states and transitions vary across different projects, so people expect them to be just a matter of convention. However every project has a "closed" state, so many people assume that the "closed" is somehow "universal". Fail: "closed" is also a convention.

> as the enhancement issues themselves do not contain any content in the description besides the template text. 

yuck

henrikbrixandersen, 2022-11-30:
GitHub supports two reasons for closing an issue, "completed" or "not planned":
[image]

They list both "won't fix" and "stale" as reasons for closing as "not planned", meaning we cannot use the reason as a search criteria when searching for stale issues/bugs. We could, however, search for issues closed as "not planned" + label = "stale" (definition 1 above) to find these and distinguish them from issues closed as "not planned" without the "stale" label (definition 2 above).

marc-hb, 2022-11-30:
"wontfix" decisions can change and wontfix issues can be re-opened. Again it's all about conventions but most of the time I see "wontfix" used as "looks _valid_ but too hard, closing for now". Granted: less likely to be re-opened than "stale" but still possible.

nashif, 2022-12-07:
summary of proposed solutions:

https://docs.google.com/document/d/1lmW48a6-2Wx-JIroga6tuy7uK27ivjJSU36nNjdaEGI/edit?usp=sharing

mbolivar-nordic, 2022-12-07:
Process WG:

- @nashif I think there's room for a stale bot. But my proposal is that we start with a triage process, and also start actively dealing with things that are coming in while we're cleaning the old stuff. Basic idea is to use GitHub projects with automating entry of issues into the project, identify a forum for where we'll triage them, and have a big triage process where we close invalid issues or assign them to relevant maintainers. I think the most important point is to make this part of the maintainer responsibility. It's also not widely understood that assignment is not always responsibility: you can either close it or assign it to someone else. This has to be the responsibility of maintainers or it won't scale.
- @mbolivar-nordic any thoughts on the appropriate forum?
- @nashif maybe need a separate triage meeting. Not something I want to push into existing forums, since they are already very busy. In my opinion we need to decouple this from meetings, and have responsible people. Can have a person do an initial "trivial", "no", "escalate to wider forum". We need to be able to deal with this without meetings, and finding volunteers of people to deal with this. Worst case scenario, 1 out of 20 enhancements go to a dedicated meeting.
- @mbolivar-nordic would the responsible people be permanent or on rotation?
- @nashif rotation. Hopefully this can be used as a model for getting more people involved from the TSC and the community at large.

mbolivar-nordic, 2022-12-14:
Process WG: postpone as @nashif was not present

mbolivar-nordic, 2023-01-11:
Process WG: skipped due to lack of time

mbolivar-nordic, 2023-01-18:
Process WG: we'll keep this on the agenda for next week; @nashif to take a look offline

mbolivar-nordic, 2023-01-25:
Process WG: @nashif to come up with a proposed workflow for how to perform this triage.

- @nashif not clear to me what next steps are; need to discuss and agree on the triage process
- @mbolivar-nordic can you outline your proposed solution for this in the google doc?
- @nashif Sure - we identified the problem and proposed solution for stale bot. But we need to figure out a triage process for all types of issues, not just enhancements. It's also features, RFCs, etc. This is also similar to pull requests which aren't getting reviews, which is part of another issue we're discussing. We need to identify a forum for enhancements -- we have various working groups already, but I'd like to figure out a way to triage other types of issues offline if possible. The goal would be to identify assignees to respond to things in a triage process without leaving them open for so long that the bot comes and closes without any comments at all from the maintainer. I also would like to dedicate time in a meeting like dev-review for discussion and triage. 
- @mbolivar-nordic so is it a good summary to start with an offline triage process along with a meeting where we discuss things?
- @nashif yes -- offline triage process should decide if items need further discussion, and if so, in what forum (WG or TSC, etc.). Probably something like dev-review's format where we have a small number of items to close on things, but not to drive the entire process.
- @galak I think that makes sense for a certain category. For RFCs that makes sense. For enhancements it doesn't necessarily make sense. If I think about an enhancement, either the maintainer is willing to take it on, or there's no resource to take it on, and it should just go away. So I think we should discuss the flow and what makes sense for each of those categories (enhancement, RFC, ...). I think a lot of things are stale because there are no resources -- I think those are reasonable to close them. I don't think it's reasonable to discuss them if nobody's willing to work on it. This is different from an RFC where someone is willing to work on it.
- @nashif in my opinion, where we fail is not closing things where the maintainer doesn't think the enhancement is good for zephyr. For ideas that are good but nobody has time to work on it, maybe we can mark it in some way, like "good first issue", or "Google Summer of Code", etc. If that tagging happens and nobody shows up, we can let the bot close it.
- @mbolivar-nordic I am curious on what your thoughts on the concrete workflow would be for the initial triage
- @nashif I can come up with a proposed workflow for that
- @galak I do think it would be good to see a state machine workflow; that'd give us some more to talk about in terms of how things flow through
- @nashif OK, I'll propose a workflow for both triage of existing issues and ongoing triage of newly filed issues

mbolivar-nordic, 2023-02-08:
Process WG: deferred one last time. Will put back to "on hold" next week if no discussion.

mbolivar-nordic, 2023-02-08:
Process WG:

Continuing discussion of https://github.com/zephyrproject-rtos/zephyr/pull/52731. General request to please review. General discussion of timelines seems to be converging on consensus.

- @keith-zephyr last time we discussed PR escalation in detail. Latest update to my PR documents dev-review meeting's work on this. I moved escalation language to a new "PR Review Escalation" section. Not sure what the process is going to be in dev-review since it's only happened once, but to summarize: identify PRs without assignee, then identify PRs without comments or reviews, then find otherwise stale PRs that need reviewer approval. Welcome comments.
- @mbolivar-nordic anything potentially contentious for today?
- @keith-zephyr I did put some timelines in for what individual authors can do to escalate: ping reviewer after a few days, ping pr-help on discord after a week, then add dev-review after 2 weeks.
- @nashif sounds reasonable to me; hopefully we'll start catching some of these earlier through our work in dev-review.
- @fabiobaltieri I think a few days sounds a bit tight. I would say 1 week for pinging the author.
- @nashif maybe put concrete numbers of days rather than "few days"
- @keith-zephyr one additional question is how long to wait to ping after sending an update to address a review?
- @fabiobaltieri depends on the change; sometimes it can be as long as the initial review
- @galak I think it's best to keep the rules consistent as the initial review to avoid the decision tree getting more complex
- @keith-zephyr for initial ping -- 3 days, 4 days?
- @nashif well, 1 or 2 days is not realistic. We have hundreds of PRs coming in. Getting things reviewed and merged in 1 or 2 days is not always possible. I would say a week, which is 5 days.
- @galak so 1 week for a ping, 2 weeks for pr-help, 3 weeks for dev-review?
- @keith-zephyr would people object to 1 week for the ping, 2 weeks for pr-help or dev-review?
- (no objections)
- @galak I would also note that having someone ping me on discord is much more effective. I get so much noise on GitHub that I can't always tell when someone pings me.
- @keith-zephyr I was reluctant to encourage people to ping individual reviewers on discord. 
- @mbolivar-nordic this could also threaten to make discord noisy as well
- @galak I'm suggesting it as an "OR" -- not to remove pinging on the PR, but I just know that that's effective and people do that
- @nashif I have something to add here -- we do have a developer advocate here too, who works for the developers. This is something he needs to be monitoring as well: reviewing and improving the developer experience. He may need to be the front-end for people who are unhappy with the process. We need to get his feedback (@kartben)
- @galak do we want to document dev-review already, or is that still a "trial balloon"?
- @keith-zephyr I did want to include it so developers are aware
- @galak sure, that makes sense, just wondering if the triage activity is something we're committing to keeping going
- @nashif I would do it one more time before we document it
- @MaureenHelm I'm fine either way; I appreciate Keith took a stab at it, recognizing we may tweak it
- @galak is there a process for making issues for changes we're making? Does @mbolivar-nordic want to make a process issue for this?
- @mbolivar-nordic I'm fine with that, but it's up to @MaureenHelm since she's running the meeting
- @MaureenHelm I'm inclined to keep it in the minutes for now
- @galak that's fine, but I'm interested in knowing how we're keeping track of things like asking collaborators for reviews on PRs if there's no maintainer -- how are we going to follow through on those?
- @MaureenHelm hadn't thought about that yet
- @nashif this may be too much to ask -- it's a lot of responsibility and we may not want people to have to be asked for every change on something they're just a collaborator on?
- @galak that's the exact discussion I want to be having somewhere -- how do we take the patterns we see in dev-review to this meeting?
- @MaureenHelm I can take the action to write up process changes from dev-review and get them brought to this meeting, but I don't think we should have a running issue. The minutes capture that.
- @nashif an additional thing I want to bring up is that we should ask maintainers to report on the state of their system and where they are taking it. "This is what's happening, these are my challenges (I need collaborators/help), these are the things I'm focusing on, ...". If someone doesn't want to do this, ok, you're not a maintainer. You can also say "I'm doing well, no problems, etc.". Trying to figure out the format for this. I don't want to wait 2 years to remove a maintainer who is completely inactive. I'm not trying to be intrusive, but this would give us a way to check in on the activity status of maintainers. Just wanted to throw this in to have people think about it as we try this out.

keith-zephyr, 2024-01-17:
The backlog of issues and and RFCs probably needs another look by the Process WG.  
* [1644 open issues](https://github.com/zephyrproject-rtos/zephyr/issues?q=is%3Aissue+is%3Aopen+-label%3Arfc)
* [205 open RFCs](https://github.com/zephyrproject-rtos/zephyr/issues?q=is%3Aissue+is%3Aopen+label%3Arfc+)

keith-zephyr, 2024-02-14:
Process WG summary
@keith-zephyr to submit PR to cleanup Feature Tracking
@nashif to create template for projects

Propose to have Arch WG manage the unassigned feature requests.

-------------
Nag bot seems to have some success.
@nashif created script to go through unassigned issues.  Started at +600 unassigned, now down to ~150
@nashif Lots of RFCs still open, but was able to auto-triage most.  For issues that were not triaged added a "Needs review" label
@keith-zephyr bot encourages owner to add context to the issue.  But we're not currently auto closing issues.
@nashif - if we decide to add auto-close, the bot will need to mark stale and send warnings before final close.  This is a good goal to get to, but we're not ready for this.
@keith-zephyr - is it worthwhile to close really old issues
@nashif - Wants to clarify enhancement requests and process first.   Need guidelines so that enhancements are managed. 
Some requests (support a specific board or driver) are mostly noise.  The requester should talk to the vendor or write the driver themselves.
@fabiobaltieri - can we clarify in the template what the maintainers are responsible for?  Discourage users from requesting new drivers as if this was a commercial product.
@kartben - How do manage wishlists?  This does provide an opportunity for engagement for new users, and provides insights about what users want.
@keith-zephyr - what about a label for wishlist items?  @nashif we used to have a "hardware-support" label, but this sent the wrong message.
@nashif - for SoCs and areas that use Github for managing backlog, wishlist items could be useful, but low priority.  The maintainer should have the power to close issues that won't be supported/added.
@dleach02 - many requests come in through other channels
@dleach02  - NXP generally doesn't use github, org uses Jira.  
@nashif - need to adjust language on https://app.codecov.io/gh/zephyrproject-rtos/zephyr/blob/main/include%2Fzephyr%2Fdrivers%2Fi2c.h#L1041
@nashif - this is to large to manage at the Process WG.  Feature request need to move to the appropriate group and then filter to the maintainer.
@nashif - need to set the expectations for feature requests.
@nashif - maintainers currently hesitate to close enhancement requests because we're missing a process.
@dleach02 - would it be reasonable to create project areas for vendor specific items?  @nashif yes this can work - if vendors are willing this helps Zephyr
@nashif we can create a project template for vendors.  
@keith-zephyr - we need more projects created to manage the feature request
@nashif - 1st step, simplify the Feature Tracking page.  Get rid of hardware support.
@keith-zephyr - can we combine enhancement and feature request.  @nashif this could work.
@nashif - generally people don't read the documentation, so simplifying makes sense.
@keith-zephyr suggest to simplify to "enhancment".  This includes both small incremental changes and requests for a new subsystem.
@nashif - Feature tracking can be simplified to "enhancement", "RFC", and "bug".
@kartben - Who owns the process? @keith-zephyr - this would be the project area, but we need to define more projects to cover the functional areas.
@keith-zephyr - to make a first pass at cleaning up Feature tracking page and to remove the feature request from the the Issue template
@nashif - it would be helpful for template to point back to the documentation
@nashif - will work on creating a project template for managing enhancements
@nashif - There is still an open issue for dealing with unassigned issues.  Bot cannot assign to areas without a maintainer.
@keith-zephyr - could Dev review triage the unassigned issues?  @nashif - doesn't like the idea, Dev review is overloaded with responsibilities
@nashif Arch WG could take this up at end of meeting once the backlog is dealt with.  This is also the correct audience.
@nashif - Need to clarify improve language for the responsibility of the issue assignee.