zephyrproject-rtos/zephyr #38570: Process: binary blobs in Zephyr

mbolivar-nordic, 2021-09-15:
This tracks ongoing discussions around the distribution or inclusion of binary blobs related to Zephyr development.

There is a live document describing the problem and possible policy outcomes:

https://docs.google.com/document/d/16QU8OT4pnKuhszZHIYfsj1WInN-LRXjPdt87hxG5IRU/edit#

mbolivar-nordic, 2021-09-15:
Process WG:

- distributing binary blobs in the default distribution (YY path through the problem statement's decision diagram) is explicitly rejected due to lack of backers and doubtful acceptance by the board
- @mbolivar-nordic to provide a detailed proposal for what it would look like for blobs to be totally out of scope (N path)
- remaining WG time on this issue will be devoted to rejecting/accepting particular policies for mechanisms that users can use to fetch blobs (YN path)

The N and YN paths will be presented as a single up-or-down vote (to approve the YN path or reject it and therefore implicitly accept the N path) to the TSC.

mbolivar-nordic, 2021-09-22:
Process WG:

- additional decisions were taken in the live document linked to in the issue description

mbolivar-nordic, 2021-11-17:
Process WG:

- No consensus on finding a maintainer before getting a vote, agree to continue discussion, but also involve Espressif and SiLabs with a view towards getting them involved

closed 2022-02-23 by mbolivar-nordic

marc-hb, 2023-01-02:
First commit adding a new `west blob` command: 336aa9dc88
First documentation commit 2c301fe22f53468