zephyrproject-rtos/zephyr #38578: STM32L0X ADC hangs

rlezuo1, 2021-09-15:
**Describe the bug**
ADC conversions on STM32L0x (i.e. L051) hang forever

**To Reproduce**
Steps to reproduce the behavior:
1. adc_channel_setup(...)
2. adc_read(...)
3. will hang forever in:LL_ADC_IsCalibrationOnGoing or adc_context_wait_for_completion depending on struct adc_sequence calibrate setting

**Expected behavior**
Will not hang the ADC unit

**Impact**
ADC unit is trashed, system becomes usable

**Additional context**

I've identified two issues
1) struct adc_sequence member calibrate must be set to 0, this avoids a second calibration in start_read() which must not be trigged when ADC is enabled (according to device RM). Zephyr's driver has the ADC enabled here.

2) similar issue with call to LL_ADC_SetOverSamplingScope in start_read, which *seems* to touch CKMODE bits and trash the unit (although debugger ADC register read-out may be broken at that point already), the register can be written when disabling the adc (see attached patch).
[adc.patch.txt](https://github.com/zephyrproject-rtos/zephyr/files/7173037/adc.patch.txt)

3) when not calibrating and applying the patch the ADC unit is not trashed, meaningful conversions are performed

FRASTM, 2021-09-16:
@rlezuo1 , I take the _samples/drivers/adc_ application as an example, because we see the sequence  adc_channel_setup() then adc_read() that you are describing as faulty. 
For 1st issue, do you mean that **sequence.calibrate = false**, is required in the adc_sequence declaration ?
```
struct adc_sequence sequence = {
	/* individual channels will be added below */
	.channels    = 0,
	.buffer      = sample_buffer,
	/* buffer size in bytes, not number of samples */
	.buffer_size = sizeof(sample_buffer),
	.resolution  = ADC_RESOLUTION,
	.calibrate = false,
};
```

rlezuo1, 2021-09-16:
@FRASTM , exactly

FRASTM, 2021-09-16:
For #2, I see that all STM32 series that have an ADC oversampling (OVSE bit) and an oversampling ratio (OVS bits), the setting  in the ADC CFGR2 "is allowed to write this bit only when ADSTART = 0 (which ensures that no conversion is ongoing). That's why I propose to have 
```
	LL_ADC_Disable(adc);
	while (LL_ADC_IsEnabled(adc) == 1UL) {
	}
```
**before**  `switch (sequence->oversampling)`. 
Then set the LL_ADC_SetOverSamplingScope  and LL_ADC_ConfigOverSamplingRatioShift
Then `LL_ADC_Enable(adc) `afterwards.

- for the same STM32 series, changing  the ADC resolution  "is allowed to write these bits only when ADEN = 0". That's why I also propose to disable(adc) then change the LL_ADC_SetResolution then enable(adc).

FRASTM, 2021-09-16:
@rlezuo1, I put 2 different PRs, if you could take both together and check on your own target board.

rlezuo1, 2021-09-16:
there is an issue with the PRs (they do not compile), I'll wait for you to update and then try again?

rlezuo1, 2021-09-16:
with the fixed PR's I can confirm ADC is working as expected

54c35b61da (HEAD -> adc_prs) Merge remote-tracking branch 'github_FRASTM/adc_sequence_init' into adc_prs
df5e0edbbb (github_FRASTM/stm32adc_hangs) drivers: adc: stm32 disable the ADC before setting the Oversampling
114c7963f7 (github_FRASTM/adc_sequence_init) samples: drivers: adc initialize the adc_sequence calibration
c2ef1b3fe2 drivers: adc: stm32 disable the ADC before setting the Resolution
2e242c51fb (github_FRASTM/myrepo) soc: riscv: ite: it8xxx2: guard SoC Kconfig options
d63d828c8f stm32: Add FMAC support to Kconfig

*** Booting Zephyr OS build v2.6.0-rc1-3814-g54c35b61da87  ***
I: detect=2
^^^^ 
value 2 is read from ADC 

can be closed from my POV

closed 2021-09-18 by cfriedt