zephyrproject-rtos/zephyr #38581: tests-ci : kernel: scheduler: multiq test failed

hakehuang, 2021-09-16:
**Describe the bug**
multiq test is failed on v2.7.0-rc1-94-gf83dc928f613 on lpcxpresso55s28

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p lpcxpresso55s28  --sub-test kernel.scheduler
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**
```
@*** Booting Zephyr OS build v2.7.0-rc1-94-gf83dc928f613  ***
Running test suite threads_scheduling
===================================================================
START - test_bad_priorities
PASS - test_bad_priorities in 0.1 seconds
===================================================================
START - test_priority_cooperative
PASS - test_priority_cooperative in 0.101 seconds
===================================================================
START - test_priority_preemptible
PASS - test_priority_preemptible in 0.101 seconds
===================================================================
START - test_priority_preemptible_wait_prio
PASS - test_priority_preemptible_wait_prio in 0.32 seconds
===================================================================
START - test_yield_cooperative
PASS - test_yield_cooperative in 0.1 seconds
===================================================================
START - test_sleep_cooperative
PASS - test_sleep_cooperative in 0.101 seconds
===================================================================
START - test_busy_wait_cooperative
PASS - test_busy_wait_cooperative in 0.101 seconds
===================================================================
START - test_sleep_wakeup_preemptible
PASS - test_sleep_wakeup_preemptible in 0.2 seconds
===================================================================
START - test_pending_thread_wakeup
PASS - test_pending_thread_wakeup in 0.1 seconds
===================================================================
START - test_time_slicing_preemptible
PASS - test_time_slicing_preemptible in 0.501 seconds
===================================================================
START - test_time_slicing_disable_preemptible
PASS - test_time_slicing_disable_preemptible in 0.501 seconds
===================================================================
START - test_lock_preemptible
PASS - test_lock_preemptible in 0.201 seconds
===================================================================
START - test_unlock_preemptible
PASS - test_unlock_preemptible in 0.101 seconds
===================================================================
START - test_unlock_nested_sched_lock
PASS - test_unlock_nested_sched_lock in 0.101 seconds
===================================================================
START - test_sched_is_preempt_thread
PASS - test_sched_is_preempt_thread in 0.1 seconds
===================================================================
START - test_slice_reset
PASS - test_slice_reset in 1.802 seconds
===================================================================
[... 120 lines cut ...]
START - test_k_wakeup_init_null
E: (nil) is not a valid k_thread
E: address is not a known kernel object
E: syscall z_vrfy_k_wakeup failed check: access denied
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x20001830 (unknown)
Caught system error -- reason 3 1
Fatal error expected as part of test case.
PASS - test_k_wakeup_init_null in 0.45 seconds
===================================================================
Test suite threads_scheduling failed.
===================================================================
PROJECT EXECUTION FAILED

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: v2.7.0-rc1-94-gf83dc928f613

hakehuang, 2021-09-16:
@dleach02

mmahadevan108, 2021-10-01:
@hakehuang. Could you try the test again after the fix went in.

hakehuang, 2021-10-02:
@mmahadevan108 , sure, I will follow

closed 2021-10-02 by hakehuang