zephyrproject-rtos/zephyr #38586: olimexino_stm32: "no DEVICE_HANDLE_ENDS inserted" builld issue (samples/subsys/usb/audio/headphones_microphone)

erwango, 2021-09-16:
**Describe the bug**
Following build issue reported when compiling `samples/subsys/usb/audio/headphones_microphone` on olimexino_stm32

```
Traceback (most recent call last):
  File "/local/mcu/zephyrproject/zephyr/scripts/gen_handles.py", line 371, in <module>
    main()
  File "/local/mcu/zephyrproject/zephyr/scripts/gen_handles.py", line 341, in main
    assert len(hdls) < len(hs.handles), "%s no DEVICE_HANDLE_ENDS inserted" % (dev.sym.name,)
AssertionError: __device_dts_ord_170 no DEVICE_HANDLE_ENDS inserted

```

**To Reproduce**
twister -b -p olimexino_stm32 -s samples/subsys/usb/audio/headphones_microphone/sample.usb.audio.headphones_microphone

erwango, 2021-09-16:
Similar to https://github.com/zephyrproject-rtos/zephyr/issues/38181

nashif, 2021-09-29:
IMO this is a duplicate of #38181, lets close and track this in one place.

closed 2021-09-29 by nashif