zephyrproject-rtos/zephyr #38591: nucleo_f091rc: Linking issue since "align __data_ram/rom_start/end linker" (65a2de84a9d5c535167951bf1cf610c4f7967ea5)

erwango, 2021-09-16:
**Describe the bug**
Since merge of https://github.com/zephyrproject-rtos/zephyr/commit/65a2de84a9d5c535167951bf1cf610c4f7967ea5,
Some tests are booting on nucleo_f091rc platform.

Reverting this commit gets the board booting again

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b  nucleo_f091rc tests/kernel/lifo/lifo_usage
2. west flash -d build/nucleo_f091rc/

erwango, 2021-09-16:
@tejlmand Would you mind having a look?
(I can help in testing if needed)

erwango, 2021-09-22:
@tejlmand I've made some investigation on this point.

In 65a2de8, you mention:
> The following new symbols are introduced so that the data section is
aligned with other sections:
>  - __data_end
>  - __data_start
       value identical to __data_region_start but describes start of
       the section.

I can see this is the case on a target not impacted by the change (nucleo_f103rb).
However, on nucleo_f091rc, this is not true:
```
 *(SORT_BY_ALIGNMENT(.ramfunc.*))
                0x00000000200000bc                . = ALIGN (_region_min_align)
                0x00000000200000bc                _ramfunc_ram_end = .
                0x0000000000000000                _ramfunc_ram_size = (_ramfunc_ram_end - _ramfunc_ram_start)
                0x0000000008006ca4                _ramfunc_rom_start = LOADADDR (.ramfunc)
                0x00000000200000bc                __data_region_start = .

datas           0x00000000200000c0       0xb9 load address 0x0000000008006ca8
                0x00000000200000c0                __data_start = .
 *(SORT_BY_ALIGNMENT(.data))
 *(SORT_BY_ALIGNMENT(.data.*))
 .data.state.0  0x00000000200000c0        0x8 zephyr/libzephyr.a(heap-validate.c.obj)
```
If I'm reading thing correctly:
```
__data_region_start = 0x00000000200000bc
__data_start = 0x00000000200000c0
```
So the base assumption doesn't work.
No explanation on this fact so far, but this could be linked to the specific sram vector linker for this target: [sram_vector_table.ld](https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arm/st_stm32/stm32f0/sram_vector_table.ld).

I'm continuing the analysis, but don't hesitate to get in touch if this rings a bell to you

erwango, 2021-09-22:
@JordanYates It seems you introduced `sram_vector_table.ld` in https://github.com/zephyrproject-rtos/zephyr/pull/34362.
It seems that disabling `CONFIG_SRAM_VECTOR_TABLE` on nucleo_f091rc solves current point.

First question, since you disabled `CONFIG_SRAM_VECTOR_TABLE` on a number of F0 boards, was there any reason to keep it on `nucleo_f091rc` ?
Then, if disabling `CONFIG_SRAM_VECTOR_TABLE` get things working, it does not solve the point that this option doesn't play well with 65a2de8. Would you mind having a quick look on the issue I pointed above ?

JordanYates, 2021-09-22:
> First question, since you disabled `CONFIG_SRAM_VECTOR_TABLE` on a number of F0 boards, was there any reason to keep it on `nucleo_f091rc` ?

The PR was cleaning up usage of `CONFIG_IS_BOOTLOADER` in order to get secondary effects (no SRAM vector table).
`nucleo_f091rc` was not modified because it was not using `CONFIG_IS_BOOTLOADER` to modify the vector table in the first place.

> Then, if disabling CONFIG_SRAM_VECTOR_TABLE get things working, it does not solve the point that this option doesn't play well with 65a2de8. Would you mind having a quick look on the issue I pointed above?

As I don't use STM, I wasn't willing to introduce any behavior changes, so section names and symbols were not changed in my PR. Perhaps the act of moving the section declaration caused the issue, but that seems unlikely, and its using the standard `zephyr_linker_sources_ifdef` function, so it seems unlikely that this should cause issues. If the problem is just the naming, my PR predates the linked commit by 4 months.

erwango, 2021-09-23:
> The PR was cleaning up usage of `CONFIG_IS_BOOTLOADER` in order to get secondary effects (no SRAM vector table).
> `nucleo_f091rc` was not modified because it was not using `CONFIG_IS_BOOTLOADER` to [...]

ok

> As I don't use STM, I wasn't willing to introduce any behavior changes, so section names and symbols were not changed in my PR. Perhaps the act of moving the section declaration caused the issue, [...]

Sure, your PR was change was introduced earlier than 65a2de8.
I was hoping maybe you could have some clue on the way to make both work together.

PR #38817 (2021-09-24): linker: __data_region_start equal to __data_start
Fixes: #38591, #38207, #37861

The commit 65a2de84a9d5c535167951bf1cf610c4f7967ea5 aligned the data
linker symbol for sections and regions.

The data region symbol start has been placed outside the sections thus
being defined as the address of the region before alignment of the first
section in the data region, usually the `datas` section.

The symbol defining the start address of the data section is after
section alignment.
In most cases the address of the data region start and datas section
start will be identical, but not always.
The data re [... cut]
commit 15e834a6 (2021-09-24): linker: __data_region_start equal to __data_start

Fixes: #38591, #38207, #37861

The commit 65a2de84a9d5c535167951bf1cf610c4f7967ea5 aligned the data
linker symbol for sections and regions.

The data region symbol start has been placed outside the sections thus
being defined as the address of the region before alignment of the first
section in the data region, usually the `datas` section.

The symbol defining the start address of the data section is after
section alignment.
In most cases the address of the data region start and datas section
start will be identical, but not always.
The data r [... cut]
closed 2021-09-24 by cfriedt