zephyrproject-rtos/zephyr #38592: flash runners: JLink provides no notifications of failures

JordanYates, 2021-09-16:
**Describe the bug**
When flashing a board that uses the `JLink` runner, no notification is provided to the user if the flash command fails.

**To Reproduce**
1. Build any application for a board using the `JLink` runner
2. Run `west flash` with the board connected
3. Run `west flash` with the programmer disconnected

**Expected behavior**
Some error should be printed to the user

**Impact**
Annoyance

**Logs and console output**

One of these two commands was run with no board connected. Can you tell which?
```
C:\Users\Jordan\code\zephyr>west flash -d build\loci3\hello_world
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner jlink
-- runners.jlink: JLink version: 7.50a
-- runners.jlink: Flashing file: build\loci3\hello_world\zephyr\zephyr.hex

C:\Users\Jordan\code\zephyr>west flash -d build\loci3\hello_world
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner jlink
-- runners.jlink: JLink version: 7.50a
-- runners.jlink: Flashing file: build\loci3\hello_world\zephyr\zephyr.hex

C:\Users\Jordan\code\zephyr>
```

**Environment (please complete the following information):**
 - OS: Windows
 - Zephyr v2.7.0-rc1

carlescufi, 2021-09-16:
@JordanYates isn't this an issue with `JLink.exe` itself? I have noticed it tends to silently fail.
Could you provide the same log output but with `west -v flash ...` ?

JordanYates, 2021-09-16:
`JLink.exe` is definitely detecting the failure, it just doesn't appear to be returning an error code that `check_call` can pick up:
```
C:\Users\Jordan\code\zephyr>west -v flash -d build\loci3\blinky
ZEPHYR_BASE=C:\Users\Jordan\code\zephyr\zephyr (origin: configfile)
-- west flash: rebuilding
cmake version 3.20.0 is OK; minimum version is 3.13.1
Running CMake: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'build\loci3\blinky'
ninja: no work to do.
-- west flash: using runner jlink
runners.jlink: JLINKARM_GetDLLVersion()=75001
-- runners.jlink: JLink version: 7.50a
runners.jlink: JLink commander script:
r
loadfile build\loci3\blinky\zephyr\zephyr.hex
g
writeDP 1 0
readDP 1
q
-- runners.jlink: Flashing file: build\loci3\blinky\zephyr\zephyr.hex
runners.jlink: 'C:\Program Files\SEGGER\JLink\JLink.exe' -nogui 1 -if swd -speed 4000 -device nrf52 -CommanderScript 'C:\Users\Jordan\AppData\Local\Temp\tmpmp_mmg_mjlink\runner.jlink' -nogui 1
SEGGER J-Link Commander V7.50a (Compiled Jul  8 2021 18:22:32)
DLL version V7.50a, compiled Jul  8 2021 18:21:11

J-Link Command File read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...O.K.
Firmware: J-Link V10 compiled Jun 29 2021 16:11:46
Hardware version: V10.10
S/N: 820101337
License(s): GDB
VTref=2.380V
Target connection not established yet but required for command.
Device "NRF52" selected.

Connecting to target via SWD
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
Cannot connect to target.

Target connection not established yet but required for command.
Device "NRF52" selected.

Connecting to target via SWD
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
Cannot connect to target.

Target connection not established yet but required for command.
Device "NRF52" selected.

Connecting to target via SWD
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
InitTarget() start
InitTarget() end
Cannot connect to target.

Writing failed.

Reading failed.

Script processing completed.

C:\Users\Jordan\code\zephyr>
```

JordanYates, 2021-09-16:
Just validated that it is still an issue on the latest Jlink (7.54b) as well.

aunsbjerg, 2021-09-16:
In https://wiki.segger.com/J-Link_Commander under the "JLink.exe return value" section, it's noted that JLinkExe will set the ERRORLEVEL environment variable depending on the outcome of the jlink operations. Maybe that could be used instead of checking the return value?

JordanYates, 2021-09-16:
> In https://wiki.segger.com/J-Link_Commander under the "JLink.exe return value" section, it's noted that JLinkExe will set the ERRORLEVEL environment variable depending on the outcome of the jlink operations. Maybe that could be used instead of checking the return value?

I wasn't aware of this, but it doesn't appear to work.
Possibly because the command is running as a subprocess, and hence the environment is destroyed when the command finishes?

Not sure why SEGGER would think that was a better solution than just returning a value...

aunsbjerg, 2021-09-16:
> I wasn't aware of this, but it doesn't appear to work.
> Possibly because the command is running as a subprocess, and hence the environment is destroyed when the command finishes?
>
> Not sure why SEGGER [...]

You're probably right in that it's because JLinkExe is run in a subprocess.

I tried running `subprocess.check_output("jlink -invalid", shell=True)`, and it actually returns an error code. So it's possible that Jlink simply doesn't return an error if a valid script is executed but with no connected device? I can probably take a look tomorrow when I'm back at the office.

carlescufi, 2021-09-16:
> You're probably right in that it's because JLinkExe is run in a subprocess.

Funny because `shell` defaults to `False` in `check_call()`.

JordanYates, 2021-11-25:
Resolved by #39734

closed 2021-11-25 by JordanYates