zephyrproject-rtos/zephyr #38593: using RTT console to print along with newlib C library in Zephyr

AfAbdel, 2021-09-16:
Hello everyone,
I am using NrfConnect with Zephyr on Segger embedded studio .
I want tu use the math logarithm function in my code and print the result on the debug terminal. At first when enabling the "Use RTT for console" configuration, printf works, then I wanted to add the math.h log function which gave me  "Undefined reference to log" error.
To resolve the error I checked the "Newlib C library" configuration which made the log function work but then I realised the printf does not work anymore.
Printf does not throw any error but nothing is printed on the Debug terminal.
How can I use both log and printf with RTT console ?

OS: windows
Board: Nrf52DK
SDK: nrf Connect sdk v1.5.0
RTOS: Zephyr

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

carlescufi, 2021-09-16:
This is an nRF Connect SDK issue, not an upstream Zephyr one. Segger Embedded Studio and nRF Connect SDK are supported by Nordic, please open an issue in the Nordic DevZone instead.

closed 2021-09-16 by carlescufi

AfAbdel, 2021-09-20:
I posted in the Nordic DevZone and they said it was an upstream Zephyr issue
https://devzone.nordicsemi.com/f/nordic-q-a/79727/using-rtt-console-to-print-along-with-newlib-c-library-in-zephyr