zephyrproject-rtos/zephyr #38601: nucleo_f103rb: samples/posix/eventfd/ failed since "retargetable locking" addition

erwango, 2021-09-16:
**Describe the bug**
Sample `samples/posix/eventfd` is failed since https://github.com/zephyrproject-rtos/zephyr/commit/43b8002b14a7c1a38c19df19955911144c625b5e

**To Reproduce**
`twister --ninja --runtime-artifact-cleanup  -s samples/posix/eventfd/sample.posix.eventfd -p nucleo_f103rb --device-testing --device-serial /dev/ttyACM0`

**Expected behavior**
Test should pass

**Additional input**
According to bench history, it is possible that nucleo_l073rz was also impacted by same commit on same test but also few others:
```
tests/arch/arm/arm_runtime_nmi/arch.interrupt.arm.nmi
tests/benchmarks/latency_measure/benchmark.kernel.latency.stm32
tests/kernel/mem_protect/userspace/kernel.memory_protection.userspace
tests/kernel/msgq/msgq_api/kernel.message_queue
tests/kernel/mutex/mutex_api/kernel.mutex
tests/kernel/pipe/pipe/kernel.pipe
tests/kernel/pipe/pipe_api/kernel.pipe.api
tests/kernel/sleep/kernel.common.timing
tests/kernel/threads/dynamic_thread/kernel.threads.dynamic
tests/kernel/threads/thread_init/kernel.threads.init
tests/kernel/threads/tls/kernel.threads.tls.userspace
tests/kernel/timer/timer_api/kernel.timer
tests/kernel/timer/timer_api/kernel.timer.tickless
tests/lib/p4workq/lib.p4wq
tests/lib/time/libraries.libc.time
```
But this is yet to be confirmed. Will confirm as soon as I get the board.

erwango, 2021-09-16:
@stephanosio Would you mind having a look?
I'm available to do some tests if needed.

erwango, 2021-09-16:
@stephanosio There is no console output. We're getting a hard fault at the first printf occurence:
Here is an abstract
```
Breakpoint 4, 0x08008224 in _vfprintf_r ()
(gdb) si
Info : halted: PC: 0x0800a000
halted: PC: 0x0800a000
0x0800a000 in _localeconv_r ()
[....]
k_spin_unlock (key=..., l=0x20001c81) at /local/mcu/zephyrproject/zephyr/kernel/mutex.c:120
120			k_spin_unlock(&lock, key);
(gdb) 
arch_irq_unlock (key=0) at /local/mcu/zephyrproject/zephyr/include/arch/arm/aarch32/asm_inline_gcc.h:95
95		__asm__ volatile(
(gdb) 
Info : halted: PC: 0x080062a4
halted: PC: 0x080062a4
Info : halted: PC: 0x080062a8
halted: PC: 0x080062a8
z_impl_k_mutex_lock (mutex=0x20000f5c <__lock___sinit_recursive_mutex>, timeout=...)
    at /local/mcu/zephyrproject/zephyr/kernel/mutex.c:124
124			return 0;
(gdb) 
Info : halted: PC: 0x080062aa
halted: PC: 0x080062aa
Info : halted: PC: 0x080062ac
halted: PC: 0x080062ac
Info : halted: PC: 0x08009c20
halted: PC: 0x08009c20
0x08009c20 in __sinit ()
(gdb) 
Single stepping until exit from function __sinit,
which has no line number information.
Info : halted: PC: 0x08009c24
halted: PC: 0x08009c24
Info : halted: PC: 0x08009c26
halted: PC: 0x08009c26
Info : halted: PC: 0x08009c28
halted: PC: 0x08009c28
Info : halted: PC: 0x08009c2a
halted: PC: 0x08009c2a
Info : halted: PC: 0x08009c2c
halted: PC: 0x08009c2c
Info : halted: PC: 0x08009c2e
halted: PC: 0x08009c2e
Info : halted: PC: 0x08009c32
halted: PC: 0x08009c32
Info : halted: PC: 0x08009c36
halted: PC: 0x08009c36
Info : halted: PC: 0x08009c3a
halted: PC: 0x08009c3a
Info : halted: PC: 0x08009c3e
halted: PC: 0x08009c3e
Info : halted: PC: 0x08009c40
halted: PC: 0x08009c40
Info : halted: PC: 0x08009c42
halted: PC: 0x08009c42
Info : halted: PC: 0x08009c46
halted: PC: 0x08009c46
Info : halted: PC: 0x08009c4a
halted: PC: 0x08009c4a
Info : halted: PC: 0x08009c4e
halted: PC: 0x08009c4e
Info : halted: PC: 0x08009c50
halted: PC: 0x08009c50
Info : halted: PC: 0x08009c52
halted: PC: 0x08009c52
Info : halted: PC: 0x08009c54
halted: PC: 0x08009c54
Info : halted: PC: 0x08009c58
halted: PC: 0x08009c58
Info : halted: PC: 0x08009c5c
halted: PC: 0x08009c5c
Info : halted: PC: 0x08009c5e
halted: PC: 0x08009c5e
Info : halted: PC: 0x08009c60
halted: PC: 0x08009c60
Info : halted: PC: 0x08009c62
halted: PC: 0x08009c62
Info : halted: PC: 0x08009c66
halted: PC: 0x08009c66
Info : halted: PC: 0x080079cc
halted: PC: 0x080079cc
Info : halted: PC: 0x08009c6e
halted: PC: 0x08009c6e
Info : halted: PC: 0x08009c72
halted: PC: 0x08009c72
Info : halted: PC: 0x08009c74
halted: PC: 0x08009c74
Info : halted: PC: 0x08009c78
halted: PC: 0x08009c78
Info : halted: PC: 0x0800c6c8
halted: PC: 0x0800c6c8
__retarget_lock_init_recursive (lock=0x200001b0 <impure_data+432>)
    at /local/mcu/zephyrproject/zephyr/lib/libc/newlib/libc-hooks.c:374
374		*lock = malloc(sizeof(struct k_mutex));
(gdb) 
Info : halted: PC: 0x0800c6ca
halted: PC: 0x0800c6ca
Info : halted: PC: 0x0800c6cc
halted: PC: 0x0800c6cc
Info : halted: PC: 0x0800c6ce
halted: PC: 0x0800c6ce
Info : halted: PC: 0x0800720c
halted: PC: 0x0800720c
Info : halted: PC: 0x0800c6d4
halted: PC: 0x0800c6d4
708		return z_impl_k_mutex_init(mutex);
(gdb) 
Info : halted: PC: 0x0800c6d8
halted: PC: 0x0800c6d8
Info : halted: PC: 0x0800f860
halted: PC: 0x0800f860
z_impl_k_mutex_init (mutex=0x20004870) at /local/mcu/zephyrproject/zephyr/kernel/mutex.c:50
50	{
(gdb) 
Info : halted: PC: 0x0800f862
halted: PC: 0x0800f862
51		mutex->owner = NULL;
(gdb) 
Info : halted: PC: 0x0800f864
halted: PC: 0x0800f864
Info : halted: PC: 0x0800f868
halted: PC: 0x0800f868
Info : halted: PC: 0x0800f86c
halted: PC: 0x0800f86c
60		return 0;
(gdb) 
Info : halted: PC: 0x08009c7c
halted: PC: 0x08009c7c
0x08009c7c in __sinit ()
(gdb) 
Single stepping until exit from function __sinit,
which has no line number information.
Info : halted: PC: 0x08009c80
halted: PC: 0x08009c80
Info : halted: PC: 0x08009c82
halted: PC: 0x08009c82
Info : halted: PC: 0x08009c84
halted: PC: 0x08009c84
Info : halted: PC: 0x08009c86
halted: PC: 0x08009c86
Info : halted: PC: 0x08009c88
halted: PC: 0x08009c88
Info : halted: PC: 0x08009c8c
halted: PC: 0x08009c8c
Info : halted: PC: 0x08009c90
halted: PC: 0x08009c90
Info : halted: PC: 0x08009c92
halted: PC: 0x08009c92
Info : halted: PC: 0x08009c94
halted: PC: 0x08009c94
Info : halted: PC: 0x08009c96
halted: PC: 0x08009c96
Info : halted: PC: 0x08009c9a
halted: PC: 0x08009c9a
Info : halted: PC: 0x080079cc
halted: PC: 0x080079cc
Info : halted: PC: 0x08009ca2
halted: PC: 0x08009ca2
Info : halted: PC: 0x08009ca6
halted: PC: 0x08009ca6
Info : halted: PC: 0x08009ca8
halted: PC: 0x08009ca8
Info : halted: PC: 0x08009cac
halted: PC: 0x08009cac
Info : halted: PC: 0x0800c6c8
halted: PC: 0x0800c6c8
__retarget_lock_init_recursive (lock=0x20000218 <impure_data+536>)
    at /local/mcu/zephyrproject/zephyr/lib/libc/newlib/libc-hooks.c:374
374		*lock = malloc(sizeof(struct k_mutex));
(gdb) 
Info : halted: PC: 0x0800c6ca
halted: PC: 0x0800c6ca
Info : halted: PC: 0x0800c6cc
halted: PC: 0x0800c6cc
Info : halted: PC: 0x0800c6ce
halted: PC: 0x0800c6ce
Info : halted: PC: 0x0800720c
halted: PC: 0x0800720c
Info : halted: PC: 0x0800c6d4
halted: PC: 0x0800c6d4
708		return z_impl_k_mutex_init(mutex);
(gdb) 
Info : halted: PC: 0x0800c6d8
halted: PC: 0x0800c6d8
Info : halted: PC: 0x0800f860
halted: PC: 0x0800f860
z_impl_k_mutex_init (mutex=0x0) at /local/mcu/zephyrproject/zephyr/kernel/mutex.c:50
50	{
(gdb) 
Info : halted: PC: 0x0800f862
halted: PC: 0x0800f862
51		mutex->owner = NULL;
(gdb) 
Info : halted: PC: 0x0800f864
halted: PC: 0x0800f864
Info : halted: PC: 0x08003360
halted: PC: 0x08003360
z_arm_usage_fault () at /local/mcu/zephyrproject/zephyr/arch/arm/core/aarch32/cortex_m/fault_s.S:80
80		mrs r0, MSP

```

stephanosio, 2021-09-16:
@erwango Could you set `CONFIG_ASSERT=y` and check if you see any failures?

> z_impl_k_mutex_init (**mutex=0x0**) at /local/mcu/zephyrproject/zephyr/kernel/mutex.c:50

Looking at the debug log, it looks like the `malloc` in `__retarget_lock_init_recursive` failed due to insufficient SRAM.

If you enable `ASSERT`, you will likely see the following messages:
* "memory space available for newlib heap is less than the minimum required size specified by 
 CONFIG_NEWLIB_LIBC_MIN_REQUIRED_HEAP_SIZE"
* "recursive lock allocation failed"

This is because there is very little leftover SRAM, which the newlib uses for its heap, for `nucleo_l073rz` with this particular sample:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       73248 B       128 KB     55.88%
            SRAM:       18528 B        20 KB     90.47%
        IDT_LIST:          0 GB         2 KB      0.00%
```

stephanosio, 2021-09-16:
From https://github.com/zephyrproject-rtos/zephyr/issues/38044#issuecomment-914466354:

> Normally, for any statically allocated regions, twister will skip the test if there is an overflow; but here, we implicitly make use of the leftover area without the linker's knowledge, so there is no linker error and twister ends up treating this as a failure at runtime.

Due to how Zephyr-to-newlib integration is implemented, there is no easy way for the twister to filter out/skip these boards/tests -- this will be fixed eventually in the next release as noted in the comment referenced above.

For now, we should identify all the failing sample-board combos (mostly the boards with very small SRAM and the samples with very high SRAM usage) and add `platform_exclude` to them manually. If there happen to be too many though, we can alternatively consider setting the `min_ram` in the `testcase.yaml` or `sample.yaml` so that twister does not run these tests on the devices with little SRAM.

erwango, 2021-09-17:
Thanks for the help @stephanosio 

> If you enable ASSERT, you will likely see the following messages:

Unfortunately, we don't get messages even with asserts enabled and even in debugger we don't reach `__assert_func`.

> For now, we should identify all the failing sample-board combos [...]

To save some time for others, I'll use `min_ram: 32`.

PR #38625 (2021-09-20): samples/posix: eventfd: Prevent run in twister when SRAM below 32K
This test makes use of dynamic RAM allocation done on the leftover
SRAM. Similarly to test/posix/eventfd, limit this test execution on
platforms with at least 32K of RAM available.

Fixes #38601
commit 123be4f1 (2021-09-20): samples/posix: eventfd: Prevent run in twister when SRAM below 32K

This test makes use of dynamic RAM allocation done on the leftover
SRAM. Similarly to test/posix/eventfd, limit this test execution on
platforms with at least 32K of RAM available.

Fixes #38601
closed 2021-09-20 by cfriedt