zephyrproject-rtos/zephyr #38602: modem gsm

tcpipchip, 2021-09-16:
Hi

I am compiling the GSM code to NRF52840, but i have a question

My UART0 is working (console), but where do i define the GPIO used to UART1 (GSM) ? (TX and RXD)

Thanks

ycsin, 2021-09-20:
Not sure if I understood you, you will need something like [this](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/gsm_modem/boards/frdm_uart2_dts.overlay) in an `.overlay` in your application directory or inside your `someboard.dts` file.

TX/RX gpios are usually already in the `*.dtsi` file which the `someboard.dts` includes, if you are using an in-tree board.

tcpipchip, 2021-09-20:
I am sorry sorry! Tx and rx are working! I can get IP using generic modem on Lisa u200 and Sara g350. I have to know why I can’t get the gateway and mask. Boths return 0.0.0.0

ycsin, 2021-09-23:
> I have to know why I can’t get the gateway and mask. Boths return 0.0.0.0

I have no idea what are these, @rerickson1 @jukkar any idea?

rerickson1, 2021-09-23:
> > I have to know why I can’t get the gateway and mask. Boths return 0.0.0.0
> 
> I have no idea what are these, @rerickson1 @jukkar any idea?

I would check the AT command documentation for the two modems you are using to see if they support the command to retrieve the gateway and mask.
Also, make sure your modem's firmware is up to date.

tcpipchip, 2021-09-23:
Good morning team!

I am back!
Let me explain about my tests, more details

I am using the zephyr GENERIC MODEM driver...i am testing on LISA U200, SARA G350 and SARA G450,.
There modem  works with ppp , because i tested on Windows Dialup, Raspberry ppp, OpenWrt ppp and finally on Nuttx Rtos.

In the Nuttx Rtos, i see that really not  back the  gateway and dns, then really looks normal, because i can PING to anywhwere, look
[image]

The only difference on ZEPHYR generic modem is that there is no Authentication (PAP) during PPP, and i need that with my SIM operator to works.

Maybe if the ZEPHYR generic modem has a PAP option will solve the problem! Ahh, i remember that the generic modem can make DNS during ping...

I hope that help you this comments

closed 2021-09-23 by nashif