zephyrproject-rtos/zephyr #38606: drivers: adc: stm32h7: Oversampling only supported for ADC3

heinwessels, 2021-09-16:
**Describe the bug**
This [PR](https://github.com/zephyrproject-rtos/zephyr/pull/36714/) provided the ability to do ADC oversampling, which then also added support for the H7 in this [PR](https://github.com/zephyrproject-rtos/zephyr/pull/37392). This will however not set the ratio correctly.

According to the [Zephyr ADC docs](https://docs.zephyrproject.org/latest/reference/peripherals/adc.html#c.adc_sequence.oversampling) the value `sequence->oversampling` should be `2^N`. This means if you require a ratio of 256 you would define `sequence->oversampling=8`. But using ADC1 or ADC2 with the current code if you set `sequence->oversampling=8` it will result in a ratio of 7, and not 256. It requires the ratio in a different format than for ADC3 for which it seems this driver was written.

**For ADC1 and ADC2:**
> OSVR[9:0]: Oversampling ratio
This bitfield is set and cleared by software to define the oversampling ratio.
0: 1x (no oversampling)
1: 2x
2: 3x
...
1023: 1024x

**For ADC3:** 
> OVSR[2:0]: Oversampling ratio
This bitfield is set and cleared by software to define the oversampling ratio.
000: 2x
001: 4x
010: 8x
011: 16x
100: 32x
101: 64x
110: 128x
111: 256x

If using ADC1 or ADC2 I would recommend simply stating:
```
uint16_t oversampling_ratio = (1<<sequence->oversampling) - 1;
...
LL_ADC_ConfigOverSamplingRatioShift(adc, oversampling_ratio, ...);
```

heinwessels, 2021-09-16:
There is also a small bug in the HAL drivers which affects the H7 ADC Oversampling in some cases.
https://github.com/STMicroelectronics/STM32CubeH7/issues/177

FRASTM, 2021-09-17:
Moreover, it depends on the soc device in the stm32H7 family:
stm32h723x has ADC1/ADC2  with Oversampling ratio 0-1023  and ADC3 with Oversampling ratio  [2:0] 
stm32h743x has ADC1/ADC2  with Oversampling ratio 0-1023

Additionnaly when the Oversampling ratio is  10bit-wide  the PCSEL register exists

FRASTM, 2021-09-17:
Additionnaly when the Oversampling ratio is  10bit-wide  the PCSEL register exists  (consequently, on stm32h723x, ADC3 has no PCSEL register), **./drivers/adc/adc_stm32.c** :
```
@@@ -386,7 -386,7 +386,7 @@@ static int start_read(const struct devi
  	 * This register controls the analog switch integrated in the IO level.
  	 * NOTE: There is no LL API to control this register yet.
  	 */
--	adc->PCSEL |= channels & ADC_PCSEL_PCSEL_Msk;
++	LL_ADC_SetChannelPreSelection(adc, index);
  #endif
  
  #if defined(CONFIG_SOC_SERIES_STM32F0X) || \
```

heinwessels, 2021-09-17:
> what about

IMO it would be simpler to define the oversampling once, and not bother with the big `switch`. All H7 devices will need the `1<<sequence->oversampling`, but only if it's not ADC3.
```

static const uint32_t table_oversampling_shift[] = {
	0,
	LL_ADC_OVS_SHIFT_RIGHT_1,
	LL_ADC_OVS_SHIFT_RIGHT_2,
	LL_ADC_OVS_SHIFT_RIGHT_3,
	LL_ADC_OVS_SHIFT_RIGHT_4,
	LL_ADC_OVS_SHIFT_RIGHT_5,
	LL_ADC_OVS_SHIFT_RIGHT_6,
	LL_ADC_OVS_SHIFT_RIGHT_7,
	LL_ADC_OVS_SHIFT_RIGHT_8,
};

	uint32_t oversampling_ratio = sequence->oversampling;
#if defined(CONFIG_SOC_SERIES_STM32H7X)
	/* All H7 devices' ADC1+2 is 10bit */
	if (adc != ADC3) {
		oversampling_ratio = 1<<sequence->oversampling;
	}
#endif
	if (sequence->oversampling == 0) {
		LL_ADC_SetOverSamplingScope(adc, LL_ADC_OVS_DISABLE);
	} else if (sequence->oversampling <= 8) {
		LL_ADC_SetOverSamplingScope(adc, LL_ADC_OVS_GRP_REGULAR_CONTINUED);
#if defined(CONFIG_SOC_SERIES_STM32H7X) && defined(ADC_VER_V5_V90)
		/* Set bits manually to circumvent bug in LL Libraries
		 * https://github.com/STMicroelectronics/STM32CubeH7/issues/177
		 */
		if (adc == ADC3) {
			MODIFY_REG(adc->CFGR2, ADC_CFGR2_OVSS,
				table_oversampling_shift[sequence->oversampling]);
			MODIFY_REG(adc->CFGR2, ADC3_CFGR2_OVSR,
				(oversampling_ratio - 1UL) << ADC3_CFGR2_OVSR_Pos);
		} else {
#endif
			LL_ADC_ConfigOverSamplingRatioShift(adc, oversampling_ratio,
				table_oversampling_shift[sequence->oversampling]);
#if defined(CONFIG_SOC_SERIES_STM32H7X) && defined(ADC_VER_V5_V90)
		}
#endif
	} else {
		LOG_ERR("Invalid oversampling");
		return -EINVAL;
	}
```

---

> For H7 ADC1/2, any sequence->oversampling from 0..1023 is accepted, but should the driver configure it ?

If I understand your question correctly, then I believe the driver should do as the docs say and do `oversampling_ratio = 1<<sequence->oversampling;`. This means you won't be able to have custom ratio's of 100 or 200, and always `2^N`, but I do not believe that would be a problem.

-- 

I'm also not sure why you're using `2 << (sequence->oversampling - 1)` instead of simply stating `1<<sequence->oversampling`? It will result in the same answer.

FRASTM, 2021-09-17:
changing from `2 << (sequence->oversampling - 1) `to `1<<sequence->oversampling`