zephyrproject-rtos/zephyr #38609: RFC: API Change: IPM

carlocaione, 2021-09-16:
### Problem Description:
The IPM subsystem is used to send signals (or small messages) to other processors / CPU clusters. The current API has one big limitation: it assumes that the platform has only two CPUs (or one single channel). Looking at the current API the `ipm_send()` function doesn't take any parameter in input related to the destination CPU and the `ipm_register_callback()` function is only able to register one single callback, so the implicit assumption is that the IPM device is sending / receiving signals to / from a single remote endpoint / CPU.

When the hardware allows multiple channels (that is multiple CPUs) the solution adopted for example by the NRFX IPC driver was to allocate several instances of the same device (see https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/ipm/ipm_nrfx_ipc.c#L236) so that (among other things) was possible to register a different callback for each instance. This is of course highly discouraged when using DT because the IPM device instance should have only one single node into the DT.

### Proposed change:
The proposed change is to add a new `channel` parameter to the `ipm_send()` and `ipm_register_callback()` functions so that the user driver can send data through a specific channel on TX and hookup a callback on a specific channel on RX. We are also introducing a couple of DT helpers to allow the user devices to specify the channels of interest directly into the DT.

### Detailed RFC:
Everything is detailed in the PR https://github.com/zephyrproject-rtos/zephyr/pull/38523 and in the `ipm.h` file at https://github.com/zephyrproject-rtos/zephyr/blob/ea7831e7b3f20371181f976ae610dd5730b5c4b4/include/drivers/ipm.h#L22-L57

carlocaione, 2021-09-17:
With this change we can now fully support the new mailbox bindings as proposed in #38616

carlescufi, 2021-09-21:
API meeting:

- Change looks fundamentally fine
- Switch from typedef to struct for channels
- Switch from int to k_timeout_t for timeouts

carlocaione, 2021-09-22:
> Switch from int to k_timeout_t for timeouts

@carlescufi I'm not sure anymore about this change. Looking at the actual use of the `wait` parameter it is used mostly as a `boolean` value (wait or not) than an actual timeout value. So my suggestion is to leave it as is or change it to `boolean` instead.

carlocaione, 2021-10-13:
Actual PR https://github.com/zephyrproject-rtos/zephyr/pull/38523

carlocaione, 2021-10-13:
> If you instead expand this to an open/close semantics to allocate/access a channel on the send side, make that return something like a 'struct ipm_channel' pointer and use that instead of the ipm_dev + channel, you won't need to touch pieces of code that send the messages along these channels if they end up being renumbered. It would make for a more durable approach, with little to no overhead compared to the proposed ipm_dev + channel id arguments.

Something close to this was already proposed in the PR with the introduction of the `struct ipm_dt_spec` and the various `*_dt()` helpers (see from https://github.com/zephyrproject-rtos/zephyr/blob/94a5f119d038a1d1e40d68d2e18f35105ef875d1/include/drivers/ipm.h#L355-L358 to the end of the file).

In that case all the information is coming from the DT, so if the channel end up being renumbered the user must only have to change the channel numbers in the DT but not the actual code.

olofj, 2021-10-13:
> Something close to this was already proposed in the PR with the introduction of the `struct ipm_dt_spec` and the various `*_dt()` helpers

The ergonomics of needing to add dt client interfaces at the call sites is even worse than the hard-coded use cases. Getting it from DT at the time of "open" might be reasonable, but having to do it everywhere doesn't make sense.

So, my original feedback still stands, better abstraction on the sending side would be beneficial here.