zephyrproject-rtos/zephyr #38612: Fault with assertions enabled prevents detailed output because of ISR() assertion check in shell function

lairdjm, 2021-09-16:
**Describe the bug**
If we have a firmware configured with a shell, a log and assertions enabled, when a fault happens the details should get printed to the screen, but this does not happen due to https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/shell/shell_ops.c#L481 in which cause we see the following output:
```
ASSERTION FAIL [!k_is_in_isr()] @ WEST_TOPDIR/zephyr/subsys/shell/shell_ops.c:481
Thread context required.
[01:39:59.425,964] <err> os: ***** MPU FAULT *****
```

Because the fault prompted an ISR to handle the fault, this means we lose all debug information because of this check which then reboots the module. I'm sure in a normal firmware you would want to check this, but this check is really not helpful for firmware development.

**Expected behavior**
If there is a fault, to not have the debug log reduced to a single line which makes it worthless because of this assertion check and for the full log to be printed

**Impact**
Major annoyance, I'm just removing the assertion check on this line in my local build because it's so annoying

jakub-uC, 2021-09-16:
@lairdjm Are you using function `z_shell_fprintf`? I am just wondering how you are getting it?
This function is internal to the shell module and shall not be used by the user application or any other module.

lairdjm, 2021-09-16:
We are not calling it from our code full stop, FAULT_DUMP is 2 which means when a fault occurs, the data gets dumped to the screen, and it seems to be when that happens it's hitting this assert.

jakub-uC, 2021-09-21:
@lairdjm 
I shall have a fix shortly.

PR #38960 (2021-10-01): shell: fix assert in panic mode
In panic mode, the function: z_shell_fprintf is expected to be called from an interrupt context. Therefore, the dedicated assert
cannot be checked in this case.

Fixes #38612
commit 8b8a3bbf (2021-10-01): shell: fix assert in panic mode

In panic mode, the function: z_shell_fprintf is expected to be
called from an interrupt context. Therefore, the dedicated assert
cannot be checked in this case.

Fixes #38612
closed 2021-10-01 by cfriedt