zephyrproject-rtos/zephyr #38619: PM: Simplify device power states

ceolin, 2021-09-16:
## Introduction

Currenty Zephyr defines distint power states for SoC and devices. How they are related is not
clear and a source of problems. This RFC aims to provide some clarity to this area and provide
more flexibility for applications.

### Problem description

There are four power states for devices and they are used by the
power subsystem according the power state that the system is
transitioning to.

[image]

This relationship is arbitrary (hardcoded) and expects devices drivers to either implement them or return
ENOTSUP. Otherwise the system will fail to transition to a sleep state and continue active.

The majority of device end up just implementing SUSPENDED and ACTIVE states. Some
of them implements LOW_POWER and SUSPENDED the same way.

Another problem with multiple states is that this is disconnected with what device runtime API provides, 
suspend (pm_device_put) and active (pm_device_get) only.

### Proposed change

Remove LOW_POWER. The system will basically use suspended and active
states. Devices should decide which state to use if they support
more than one low power state.

More information has to be provided to devices to them be able to
select the most appropriated state though. So, the power management
callback implemented by drivers has to be changed. Two additional
parameters are proposed:

```
uint8_t flags; /* Who is asking for it ? PM_DEVICE_RUNTIME, PM_SUBSYSTEM, ... */
enum pm_state soc_state; /* SoC pm state */
```

The final version of the callback will look like:

```
typedef int (*pm_device_control_callback_t)(const struct device *dev,
					    enum pm_device_action action,
					    enum pm_state soc_state,
					    uint8_t flags);
```

Most drivers will not be impacted right now, since they already support only SUSPENDED and ACTIVE.  Devices that support multiple states will now be abl  to do it properly having enough information to select the best one for a given situation.
The power subsystem will be simpler since will not need to make assumptions about device power states.

### Dependencies

This change will impact how device drivers support PM.

### Concerns and Unresolved Questions

How applications can fine tune which low power state a device should use (if more than one is implemented). A new function will possibly be added to do this. This can be very fancy like allow to map all possible combinations of `soc_state` and `flags` (called from PM_DEVICE_RUNTIME, from PM_SUBSYSTEM, from another host OS, ...) or as simple as select a particular state every time the device is requested to suspend.

## Alternatives

ceolin, 2021-09-16:
@nashif @erwango @gmarull @JordanYates @aurel32  ^^^^

aurel32, 2021-09-21:
> More information has to be provided to devices to them be able to
> select the most appropriated state though. So, the power management
> callback implemented by drivers has to be changed. Two [...]

Would that be possible to also add the `soc_substate` in addition to the `soc_state`? For example on STM32, `PM_STATE_SUSPEND_TO_IDLE` is mapped to STOP0, STOP1 and STOP2 modes using substates. Depending on the substate, the various peripheral might need specific actions or not.

ceolin, 2021-09-21:
> 
> Would that be possible to also add the `soc_substate` in addition to the `soc_state`? For example on STM32, `PM_STATE_SUSPEND_TO_IDLE` is mapped to STOP0, STOP1 and STOP2 modes using substates. Depending on the substate, the various peripheral might need specific actions or not.

Yes, it should probably be the `pm_state_info` struct instead of only the enum.

mmahadevan108, 2021-09-23:
Can `Device_State_Off` be removed as well so there are only 2 states for the device i.e `suspend `& `active`. Drivers can use the `soc_state` argument to decide the low power operation to perform

gmarull, 2021-09-27:
I'd not extend callback, but provide some APIs to obtain contextual information if the driver requires it. For example, `pm_state_get_next()` instead of passing `enum pm_state soc_state`. Many devices will never care about soc_state, so better make their life easier.

https://github.com/zephyrproject-rtos/zephyr/issues/37056 also describes some problems outlined here, together with information that can be valuable for this discussion.

gmarull, 2021-09-27:
> Can `Device_State_Off` be removed as well so there are only 2 states for the device i.e `suspend `& `active`. Drivers can use the `soc_state` argument to decide the low power operation to perform

That's indeed a good point.

ceolin, 2021-09-27:
> > Can `Device_State_Off` be removed as well so there are only 2 states for the device i.e `suspend `& `active`. Drivers can use the `soc_state` argument to decide the low power operation to perform
> 
> That's indeed a good point.

I was thinking about it as well. Unless someone else come up with a case where it makes sense, I agree that ACTIVE and SUSPENDED are enough.

ceolin, 2021-09-27:
> I'd not extend callback, but provide some APIs to obtain contextual information if the driver requires it. For example, `pm_state_get_next()` instead of passing `enum pm_state soc_state`. Many [...]

That is a good point, just wondering if making it a different function, instead of a parameter, will not make it harder for drivers developers that need it.