zephyrproject-rtos/zephyr #38620: STM32 uart driver prevent system to go to deep sleep

vgeneves, 2021-09-17:
**Describe the bug**
When using uart driver on STM32 (tested on `nucleo_wb55rg`) system can't go to sleep.

When printing logs (or printk) in the console using `uart_poll`, transfer complete isr is only called for the last byte of the string and not the previous ones.
In consequence `pm_constraint_release(PM_STATE_SUSPEND_TO_IDLE);` is not called and prevent system going to sleep.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nucleo_wb55rg -s zephyr/samples/boards/stm32/power_mgmt/blinky/ -d build-pm
2. west flash
3. west debugserver

**Expected behavior**
Transfert complete isr is called for each bytes and `pm_constraint_release` is called.

**Impact**
System can't go to deep sleep state

**Logs and console output**
From gdb after printing some data (file `pm/pm_ctrl.c`):
```
p power_state_disable_count
$1 = {0, 0, 76, 0, 0, 0, 0}
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain : zephyr-sdk.0.13.0
 - Commit SHA : 81595ed4db2081e227d6425acb25b6eec8191787

**Additional context**
Add any other context about the problem here.

ycsin, 2021-09-17:
@erwango ^^^

Hey @vgeneves , there's an inflight PR about a patch for the uart driver #38521, maybe you can try that first?

vgeneves, 2021-09-17:
Hi,

I just tried the patch but it does not solve the problem.

erwango, 2021-09-17:
Looking to following figure, it's clear that current implementation (`pm_constraint_set` on `poll_out`, `pm_constraint_release` on TC ISR) can't work for a stream of bytes:
[image]
For a stream of bytes, we should not call `pm_constraint_set` for every `poll_out` calls, but rather we should call it only once for each frame, which means that we should be able to detect streams transmissions, what we don't do today.

#38521 should take that into account.

erwango, 2021-09-17:
@vgeneves Would you mind checking https://github.com/zephyrproject-rtos/zephyr/pull/38521 ?

vgeneves, 2021-09-21:
I replied in the PR thread https://github.com/zephyrproject-rtos/zephyr/pull/38521. This fixes the issue on my side.