zephyrproject-rtos/zephyr #38622: subsys/usb: CONFIG_USB_DEVICE_STACK resulted in 10kb increase in firmware size

ycsin, 2021-09-17:
**Describe the bug**
I was getting this in my compilation logs when if I enable `CONFIG_USB_DEVICE_STACK` when I'm building the mcuboot for a G0B1 custom board.

```
CMake Warning at C:/Users/yongc/zephyrproject/zephyr/CMakeLists.txt:717 (message):
  No SOURCES given to Zephyr library: modules__hal_nxp__usb

  Excluding target from build.
```

The firmware increased 10kb in size with `CONFIG_USB_DEVICE_STACK`, I certainly expect the size to increase as this option will compile the entire USB subsys and driver, but the warning makes me suspicious if some unrelated things were compiled together with `CONFIG_USB_DEVICE_STACK`.

**To Reproduce**
Build mcuboot with or without `CONFIG_USB_DEVICE_STACK` and check the size difference

**Environment (please complete the following information):**
 - OS: win10
 - gnuarmemb
 - f83dc928f6

ycsin, 2021-09-17:
@jfischer-no

mmahadevan108, 2021-09-17:
PR https://github.com/zephyrproject-rtos/zephyr/pull/38471 should fix the warning. 
commit: 
https://github.com/zephyrproject-rtos/zephyr/pull/38471/commits/c5190f26e4ee1ced6b6b92bc03d714d2cbf82104

carlescufi, 2021-09-22:
I think 10KB when enabling USB is not unreasonable. Let us close this issue with #36471 and then you can open a new enhancement if you would like to track the size of the USB stack.

PR #38471 (2021-09-28): Cleanup CMake code to avoid creation of empty Zephyr libraries
Fixes: #38403
Fixes #38622

This is a series of commits to followup on #37512 by ensuring empty Zephyr libraries are not created.

With this series of commits, then several occurrences of:
```
No SOURCES given to Zephyr library: <foobar>
    
Excluding target from build.
```

has now been fixed.

---
This change is [[image]](https://reviewable.io/reviews/zephyrproject-rtos/zephyr/38471)
closed 2021-09-28 by carlescufi